INTI-CMNB/KiBot #304: [BUG] When using pcb_print output with drill_marks enabled, lot's of assert warnings are emitted

matthiesenj, 2022-09-27:
**Describe the bug**
When running pcb_print with drill_marks enabled on a project with drilled holes, a lot of asserts are emitted in the console. On a semi-complex project, hundreds of these asserts are emitted each time pcb_print is invoked.

Asserts:
> ./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
./include/macros.h(132): assert "lower <= upper" failed in Clamp().
...

This happens when `drill_marks` is set to 'full' or 'small', but not with none. It looks like a C-style assert message, so I wonder where it comes from.

**To Reproduce**
Run kibot on the attached Kicad project: `kibot -e assert.sch -c assert.kibot.yaml` and observe the console output.

[assert.zip](https://github.com/INTI-CMNB/KiBot/files/9656928/assert.zip)

**Expected behavior**
I expect these asserts not to happen/show.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image: `kicad_auto:latest` on Windows 10

set-soft, 2022-09-27:
These asserts are generated by KiCad code

set-soft, 2022-09-27:
The asserts were originated by setting vias with diameter == drill == 0 to make them invisible. So now on KiCad 5 we use diameter = 2 > drill = 0

commit ee967235 (2022-09-27): [Fixed] KiCad 5 "assert "lower <= upper" failed in Clamp()"

Closes #304
closed 2022-09-27 by set-soft