INTI-CMNB/KiBot #311: [HOW]

N0cynym, 2022-09-30:
**What do you want to achieve?**
Let the `extra_fields` option in `interactive_bom` [urlify](https://github.com/openscopeproject/InteractiveHtmlBom/issues/82#issuecomment-1228665488) my URL string

**Do you have some configuration file (.kibot.yaml) that you are using?**

```yaml
  - name: 'interactive_bom'
    comment: "Interactive Bill of Materials in HTML format"
    type: ibom
    dir: .
    options:
      name_format: 'ibom'
      show_fabrication: true
      hide_silkscreen: true
      highlight_pin1: true
      extra_fields: 'Datasheet'
      checkboxes: 'Sourced,Placed,Bogus'
      output: MyProject.ibom.html
```

**Environment (please complete the following information):**
I'm running KiBot:
- In `GitLab`
- Using docker image kicad_auto:dev_k6`

set-soft, 2022-09-30:
Hi @N0cynym !
The mechanism described in https://github.com/openscopeproject/InteractiveHtmlBom/issues/82#issuecomment-1228665488 is really weak, you must be able to write to the *web* folder and modify the plug-in. I think @qu1ck should consider adding a command line option to provide a path for *user* files. Otherwise the solution is not usable in many cases.

That said, I think this problem could be solved from the KiBot side, adding some transformation filter to *urlify* selected fields.

N0cynym, 2022-09-30:
Ok so it's as I thought.

So best is to reopen [issue](https://github.com/openscopeproject/InteractiveHtmlBom/issues/327) and update description for the pipeline usecase so that @qu1ck could get a idea why a flag like `urlify` as command would be great to have for [InteractiveHtmlBom](https://github.com/openscopeproject/InteractiveHtmlBom)!?

set-soft, 2022-09-30:
I think he will pick the idea from this discussion, but I'll try to add the above mentioned filter.

closed 2022-09-30 by set-soft

qu1ck, 2022-09-30:
I'll add the cli flag to specify user file dir.

qu1ck, 2022-09-30:
Actually after implementing it I realized that I will have to scrap the whole thing when proper template support will be added because it will rework the user files customization entirely. So I'd rather prioritize doing that instead of adding a short lived feature.

commit a9d5b8cd (2022-09-30): New filter `urlify` to convert URLs in fields to HTML links

Closes #311