INTI-CMNB/KiBot #318: [BUG] Old version of pcbdraw causes failure in plotting advanced contours/panels

korken89, 2022-10-11:
**Describe the bug**
When generating output images with `pcbdraw` the output is distorted due to bugs that are fixed in v1.0.0.
So it would be awesome if the dockers could be updated with the new version of `pcbdraw`.

Thanks for a great tool!

**Screenshots**

Before v1.0.0 and with v1.0.0

[image]

[image]

set-soft, 2022-10-12:
Hi @korken89 !
PcbDraw 1.0.0 isn't ready yet. I have a working fork, but it still have plenty of patches not yet merged by the upstream.
I can add 1.0.0 to the docker images, but this won't be the upstream 1.0.0, making people think they could use the upstream 1.0.0.
I see various options here:
1. Wait until the 1.0.0 fixes all the needed details, is going really slow.
2. Backport the fixes. Do you know what fixes are needed? The 0.9.0 found in the docker images has various patches to fix problems with the edge
3. Just drop the support for upstream pcbdraw and only support patched versions.

korken89, 2022-10-12:
Thanks for checking!

Unfortunate that it was not ready, I made an original issue here but the answer indicated that upgrading to 1.0.0 should solve the issues: https://github.com/yaqwsx/PcbDraw/issues/129
I looked at the commits between 0.9.0 - 1.0.0, and I can't really find what caused it to start working, so backporting needs someone better than me at Python :)

If it goes as slow as you say it could be worth just using the patched versions until the upstream catches up, but I'm unsure of what issues this might bring.

set-soft, 2022-10-12:
Hi @korken89 !
I'm doing a new release. After this I'll add some experimental patched version to the development images.

set-soft, 2022-10-12:
Hi @korken89 !
Do you have an example that shows this problem? Perhaps a simplified version of your design, I guess the problem is with the outline, not the traces, so you can remove them if they are private.

korken89, 2022-10-12:
Hi, this has the issue: https://github.com/korken89/board-render-error

I run `kibot --verbose -b panel.kicad_pcb -d Fabrication -s run_drc,run_erc board_image_top board_image_bottom`

with the following config:

```
# Example KiPlot config file
kiplot:
  version: 1

preflight:
  run_erc: true
  update_xml: true
  run_drc: true
  check_zone_fills: false
  ignore_unconnected: false

global:
  drc_exclusions_workaround: true
#   kiauto_wait_start: 60
#   kiauto_time_out_scale: 2

outputs:
  - name: 'board_image_top'
    comment: "Image of PCB"
    type: pcbdraw
    options:
      bottom: false
      format: "png"

  - name: 'board_image_bottom'
    comment: "Image of PCB"
    type: pcbdraw
    options:
      bottom: true
      format: "png"

  - name: 'print_sch'
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: .
    options:
      output: Schematic.pdf

  - name: "interactive_bom"
    comment: "Interactive Bill of Materials (HTML)"
    type: ibom
    dir: Interactive BoM
    options:
      blacklist: "DNF*"

  - name: 'gerbers'
    comment: "Gerbers for the board house"
    type: gerber
    dir: Gerbers
    options:
      # generic layer options
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: false
      use_aux_axis_as_origin: false
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: true
      check_zone_fills: true

      # gerber options
      line_width: 0.1
      subtract_mask_from_silk: false
      use_protel_extensions: false
      gerber_precision: 4.6
      create_gerber_job_file: true
      use_gerber_x2_attributes: true
      use_gerber_net_attributes: true

    layers: "all"
  
  - name: 'zip_gerbers'
    comment: "ZIPed gerbers"
    type: compress
    dir: ./
    options:
      files: 
        - source: "Gerbers/*"

```

set-soft, 2022-10-13:
Hi @korken89 !
Thank you very much for creating the example file.
I verified v0.9.0-4 renders it wrongly. 
I then applied https://github.com/yaqwsx/PcbDraw/commit/a7088f5c653b6a7e9a3ffe166bfa22a391d2019b , as you suggested. It fixed the problem:

[image]

So now I'm releasing v0.9.0-5. The images should be updated.