INTI-CMNB/KiBot #319: [BUG] diff output for schematic only first page

Crzyrndm, 2022-10-14:
Currently attempting to add diff output for Github PR for a Kicad6 project

PCB diffs work amazingly well, the schematic diff however is only generating a single page (page 1) and none of the other sheets in the project. This particular project has about 15 sheets with the first sheet being basically just links between sheets,

```yaml
outputs:
  - name: "schematic-diff"
    comment: "schematic diff"
    type: diff
    dir: diff/
    options:
      output: "${PROJECT}-${VERSION}-%i.%x"
      cache_dir: .cache/
      pcb: false
      only_different: false
      # comparison is with current commit
      new: HEAD
      new_type: git
      # original
      old: KIBOT_LAST-10 # placeholder
      old_type: git
```

```yaml
      - uses: INTI-CMNB/KiBot@v2_dk6 # have tried v2_k6 and v2_dk6
        with:
          # Required - kibot config file
          config: config.kibot.yml
          # optional - prefix to output defined in config
          dir: output
          # optional - schematic file
          schema: "schematic.kicad_sch"
          # optional - PCB design file
          board: "pcb.kicad_pcb"
```

NOTE:
`${PROJECT}` and `${VERSION}` are set in preflight `set_text_variables`

set-soft, 2022-10-14:
Hi @Crzyrndm !
The development version should support it now. 
Thanks for reporting!

closed 2022-10-14 by set-soft

Crzyrndm, 2022-10-17:
@set-soft this doesn't seem to be working
```yaml
  - name: "schematic-diff-branch"
    comment: "schematic diff against main"
    type: diff
    dir: diff/
    options:
      output: "${PROJECT}-${VERSION}-branch-%i.%x"
      cache_dir: .cache/
      pcb: false
      only_different: true
      only_first_sch_page: false # should be the default?
      # comparison is with current commit
      new: HEAD
      new_type: git
      # original
      old: b428c6b047400a179d6b628271ace56436c30c5f # hardcoded for testing
      old_type: git
```
^^ results in "No diff" (or a title page with no changes visible if "only different" is false)

set-soft, 2022-10-18:
Hi @Crzyrndm !
Do you have a public repo with this?

Crzyrndm, 2022-10-18:
Not the primary project I'm setting up. I will see if I can reproduce with an example

set-soft, 2022-10-19:
You can also try enabling debug information and looking at the output to see why you are getting two identical copies during the compare stage.

commit 25d9806b (2022-10-14): [Diff] Added option to compare only the first schematic page.

- So now the default is to compare all the schematic pages.
Closes #319