INTI-CMNB/KiBot #320: [FEATURE] Additional arguments with the Github action

Crzyrndm, 2022-10-14:
The [Github action](https://github.com/INTI-CMNB/KiBot#usage-of-github-actions) appears to not provide any "additional arguments" field that could be used for e.g. [preprocessing](https://github.com/INTI-CMNB/KiBot#doing-yaml-substitution-or-preprocessing)

I came across this while trying to feed in the "old" commit for a diff based on some previous steps in the workflow (depending on context, diff against base branch or the previous release)

a simple `additional_args` field that appends to the kibot invocation would be appreciated

set-soft, 2022-10-14:
Hi @Crzyrndm !
I added `additional_args`, can you test it?
Be careful about quotes, they will be problematic, as usually when handling shell stuff.
The current behavior should be to just concatenate the text at the end of the kibot invocation.

Crzyrndm, 2022-10-14:
```yaml
- uses: INTI-CMNB/KiBot@v2_dk6
        with:
          config: config.kibot.yml
          dir: output
          schema: "schematic.kicad_sch"
          board: "pcb.kicad_pcb"
          additional_args: -E OLD_COMMIT=1234
```

`Error: INTI-CMNB/KiBot/v2_dk6/action.yml (Line: 68, Col: 8): Unrecognized named-value: 'additional_args'. Located at position 1 within expression: additional_args`
?

GarethID, 2022-10-17:
Hi @set-soft , I have the same issue as @Crzyrndm, even without using additional arguments. Exists on both v2_k6 and v2_dk6 versions through GitHub actions, can't get any CI to run as it all errors with the same `Error: INTI-CMNB/KiBot/v2_dk6/action.yml (Line: 68, Col: 8): Unrecognized named-value: 'additional_args'`

Problem doesn't seem to exist in v2_k6_1_3_0 so I have reverted to that to stay online. Thanks for your hard work so far!

set-soft, 2022-10-17:
Hi @Crzyrndm and @GarethID !
Sorry for the glitch, now seems to be ok. ([example](https://github.com/INTI-CMNB/kibot_variants_arduprog/actions/runs/3264254533/jobs/5364640487))

Crzyrndm, 2022-10-17:
Can confirm this is now working as intended (including the `additional_args` field) 👍 
Thanks for this

GarethID, 2022-10-17:
Yep, can confirm it's working too. Great work! Thank you.

commit 090c5943 (2022-10-14): [GitHub Action] Added a mechanism to pass extra arguments

- Currently experimental
Closes #320