INTI-CMNB/KiBot #321: [BUG] Running GitHub Action ERC and DRC failing with "ERROR:Unknown output"

Informaticore, 2022-10-22:
**Describe the bug**
Running the actions ERC and DRC are failing with `ERROR:Unknown output` on the hierarchical sheets of my project.

```
kibot -c kibot.yml -e pcb/*.kicad_sch -b pcb/*.kicad_pcb -d . -s run_erc -v -i
  shell: sh -e {0}
  env:
    schema: pcb/*.kicad_sch
    board: pcb/*.kicad_pcb
    dir: .
ERROR:Unknown output `pcb/esp32-s2-solo.kicad_sch` (kibot - kiplot.py:437)
Error: Process completed with exit code 6.
```

Informaticore, 2022-10-22:
To fix it I set the kicad_sch file and kicad_pcb explicitly to the one file which is used and ignored the hierarchical files - with that approach it is fixed and works

set-soft, 2022-10-24:
Hi @Informaticore !
I'm not sure how did you get this, but using `*.kicad_sch` isn't a good idea if you have more than one SCH file. The shell will expand it to all the matching names, but `-e` takes just one argument.
Also note that KiBot doesn't need it, if you don't specify a name it will take the first found. In your case you don't really need to use `-b`. You most probably need to use `-e` because you have more than one file in the same dir.

Informaticore, 2022-10-24:
Hi @set-soft ,
thanks, I will try it out. I do have multiple SCH file but due to the hierarchical pages they belong together anyway. If I give it the one (main) file all works well. Not sure which one it would use if I leave it blank, might be the wrong one if it takes "the first one found".

Informaticore, 2023-03-27:
@set-soft I tried it again with a new project and latest version and the issue is the same. In the action file of GitHub is a section env:
[image]

the default here is that the path is set to 
```
schema: "pcb/*.kicad_sch"
board: "pcb/*.kicad_pcb"
```
But that seems not to work with hierarchical pages still. In KiCad I am using the following to create new schematic files:
[image]
so called hierarchical page. For the KiBot it only needs the initial / main file to generate everything, there is no need to look for all * files. 

Is there a solution to this somehow? 

To fix it I was setting the path explicitly like in the image above, with that everything works just fine.

Informaticore, 2023-03-31:
This is the whole file:

```
# SPDX-FileCopyrightText: 2022 nerdyscout <https://github.com/nerdyscout/>
#
# SPDX-License-Identifier: CC0-1.0

name: KiBot

on:
  workflow_dispatch:
  push:
    paths:
    - "pcb/*.kicad_sch"
    - "pcb/*.kicad_pcb"
    - "pcb/*.kicad_pro"
    - "**/*kibot.yml"
  pull_request:
    paths:
    - "pcb/*.kicad_sch"
    - "pcb/*.kicad_pcb"
    - "pcb/*.kicad_pro"
    - "**/*kibot.yml"

env:
  schema: "pcb/so-dimm-debugger-programmer.kicad_sch"
  board: "pcb/so-dimm-debugger-programmer.kicad_pcb"
  dir: .

jobs:
# checks
  ERC:
    runs-on: ubuntu-latest
    container:
      image: setsoft/kicad_auto:ki6
    steps:
      - uses: actions/checkout@v3
      - name: run kibot
        run: | 
          kibot -c kibot.yml -e ${{env.schema}} -b ${{env.board}} -d ${{env.dir}} -s run_drc -v -i
  DRC:
    runs-on: ubuntu-latest
    container:
      image: setsoft/kicad_auto:ki6
    steps:
      - uses: actions/checkout@v3
      - name: run kibot
        run: | 
          kibot -c kibot.yml -e ${{env.schema}} -b ${{env.board}} -d ${{env.dir}} -s run_erc -v -i
# documentation
  docs:
    runs-on: ubuntu-latest
    needs: [ERC]
    container:
      image: setsoft/kicad_auto:ki6
    steps:
      - uses: actions/checkout@v3
      - name: run kibot
        run: | 
          kibot -c kibot.yml -e ${{env.schema}} -b ${{env.board}} -d ${{env.dir}} -s run_erc,run_drc -v \
            ibom.html octopart.csv schematic.pdf schematic.svg board_top.pdf board_bottom.pdf
      - uses: actions/upload-artifact@v3.1.1
        if: ${{success()}}
        with:
          name: ${{github.event.repository.name}}_docs
          if-no-files-found: error
          path: |
            ${{env.dir}}/docs/**
            !${{env.dir}}/**/*.ogv
            !${{env.dir}}/**/*.log
# fabrications
  gerbers:
    runs-on: ubuntu-latest
    needs: [DRC]
    container:
      image: setsoft/kicad_auto:ki6
    steps:
      - uses: actions/checkout@v3
      - name: run kibot
        run: |
          kibot -c kibot.yml -e ${{env.schema}} -b ${{env.board}} -d ${{env.dir}} -s run_erc,run_drc -v \
            gerbers.gbr drills.gbr drills.drl pickandplace.csv bom.csv
      - uses: actions/upload-artifact@v3.1.1
        if: ${{success()}}
        with:
          name: ${{github.event.repository.name}}_gerbers
          if-no-files-found: error
          path: |
            ${{env.dir}}/gerbers/**
            !${{env.dir}}/**/*.ogv
            !${{env.dir}}/**/*.log
# cad
  cad:
    runs-on: ubuntu-latest
    needs: [DRC]
    container:
      image: setsoft/kicad_auto:ki6
    steps:
      - uses: actions/checkout@v3
      - name: run kibot
        run: | 
          kibot -c kibot.yml -e ${{env.schema}} -b ${{env.board}} -d ${{env.dir}} -s run_erc,run_drc -v \
            model.step openboardview.brd autocad.dxf
      - uses: actions/upload-artifact@v3.1.1
        if: ${{success()}}
        with:
          name: ${{github.event.repository.name}}_cad
          if-no-files-found: error
          path: |
            ${{env.dir}}/cad/**
            !${{env.dir}}/**/*.ogv
            !${{env.dir}}/**/*.log
# images
  render:
    runs-on: ubuntu-latest
    needs: [DRC]
    container:
      image: setsoft/kicad_auto:ki6
    steps:
      - uses: actions/checkout@v3
      - name: run kibot
        run: | 
          kibot -c kibot.yml -e ${{env.schema}} -b ${{env.board}} -d ${{env.dir}} -s run_erc,run_drc -v \
            board_black_top.svg board_black_bottom.svg
      - uses: actions/upload-artifact@v3.1.1
        if: ${{success()}}
        with:
          name: ${{github.event.repository.name}}_img
          if-no-files-found: error
          path: |
            ${{env.dir}}/img/**
            !${{env.dir}}/**/*.ogv
            !${{env.dir}}/**/*.log
```

Informaticore, 2023-04-03:
So the error was that it could not finde the kicad_sch file (when it was set to *) - setting it to the exact name of the main sch file fixes it. But now I updated to KiCad 7 and receive a different error:

```
ERROR:PcbDraw error: Cannot locate resource /usr/share/pcbdraw/styles/set-black-cu.json; explored paths:
- /usr/share/kibot/pcbdraw
- /github/home/.local/share/pcbdraw
- /usr/share/pcbdraw (kibot - out_pcbdraw.py:500)
```

set-soft, 2023-04-03:
> So the error was that it could not finde the kicad_sch file (when it was set to *) - setting it to the exact name of the main sch file fixes it.

This "error" has nothing to do with KiBot. Is how things work. You don't have to use `*.kicad_sch` for a project with more than one schematic. If the documentation says this please let me know where and I'll fix it. Otherwise this is just basic command line knowledge.

> But now I updated to KiCad 7 and receive a different error:

Please read the docs first, your workflow is using outdated images.

closed 2023-04-03 by set-soft

Informaticore, 2023-04-04:
This is not fixed with latest images. I am using now: image: ghcr.io/inti-cmnb/kicad7_auto:1.6.1

set-soft, 2023-04-04:
What do you mean with "latest images"?