INTI-CMNB/KiBot #323: [BUG] Diff Fails if old file does not exist

jtmullen, 2022-11-01:
**Describe the bug**
When a File is _new_  the diff output gives an error that the file does not exist when attempting to retrieve the old file. 

Shell Output:
```bash
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.4.0', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 386, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 486, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 477, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 404, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_diff.py", line 545, in run
    super().run(name)
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 163, in run
    self.options.run(self.expand_filename(output_dir, self.options.output))
  File "/usr/lib/python3/dist-packages/kibot/out_diff.py", line 479, in run
    self.do_compare(self.old, self.old_type, self.new, self.new_type, name, name_ori)
  File "/usr/lib/python3/dist-packages/kibot/out_diff.py", line 397, in do_compare
    old_hash = self.cache_obj(old, old_type)
  File "/usr/lib/python3/dist-packages/kibot/out_diff.py", line 374, in cache_obj
    return self.cache_git(name)
  File "/usr/lib/python3/dist-packages/kibot/out_diff.py", line 328, in cache_git
    hash = self.cache_file()
  File "/usr/lib/python3/dist-packages/kibot/out_diff.py", line 178, in cache_file
    return self.cache_pcb(name) if self.pcb else self.cache_sch(name)
  File "/usr/lib/python3/dist-packages/kibot/out_diff.py", line 166, in cache_sch
    sch = load_any_sch(name, os.path.splitext(os.path.basename(name))[0])
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 247, in load_any_sch
    sch.load(file, project)
  File "/usr/lib/python3/dist-packages/kibot/kicad/v6_sch.py", line 1823, in load
    with open(fname, 'rt') as fh:
FileNotFoundError: [Errno 2] No such file or directory: '/home/jt/template-pcb/template-pcb.kicad_sch'
```

**To Reproduce**
Set up a diff output (my config below) and have old reference a git branch that does not (yet) have the KiCad project on it. 

Output kibot settings: 
```yml
outputs:
  ## Visual Diff of the Project
  - name: 'visual-diff-schematic'
    comment: 'Generate PDF with differences for schematics'
    type: 'diff'
    dir: './diff/'
    options:
      old_type: 'git'
      old: 'main'
      new_type: 'current'
      only_different: true
      diff_mode: 'red_green'

```

**Expected behavior**
Since the entire schematic is new the output should just be the full schematic all highlighted in green as new. 

**Environment (please complete the following information):**
Running in both: 
- WSL2 Ubuntu (on Windows 11)
- GitHub Actions custom action in kicad_auto:ki6 docker

**Additional context**
Changing "old" to a previous commit with a different version works fine, so the rest of the config is good. 

For this application running in CI I cannot have this action fail whenever someone creates a new sheet and would ideally get an output as described above - at the very least it can't fail.  With this behavior the work around would have to be a script that checks which files are on both branches and calls kibot only on those.  If I want the output to include the new sheets I would also have to separately generate a pdf of those which would be a whole additional output to configure.

jtmullen, 2022-11-02:
Perhaps this is actually a feature request as I think about it more. I can see that some applications failing might be the desired output. Perhaps an option where file that don't exist will be substituted with a "blank" schematic/pcb for the diff purposes.

set-soft, 2022-11-02:
Hi @jtmullen !
I agree that a better handling is needed.
But note this problem is because you are explicitly referencing a nonexistent file. If you add/remove pages to a schematic they will be handled properly. The problem here is that there is no schematic at all.
Please be patient because I won't have much time during the next weeks.

jtmullen, 2022-11-02:
Sounds good, thanks for the clarification!

set-soft, 2022-11-05:
Hi @jtmullen !
Now should work. By default you'll get a diff against a blank file. An option enables an error.
Please let me know if it works for you.

closed 2022-11-05 by set-soft

jtmullen, 2022-11-06:
That sounds great, thanks! I will give it a try this week

commit c3f426a1 (2022-11-05): [Diff][Fixed] Problems when missing file from repo

- When comparing to a repo point where the PCB/SCH didn't exist yet.
Closes #323