INTI-CMNB/KiBot #326: [BUG] some npths cause report generator calculating unreasonably small minimum OAR values and warnings

hkleen, 2022-11-04:
**Describe the bug**
Some footprints that contain non plated through holes cause warnings about either a negative OAR or a really small OAR when generating a report. The minimum outer annular ring will be reported as 0.01 mm (0 mils). One footprint that shows this behavior is "Connector:Tag-Connect_TC2030-IDC-FP_2x03_P1.27mm_Vertical" (see example project in attached .zip file). I guess this is due to kibot trying to deduce the correct drill size for a given hole size when calculating the report values.

**To Reproduce**
[npth_oar_problem.zip](https://github.com/INTI-CMNB/KiBot/files/9937317/npth_oar_problem.zip)
run kibot on the project within the .zip file.

**Expected behavior**
There should be no warning regarding non plated holes, at least if the original hole size is equal to pad size (so the pad is removed completely). The minimum OAR should not be reported as 0.01 mm, but the actual minimum OAR should be reported.

**Environment (please complete the following information):**
Kibot (fresh dev checkout) running on Linux (Ubuntu 22.04)

set-soft, 2022-11-05:
Hi @hkleen !
Thanks for reporting.
The above patch should fix the OAR computation, and remove some of the warnings.
I kept other warnings and added clarifications.
The problem here is the use of drill sizes that doesn't correspond to real drill tools, i.e. 2.3749 mm. The manufacturer will use 2.35 mm or perhaps 2.4 mm. But if 2.35 mm is used (the closer value) you could get a really thin copper ring, that might be a  problem. So it will generate a warning.

closed 2022-11-05 by set-soft

hkleen, 2022-11-07:
Great! works fine now. Keeping the warning is fine, especially since it now provides a hint for solving the issue. Although I'd hope that the PCB manufacturer would adjust the pad size according to the chosen drill size.

Thanks for the quick fix!

set-soft, 2022-11-07:
Hi @hkleen !

> Although I'd hope that the PCB manufacturer would adjust the pad size according to the chosen drill size.

Do you mean choosing a larger drill tool instead of a smaller one? To clarify: rounding up even when the closest value is smaller?
This seems to be logic in this case, but I'm not sure this is really done this way.

hkleen, 2022-11-07:
Hi @set-soft !
What I meant is that I would expect the PCB manufacturer to also reduce the copper pad size if they round to a smaller drill tool size, so that there would be no remaining copper ring.
I just checked the gerber files generated by KiCAD and it seems that at least for non plated holes there is no copper pad generated (see screenshot)
[image]

set-soft, 2022-11-07:
> I just checked the gerber files generated by KiCAD and it seems that at least for non plated holes there is no copper pad generated (see screenshot)

Yes, but remember KiCad is using an hypothetical drill tool that doesn't exist in real world, nobody has a  2.3749 mm drill tool.

hkleen, 2022-11-08:
Of course no MFG has a 2.3749 mm drill tool. But even if the MFG chooses a 2.35 mm drill (or even smaller for that matter) there is really no danger of it leaving a residual annular ring around the hole because there is no copper pad specified in the gerber files. Unless I get something totally wrong here, in that case I'm really sorry.
This is what I wanted to show by including the Screenshot: Purple is top copper, green (not really visible is bottom copper), holes are hidden. So as you can see, it seems that for non plated holes KiCAD suppresses generating a copper pad in the gerber files (at least if the specified hole size matches the specified pad size, I haven't tried anything else yet). Since there is no copper pad, there won't be any residual annular ring, regardless of the actual drill size chosen.

In summary what I meant to say is that for plated through holes these checks of annular ring width after rounding to a real world drill size make sense of course. For NPTH (at least as I understand it) there is not really any problem even if the drill size is rounded down.

set-soft, 2022-11-08:
Hi @hkleen !

You are right, KiCad doesn't generate a pad when the sizes are the same and this is an NPTH hole.

commit c2cd439b (2022-11-05): [Report][Fixed] Problems for NPTH holes with some sizes

- When using NPTH holes with sizes that doesn't correspond to
  real drill tools. It generated bogus reports about wrong OARs.

Close #326