INTI-CMNB/KiBot #327: [BUG] `type: bom` join on columns that may be missing error/infinite loop

Crzyrndm, 2022-11-07:
**Describe the bug**

I'm adding KiBot to a Legacy project where alternative part numbers (MPNs) for some parts are listed using `MPN_1`, `MPN_2`, ...
I wanted to combine these into a single "alternatives" column in the BoM output and ran into an exception

BoM configuration
```yaml
  - name: bom
    type: bom
    options:
      format: XLSX
      columns:
        - References
        - Value
        - MPN
        - Quantity Per PCB
        - Footprint
        - Datasheet
        - name: alternatives
          comment: alternative parts to use if no stock available of primary MPN
          join:
            - field: MPN_1
              text_after: "\n"
            - field: MPN_2
              text_after: "\n"
            - field: MPN_3
              text_after: "\n"
            ...
```
I just listed n=1-10 fields thinking I would future proof it a little (incase more get added no need to update the script)  but that generated this error
```
ERROR:Trace stack: (kibot - error.py:33)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 379, in config_output
    out.config(None)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 142, in config
    super().config(parent)
  File "/usr/local/lib/python3.9/dist-packages/kibot/optionable.py", line 206, in config
    self._perform_config_mapping()
  File "/usr/local/lib/python3.9/dist-packages/kibot/optionable.py", line 181, in _perform_config_mapping
    v.config(self)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_bom.py", line 628, in config
    super().config(parent)
  File "/usr/local/lib/python3.9/dist-packages/kibot/optionable.py", line 206, in config
    self._perform_config_mapping()
  File "/usr/local/lib/python3.9/dist-packages/kibot/optionable.py", line 192, in _perform_config_mapping
    nv.config(self)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_bom.py", line 155, in config
    raise KiPlotConfigurationError("Missing or empty `field` in columns list ({})".format(str(self._tree)))
ERROR:In section 'bom' (bom): Missing or empty `field` in columns list ({'name': 'alternatives', 'comment': 'alternative parts to use if no stock available of primary MPN', 'join': [{'field': 'MPN_1', 'text_after': '\n'}, {'field': 'MPN_2', 'text_after': '\n'}, {'field': 'MPN_3', 'text_after': '\n'}]}) (kibot - error.py:40)
```
The parts in the project have a variable number of listed alternatives (0-3)

Desired output would be a single column (`alternatives`) which is a newline separated list of all values from the listed columns with empty/missing columns being ignored. If none of the listed columns are filled in then empty column output

**Environment (please complete the following information):**
Where are you running KiBot: Github Actions on a Kicad6 project
```yaml
      - uses: INTI-CMNB/KiBot@v2_dk6
        with:
          config: config.kibot.yml
          dir: output
          schema: "schematic.kicad_sch"
          board: "pcb.kicad_pcb"
          verbose: 2 # Debug info level [0-3]
```

set-soft, 2022-11-07:
Hi @Crzyrndm !
Can you post the full config for the bom output?
The error is not an exception, is a report saying one of the entries in the list is empty. A typo (i.e. feild instead of field)?
You get the stack trace because you enabled debug, in this way we can know how the code reached the error message.

About the time-out: Could it be some output that uses KiCad? (most outputs that uses KiAuto tool). The current implementation, using an interposer lib, is quite stable, but you can hit a very long time-out when the servers experiment high load. Not to mention that sometimes the GitHub status gets unsynced, the job looks unfinished, even when it really finished.

Crzyrndm, 2022-11-07:
> The error is not an exception, is a report saying one of the entries in the list is empty. A typo (i.e. feild instead of field)? You get the stack trace because you enabled debug, in this way we can know how the code reached the error message.

Thanks for the hint. I needed to set a (fake) field as well as the join. As soon as I added that it works
```diff
        - name: alternatives
+        - field: alternatives
          comment: alternative parts to use if no stock available of primary MPN
          join:
            - field: MPN_1
```

Timeout could definitely be one of the KiCAD outputs. I've added a job timeout now so that shouldn't be an issue.

closed 2022-11-07 by Crzyrndm