INTI-CMNB/KiBot #328: [BUG] "addsublayer" not managed in dual dielectric layers

technosvitman, 2022-11-09:
This my stackup generated by kicad : 

(stackup
      (layer "F.SilkS" (type "Top Silk Screen") (color "White"))
      (layer "F.Paste" (type "Top Solder Paste"))
      (layer "F.Mask" (type "Top Solder Mask") (color "Green") (thickness 0.01))
      (layer "F.Cu" (type "copper") (thickness 0.018))
      (layer "dielectric 1" (type "prepreg") (thickness 0.12 locked) (material "PR2116") (epsilon_r 4.18) (loss_tangent 0.022))
      (layer "In1.Cu" (type "copper") (thickness 0.018))
      (layer "dielectric 2" (type "core") (thickness 0.2 locked) (material "FR4") (epsilon_r 4.18) (loss_tangent 0.022))
      (layer "In2.Cu" (type "copper") (thickness 0.018))
      (layer "dielectric 3" (type "prepreg") (thickness 0.12 locked) (material "PR2116") (epsilon_r 4.18) (loss_tangent 0.022)
        addsublayer (thickness 0) (material "PR2116") (epsilon_r 1) (loss_tangent 0))
      (layer "B.Cu" (type "copper") (thickness 0.018))
      (layer "B.Mask" (type "Bottom Solder Mask") (color "Green") (thickness 0.01))
      (layer "B.Paste" (type "Bottom Solder Paste"))
      (layer "B.SilkS" (type "Bottom Silk Screen"))
      (copper_finish "None")
      (dielectric_constraints yes)
      (edge_connector bevelled)
    )

kibot return the followinf error : 

Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.4.0', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 372, in main
    outputs = cr.read(cf_file)
  File "/usr/lib/python3/dist-packages/kibot/config_reader.py", line 529, in read
    self._parse_global(data.get('global', {}))
  File "/usr/lib/python3/dist-packages/kibot/config_reader.py", line 244, in _parse_global
    glb.config(None)
  File "/usr/lib/python3/dist-packages/kibot/globals.py", line 347, in config
    self.get_stack_up()
  File "/usr/lib/python3/dist-packages/kibot/globals.py", line 335, in get_stack_up
    ly = PCBLayer.parse(e)
  File "/usr/lib/python3/dist-packages/kibot/kicad/v6_sch.py", line 1514, in parse
    i_type = _check_is_symbol_list(i)
  File "/usr/lib/python3/dist-packages/kibot/kicad/v6_sch.py", line 34, in _check_is_symbol_list
    raise SchError('Orphan symbol `{}`'.format(e.value()))
kibot.kicad.v5_sch.SchError: Orphan symbol `addsublayer`

set-soft, 2022-11-10:
Hi @technosvitman !
Thanks for reporting, didn't notice KiCad allows to specify multiple dielectrics, which is common.
You'll need the `dev` version of KiBot. Let me know if it works for you.

closed 2022-11-10 by set-soft

technosvitman, 2023-01-17:
Hi @set-soft,
sorry for the high latency! I update to the latest master and it works pretty fine!!!
Thanks a lot and congratulations for this nice tool!!!!!

commit 5e340494 (2022-11-10): [Stack-up][Fixed] Problems when using more than one dielectric

Fixes #328