INTI-CMNB/KiBot #329: [BUG] Default inner layer extension for JLCPCB is misleading

andreasWallner, 2022-11-13:
First off thanks for the great tool, this is only a minor inconvenience - but I still wanted to raise it. I'd also be happy to open a PR but wanted to open a bug first as I was not sure about your preferred process.

**Describe the bug**
The default for JLCPCB generation uses `gp%n` as the extension for inner layers, but generates them as positive layers. `gpX` is often used for negative layers and has now twice produced confusion at JLCPCB (for sure easy to rectify though).

**To Reproduce**
Run `kibot --quick-start` which generates a JLCPCB section in the yaml file like:
```
- name: JLCPCB_gerbers
  comment: Gerbers compatible with JLCPCB
...
  options:
...
    inner_extension_pattern: .gp%n
```

**Expected behavior**
Extension should be `g%n`, [JLCPCB](https://jlcpcb.com/help/article/f11e533d5201448ab599d811aada79a5/ea6d13d4f1b143abb2653c324c539c06?title=How-to-generate-Gerber-and-Drill-files-in-KiCad-6) seems to expect that looking at the settings that they recommend.

**Environment (please complete the following information):**
- Ubuntu 22.04
- KiBot 1.4.0, installed via pip

**Additional**
I had a quick check online and found multiple resources that confirm that `gpX` is used for negative layers, e.g.
- https://www.eevblog.com/forum/altium/internal-power-plane/
- https://becman.com//tipsgerber.html
which was also my previous assumption. (I confess, I forgot to check and overlooked it the first time around)

set-soft, 2022-11-23:
Hi @andreasWallner !
Thanks for pointing this out.
Now the examples and quick start files should use `.gN`

closed 2022-11-23 by set-soft

andreasWallner, 2022-11-27:
@set-soft Thanks!

commit cfa54d03 (2022-11-23): [Gerber][Fixed] Extension used for JLCPCB inner layers.

- The .gpN stands for "Gerber Plane N", which are plotted negative.
- Normal (positive) inner layers should be named.gN
- We are plotting them in positive.

Fixes #329