INTI-CMNB/KiBot #330: [FEATURE]

gcormier, 2022-11-15:
I'd like to request an example in the documentation on how to use GitHub Actions cache (https://github.com/actions/cache). I think it could help speed up the actions overall, but I'm unsure on the implications of using it and if it would work well with Kibot?

set-soft, 2022-11-23:
Hi @gcormier !

I'm not familiar with GitHub cache. But I don't see how it can help. If you use the `kicad_auto` docker images then the image contains all the needed stuff. So you won't get anything from the pip cache (one of the GitHub usages). Also, we don't download things that could need to be cached. 
When using the GitHub action the docker image is created on the fly, here a cache could help. But in this case I suggest to switch to using the already created docker images.
If you find something we could cache please mention it.

gcormier, 2022-11-23:
My thought was it would help improve the pipeline speed

[image]

This is the longest action in the pipeline, 31 seconds. I think this would be done with a cache?

set-soft, 2022-11-23:
Hi @gcormier !
Are you running more than one KiBot action in the same workflow? From what I understand the cache helps when you need to get the same data in more than one step of the workflow.

gcormier, 2022-11-24:
This is the pipeline : https://github.com/gcormier/megadesk/blob/master/.github/workflows/fab_outputs.yml

My understanding (which could be wrong) is that the cache is between multiple runs of the same workflow.

"GitHub will remove any cache entries that have not been accessed in over 7 days." - why a 7 day cache if the intent is only inside 1 workflow?

set-soft, 2022-11-25:
Hi @gcormier !
You are right about cache stuff being usable between workflow runs.
But I don't see a way to cache the docker image created for the run. I found some explanations on how to cache layers ([here](https://evilmartians.com/chronicles/build-images-on-github-actions-with-docker-layer-caching)), but we don't have much stuff to cache. The action takes almost all using:

```
 FROM ghcr.io/inti-cmnb/kicad6_auto:latest
```

This is an already created image containing all the tools, and stored at GitHub. This is the step *1/7* in your captured screen. The other steps are really simple, and I don't think they can be cached. 

Most of the time is consumed downloading the layers, but none is created, they are already cached!

And you don't have any control on how the action creates the docker image, otherwise we could avoid the need of two different tags (one for KiCad 5 and another for KiCad 6).

If you find anything we could cache please create a new issue, but I don't see anthing. Also note: GitHub internal downloads are somehow slow, just storing the artifacts from my regression tests takes a long time. Also note that when I switched from storing the docker images at DockerHub to GitHub internal registry I noticed it takes more time to pull the layers, which seems ridiculous. Not a big difference, but consistent.

closed 2022-11-25 by set-soft

set-soft, 2022-12-15:
Hi @gcormier !
I found a useful use case for GitHub caches: make the KiCost prices cache persistent across KiBot runs.
The default cache is 7 days, but you destroy it as soon as the workflow finishes, using a GitHub cache is the solution.
Here I explain it: [kicost_ci_test](https://github.com/set-soft/kicost_ci_test)