INTI-CMNB/KiBot #335: [BUG] Recursive imports do not work fully.

FrankvVeelen, 2022-11-23:
**Describe the bug**
When I tried to recursively import files, in the following configuration setup: project-name.kibot.yaml imports company-default.kibot.yaml which imports jlcpcb.kibot.yaml, I got the following error: `ERROR:In section 'JLCPCB_position' (position): Unknown variant name rotated (kibot - error.py:40)`. It seems that variants are not imported correctly when recursively importing kibot yaml files.

**Expected behavior**
I expected imports to have the same behaviour independent of recursion, but they don't seem to.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a desktop system? Which OS, distro and version?
ubuntu 20.04

Unichord, 2022-11-24:
I have the same behavior. One layer of recursion works fine but two don't, not only for variants but also for outputs.
Variants have to be defined on both files, though.

set-soft, 2022-11-24:
I don't have much time right now.
A very simple example (minimal) showing the problem could help.

set-soft, 2022-11-24:
Hi @Unichord !
The above patch should solve the problem.
It was like this: you tried to import the outputs from `kibot_generated.kibot.yaml` but the intermediate import (`kibot_import.kibot.yaml`) didn't contain any output. In this case the outputs from `kibot_generated.kibot.yaml` were lost.
Now they should be visible.

commit 1f8bad0a (2022-11-24): [Imports][Fixed] Problems with recursive imports

- When the intermediate import didn't contain any of the requested
  elements (i.e. importing outputs, but no outputs in the middle).

Closes #335
closed 2022-11-24 by set-soft