INTI-CMNB/KiBot #336: [FEATURE] subsheets in schematic diffs.

harm363, 2022-11-23:
**Is your feature request related to a problem? Please describe.**
I always have multiple subsheets in my schematic. for review purpose I need to have a diff for all the sheets in the schematic
#333. right now only the main sheet is diffed against a earlier version of itself

**Describe the solution you'd like:
I would really like to have an :

    all_pages:true  
option in my kibot.yaml file to generate a diff for all of the pages at once without the need for workaround scripts.

**Describe alternatives you've considered**
right now i have a workarpound script that calls kicad-diff.py with the neccesary files.

set-soft, 2022-11-23:
Hi @harm363 !
Can you try the current development code? It now defaults to compare all sheets, and you can compare only the first using `only_first_sch_page`.

harm363, 2022-11-24:
great!, unfortunaly the first moment i have available is tuesdeay 29th of november

set-soft, 2022-11-25:
Hi @harm363 !
I'm labeling it as duplicate for #319 and clossing the issue.
If you find any problem with the subsheets please open a new issue.

closed 2022-11-25 by set-soft