INTI-CMNB/KiBot #337: [FEATURE] PDF generation options

Unichord, 2022-11-24:
I don't know if I'm asking for new features or they're already there.

When generating the PCB layout PDF, I can't seem to find a way to adapt the PCB to the sheet, using `scaling: 0`, _but with some margins_.
At the moment, the board edges are right on the edges of the sheet, which can be inconvenient for a number of reasons (printing, for example).

Also, noticed a different behavior (when opening with Acrobat Reader) generated portrait/landscape PDF of the PCBs. When portrait, the pages adapt in width, with smooth scrolling enabled. When landscape, the pages adapt in height, with smooth scrolling disabled, namely I jump from one page to the next.
I don't know if this is customizable, but I'd like to always fit the page to the screen (in both width and height) with scrolling enabled.

Thanks.

set-soft, 2022-11-24:
Hi @Unichord !
Please be more specific, which output are you referring?

About the scrolling: we are not setting any option for the way the document looks when you open it. I guess the jump/scroll is something related to the PDF reader.

Unichord, 2022-11-24:
Hi @set-soft, sorry I should have been more specific. I'm talking about this:
```
- name: basic_pcb_print_pdf
  comment: PCB
  type: pcb_print
  dir: ./
  options:
    format: PDF
    scaling: 0
    output: '%f%v-PCB.%x'
    keep_temporal_files: false
    plot_sheet_reference: false
    pages:
    - layers:
      - layer: F.Silkscreen
      - layer: Edge.Cuts
      sheet: Silk
    - layers:
      - layer: User.Drawings
      - layer: Edge.Cuts
      sheet: Drawings
    - layers:
      - layer: F.Cu
      - layer: F.Mask
        color: '#14332440'
      - layer: F.Paste
      - layer: F.Silkscreen
      - layer: Edge.Cuts
      sheet: Front copper
    - layers:
      - layer: B.Cu
      - layer: B.Mask
        color: '#14332440'
      - layer: Edge.Cuts
      mirror: true
      sheet: Bottom copper
```
I couldn't find a way to set margins around the board edges.

About the scrolling: I imagined you're not setting anything while generating the PDF, I just thought it was weird that it behaved differently when portrait and when landscape. In fact I checked Acrobat Reader settings and I solved the problem.

set-soft, 2022-11-25:
Hi @Unichord !
The 788aabeeedc6bdd77b0884e01eeb6158a89ae2e5 commit should add margins, I tried:

```yaml
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: basic_pcb_print_pdf
    comment: PCB
    type: pcb_print
    dir: ./
    options:
      format: PDF
      scaling: 0
      autoscale_margin_x: 50
      autoscale_margin_y: 25
      output: '%f%v-PCB.%x'
      keep_temporal_files: false
      plot_sheet_reference: false
      pages:
      - layers:
        - layer: F.Silkscreen
        - layer: Edge.Cuts
        sheet: Silk
      - layers:
        - layer: User.Drawings
        - layer: Edge.Cuts
        sheet: Drawings
      - layers:
        - layer: F.Cu
        - layer: F.Mask
          color: '#14332440'
        - layer: F.Paste
        - layer: F.Silkscreen
        - layer: Edge.Cuts
        sheet: Front copper
      - layers:
        - layer: B.Cu
        - layer: B.Mask
          color: '#14332440'
        - layer: Edge.Cuts
        mirror: true
        sheet: Bottom copper
```

commit 788aabee (2022-11-25): [PCB_Print][Added] Margins for the autoscale mode.

Closes #337
closed 2022-11-25 by set-soft