INTI-CMNB/KiBot #339: [BUG] component does not show up in iBoM and KiBoM

dommae123, 2022-11-28:
Hi,
We have a bug in the ibom creation where one component just does not show up in the ibom created by Kibot, although it is not marked as DNF or similar. If we run ibom locally directly from Kicad, the component is listed. If we choose another symbol for the same component in the schematic, it also works from Kibot. The same problem occurs with KiBOM, too.

For us, we use another symbol for this component now as a workaround. Still, this problem can cause huge damage if BOMs just don’t contain all the necessary components and no one will notice.

For reference, please find attached the component data which does not work for some reason:
[component from kicad_sch.txt](https://github.com/INTI-CMNB/KiBot/files/10102446/component.from.kicad_sch.txt)
[component from kicad_pcb.txt](https://github.com/INTI-CMNB/KiBot/files/10102447/component.from.kicad_pcb.txt)

Greetings,
Thomas

set-soft, 2022-11-28:
Hi @dommae123 !
Can you create a very simple project with this component and another one, so I can reproduce the problem?

dommae123, 2022-11-29:
Hi @set-soft 
Thanks for replying so rapidly!
In the zip file, you can find a simple project of a Kicad-Project (including the kibot yaml file that we used), where the pcb-connectors show up in the iBom made by the Kicad-PCB-Editor (locally on my PC), but do not show up, when created by kibot in our Azure pipeline.
Do you need more files to reproduce it?

set-soft, 2022-11-29:
Hi @dommae123 !
The problem is that testpoints are excluded from the BoM. This is the default behavior for a KiBoM variant, and they are excluded when you use the `default` variant, which is a KiBoM variant.
You just need to run KiBot using debug information (i.e. `-vvv` in the command line or `verbose: '3'` in the GitHub action). Using it I get (for the `default` variant):

```
DEBUG:Applying filter `_mechanical` to exclude (kibot - fil_base.py:154)
DEBUG:Excluding 'C1+_3V_Zellabgriff1': Field 'part' (TestPoint) matched 're.compile('test.*point', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'C1-_0V_Zellabgriff1': Field 'part' (TestPoint) matched 're.compile('test.*point', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'C2+_6V_Zellabgriff1': Field 'part' (TestPoint) matched 're.compile('test.*point', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'C3+_9V_Zellabgriff1': Field 'part' (TestPoint) matched 're.compile('test.*point', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'CELL1': Field 'part' (TestPoint) matched 're.compile('test.*point', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'JP1': Field 'part' (SolderJumper_2_Bridged) matched 're.compile('solder.*bridge', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'TP11': Field 'reference' (TP11) matched 're.compile('^TP[0-9]*', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'TP12': Field 'reference' (TP12) matched 're.compile('^TP[0-9]*', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'TP13': Field 'reference' (TP13) matched 're.compile('^TP[0-9]*', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'TP14': Field 'reference' (TP14) matched 're.compile('^TP[0-9]*', re.IGNORECASE)' (kibot - fil_generic.py:164)
DEBUG:Excluding 'TP19': Field 'reference' (TP19) matched 're.compile('^TP[0-9]*', re.IGNORECASE)' (kibot - fil_generic.py:164)
```

Looking at the Schematic (in the text you didn't include in component from kicad_sch.txt):

```
  (symbol (lib_id "Connector:TestPoint") (at 209.55 55.88 0) (unit 1)
    (in_bom yes) (on_board yes)
    (uuid 00000000-0000-0000-0000-00005f55e956)
    (property "Reference" "C1+_3V_Zellabgriff1" (id 0) (at 208.1022 54.0512 0)
```

So `C1+_3V_Zellabgriff1` (which uses an unconventional reference id) is in fact a `TestPoint`.
In fact is quite confusing, because you draw a connector using a testpoint ;-)
You could use `Conn_01x01`, `Conn_01x01_Male`, `Conn_01x01_Female`, etc. if this is a connector pin.

closed 2022-11-29 by set-soft