INTI-CMNB/KiBot #340: [HOW] set_text_variables usage

Unichord, 2022-11-29:
I can't find a way to substitute variables in both PCB/Gerbers and schematic PDFs, probably I'm not using them right.
I'm working on KiCad 6, no WSL2, no CI/CD environment.

This is my preflight (I also tried with `pcb_replace`), the rest is just a simple gerber/PDF generation, if needed just let me know:
```
preflight:
  # pcb_replace:
    # - replace_tags:
        # tag: 'name'
        # text: 'aaaaaaaaa'
  set_text_variables:
    - name: 'b_name'
      text: 'example'
      command: 'echo $KIBOT_PCB_NAME'
    - name: 'b_code'
      command: 'echo 22.4360'
```
I defined the variables as `${b_name}` and `${b_code}`, but I also tried `b_name` and `b_code`, and also `@b_name@` and `@b_code@`, with no result whatsoever. Also in the yaml file I tried setting the name as `'@b_name@'` and so on.

This is a small example of the PCB, I left just the writings and the header, if more is needed let me know:
[FMFR1_example.zip](https://github.com/INTI-CMNB/KiBot/files/10114202/FMFR1_example.zip)

Thanks.

set-soft, 2022-11-29:
Hi @Unichord !

Here is an example: 
[set_text_variables_example.zip](https://github.com/INTI-CMNB/KiBot/files/10115714/set_text_variables_example.zip)

In this example the schematic and the PCB uses `${TO_REPLACE}` twice.
Note that you need a project file.
Running KiBot you get the files included in `Generate`:
[image]
[image]

The text to replace is simply `Replaced text`

closed 2022-11-29 by set-soft

set-soft, 2022-12-01:
Hi @Unichord !
The file you sent me contains:

```json
   "text_variables": {
     "TO_REPLACE": "Replaced text"
   }
```

If you use this project KiCad texts containing `${TO_REPLACE}` will become `Replaced text`. Just rename it to `bom_w_prj.kicad_pro` and use it with the PCB and schematic I attached before and load it.

Unichord, 2022-12-12:
Hello @set-soft, sorry for the delayed reply.
I tried as you suggested, and it weirdly works. I don't know what could be the problem actually, since the preflight is identical in my configuration file (attached before) and in yours.
Still, in my configuration file it doesn't work.

set-soft, 2022-12-12:
Send me a project where it doesn't work and I'll check it in my system.

Unichord, 2022-12-12:
> Thanks! Except with your project it works, but not with mine. If I manually edit my project and replace variables, it replaces them. If I use my configuration file with your project, it works. So maybe the problem is my project. I don't know if it's an important information, but I migrated this from KiCad 5 to 6. I can send you the whole folder (privately), here is just the project file. [FMFR1.zip](https://github.com/INTI-CMNB/KiBot/files/10124008/FMFR1.zip)

Here is the project. If you need other files I can send the to you privately.

set-soft, 2022-12-12:
> Here is the project. If you need other files I can send the to you privately.

Yes, send me them privately. The above project is already configured, did you load it on KiCad?

Unichord, 2022-12-22:
Hello @set-soft, I've sent you an email with the full project a while ago. Have you checked it by chance? Have you even received it? At this point I'm not sure. Thanks.

set-soft, 2022-12-26:
Hi @Unichord !
I didn't get the files. May be some silly filter here. Can you share it using some mechanism, like Google Drive with a private link?

set-soft, 2023-09-01:
Note for future references: the main problem here was the use of `-s all` skipping **all** the preflights, including `set_text_variables`