INTI-CMNB/KiBot #341: [BUG] Kicost not runing with quick-start config

p-h-a-i-l, 2022-11-29:
**Describe the bug**
Kibot prints the error: 
`ERROR:KiCost error: `One of the internal Kitspace services may experiencing problems. Most probably is saturated, please consider using other API.` (5) (kibot - xlsx_writer.py:696)`

**To Reproduce**
running `kibot --quick-start`

**Expected behavior**
Kicost produces a costed BOM

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? No
- In a GitHub action? No
- Using a docker image? `ghcr.io/inti-cmnb/kicad6_auto:dev`

Am I missing something? Do I need to provide API Keys? I had the naive expectation to just run `kibot --quick-start`

set-soft, 2022-11-30:
Hi @p-h-a-i-l !

> Am I missing something?

I think the error is quite descriptive: Kitspace is saturated, take some time to configure an API with more requests.

> Do I need to provide API Keys?

Yes, please read KiCost documentation about it. I recommend experimenting in your local machine (KiCost runs on Linux and Windows), when you get it working then use the same credentials in your CI/CD.

I recommend trying Nexar API first, this is Octopart, but now belongs to the Altium company (cleaver move). But if you could need others if Nexar's limit is not enough for you. Note that all the currently supported APIs has free services. Of course you can pay for a Nexar service to get a generous limit.

closed 2022-11-30 by set-soft

p-h-a-i-l, 2022-12-01:
Thanks a lot, I followed your suggestion to use Nexar, works like a charm!