INTI-CMNB/KiBot #348: [HOW] add API config details to Kicost within Kibot

dommae123, 2022-12-12:
Hi there,

I want to use kicost within kibot. I already have a nexar api account (also the ids) and and mouser api account. 

I'm wondering how to get the API config details into my kibot config yaml file, because there is no section for that as far as I can see under "type: kicost"

[image]

Maybe someone can help me with that. Thank you =)

Greetings, Thomas

set-soft, 2022-12-12:
Hi @dommae123 !
I don't think adding secrets to the KiBot config is a good idea.
You should provide a configuration file.
For mouser you can use the environment variable `MOUSER_PART_API_KEY`.
Nexar API currently lacks the environment mechanism, I'll add it.
Note that when using GitHub you can use repository secrets to set the environment variables. You can also use them to create a file.
Providing the secrets from the KiBot configuration will force you to make them visible to anyone with access to the file. In a public repo this means sharing your keys.

set-soft, 2022-12-15:
Hi @dommae123 !
Note that `xlsx` has an option `kicost_config` to specify a KiCost configuration, also note that KiBot will use your default KiCost config if none specified.
The secrets can't be included in the KiBot configuration because then you can't use it in a public repo.
So the secrets must come from outside. But your question is valid: from where?
This was complex to solve, so I added a couple of features to make it easier.
Please visit: [kicost_ci_test](https://github.com/set-soft/kicost_ci_test) to learn how to do it in a CI/CD environment. Even if you use it in your local machine this example will give you some good ideas about how to separate things.

If you want to run it in your machine first verify things work using the KiCost GUI (KiCad plugin) once it works KiBot will call the KiCost API and use the same configuration.

Note that the example I mention above works with the current git code for KiBot and using KiCost 1.1.15 (which I released today).