INTI-CMNB/KiBot #350: [HOW]

N0cynym, 2022-12-16:
**What do you want to achieve?**
My kibot export of interactive_bom should remove Fab data of components they do not be part of variant.

`kibot -g variant=without_some_component -d BoM/without_some_component -s all interactive_bom`

**Do you have some PCB/Schematic to use as example?**
I can create if necessary.

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
# Example KiPlot config file

kiplot:
  version: 1

preflight:
  run_erc: true
  update_xml: true
  run_drc: true
  check_zone_fills: false
  ignore_unconnected: false

variants:
  - name: 'default'
    comment: 'default'
    type: kibom
  - name: 'without_some_component'
    comment: 'A variant where some components should not be placed'
    type: kibom
    file_id: _without_some_component
    variant: without_some_component

outputs:
  - name: 'interactive_bom'
    comment: "Interactive Bill of Materials in HTML format"
    type: ibom
    dir: .
    options:
      name_format: 'ibom'
      show_fabrication: true
      hide_silkscreen: true
      highlight_pin1: true
      extra_fields: 'Datasheet,Value Component'
      checkboxes: 'Placed,Checked'
      output: iBoM.ibom.html
```

**Environment (please complete the following information):**
Where are you running KiBot:
- GitLab CI
- kicad_auto:ki6

set-soft, 2022-12-16:
Hi @N0cynym !
This should work. The `test_ibom_variant_1` checks variants on iBoM.
It uses [ibom_variant_1.kibot.yaml](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/yaml_samples/ibom_variant_1.kibot.yaml) applied to [kibom-variante.kicad_sch](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/board_samples/kicad_6/kibom-variante.kicad_sch) and [kibom-variante.kicad_pcb](https://github.com/INTI-CMNB/KiBot/blob/dev/tests/board_samples/kicad_6/kibom-variante.kicad_pcb)
Please provide a simple example that fails and describe which component/s are failing

N0cynym, 2022-12-19:
Here I have a example of what I mean.

Ibom does correctly export material list. 
But the Fab Data are incorrect.

Pick and Place data and Paste data are also correct exported.

Is it possible to hide the not placed component Fab data like Paste data?

```
kibot -g variant=variant_1 -d BoM/variant_1 -s all  interactive_bom_variant_1
kibot -g variant=variant_2 -d BoM/variant_1 -s all  interactive_bom_variant_2
```
[Test.zip](https://github.com/INTI-CMNB/KiBot/files/10258834/Test.zip)

set-soft, 2022-12-19:
Hi @N0cynym !
Note that the above files doesn't remove anything, they just change the value for a component. I'll take a deeper look, but the test case doesn't correlate with the problem you mention.

set-soft, 2022-12-19:
Hi @N0cynym !

1. Your PCB is wrong. You have two overlapped components, this is wrong. You should play with the Value of the component. Otherwise you'll have two different references for the same physical footprint.
2. Things aren't removed from *.Fab, they are crossed, like in the schematic. In some outputs you can use `hide_excluded`, I'm adding it to iBoM, but this is not for this case.
3. You are mixing two strategies: a) Define outputs for more than one variant and b) Specify the variant in the command line. The first is good for small cases, the second scales much better.
4. I found a bug about the iBoM generation, not all variant operations were applied to iBoM

The above patch fixes the iBoM issue.
Lets clarify some stuff, here is the test file I used to verify the cross bug:

```yaml
kiplot:
  version: 1

variants:
  - name: 'default'
    comment: 'default'
    type: kibom
  - name: 'variant_1'
    comment: 'Variant with 100 Ohm parallel'
    type: kibom
    file_id: _variant_1
    variant: variant_1  
  - name: 'variant_2'
    comment: 'Variant without 100 Ohm parallel'
    type: kibom
    file_id: _variant_2
    variant: variant_2 

outputs:
  - name: 'interactive_bom_variant'
    comment: "Interactive Bill of Materials in HTML format"
    type: ibom
    options:
      name_format: 'ibom'
      show_fabrication: true
      hide_silkscreen: true
      highlight_pin1: true
      checkboxes: 'Placed,Checked'
      output: Test.ibom.html

  - name: show_pcb
    comment: "Export the PCB with variant"
    type: pcb_variant
    options:
      output: "%f.%x"

  - name: show_sch
    comment: "Export the SCH with variant"
    type: sch_variant

  - name: print
    comment: "Print the PCB front"
    type: pcb_print
    options:
      title: "Variant %V"
      format: 'PNG'
      scaling: 0
      autoscale_margin_x: 50
      autoscale_margin_y: 50
      pages:
        - sheet: Front
          layers:
            - layer: Edge.Cuts
            - layer: F.Cu
            - layer: F.Paste
            - layer: F.SilkS
            - layer: F.Fab
```

Note this config has both variants, but none output says which variant is used. So now you run:

```shell
#!/bin/sh
kibot -g variant=variant_1 -d BoM/variant_1
kibot -g variant=variant_2 -d BoM/variant_2
```

And you'll get the iBoM output, plus the PCB and SCH with the variant applied and a print. I separated R3 from R2, so you can see it properly. 

For the `variant_1`:

[image]

For the `variant_2`:

[image]

This is how things work, but not what you want. So I modified the schematic: 
1. Removed R3
2. Removed the `Config` field for R2 (is always there, just different value)
3. Added `variant_2:Value` with `0 Ohms` to R2

Then I changed the `variant_2` to be:

```yaml
  - name: 'variant_2'
    comment: 'Variant without 100 Ohm parallel'
    type: kibom
    file_id: _variant_2
    variant: variant_2 
    pre_transform: '_var_rename'
```

Just added a transform filter. So now ...

For the `variant_1`:

[image]

For the `variant_2`:

[image]

Here are all the files I used 
[ibom_issue_350.zip](https://github.com/INTI-CMNB/KiBot/files/10260049/ibom_issue_350.zip)

But this is after applying the above mentioned patch.

closed 2022-12-19 by set-soft

N0cynym, 2022-12-20:
@set-soft Thank you very much for helping me in this case!

Will the support of different values still be experimental?
Means is [this](https://github.com/INTI-CMNB/kibot_variants_arduprog#definition) no more up to date?

set-soft, 2022-12-22:
Hi @N0cynym !
This will remain, the *experimental* status is just a warning meaning *is not fully tested, be patient if some use case is not covered, just report it*. I'll clarify it, thanks for asking

commit 3c501d30 (2022-12-19): [iBoM][Fixed] Variant changes to the *.Fab weren't exported

- Now all 2D variant stuff is applied before calling iBoM

Closes #350
commit 5594fa0b (2022-12-19): [iBoM][Fixed] Variant changes to the *.Fab weren't exported

- Now all 2D variant stuff is applied before calling iBoM

Closes #350