INTI-CMNB/KiBot #351: [BUG] PCB_print

MarkusSBS, 2022-12-16:
**Describe the bug**
in the PDF F.Mask andB.mask is inverted and extra text is missing 

**To Reproduce**

  - name: 'print_pcb_pdf'
    comment: "PDF for the PCB"
    type: pcb_print
    dir: PCB_Docu
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      enable_ki6_frame_fix: true
      #sheet_reference_layout: pcb_print.kicad_wks
      scaling: 1.0
      pages:
        - layers: [ F.SilkS, Dwgs.User ]
          sheet: 'Top Silk layer'
        - layers: [ F.Mask, Dwgs.User ]
          sheet: 'Top Mask layer'
        - layers: [ F.Cu, Dwgs.User ]
          sheet: 'Top copper layer'
        #- layers: [ In1.Cu, Dwgs.User ]
        #  sheet: '1 inner layer'
        #- layers: [ In2.Cu, Dwgs.User ]
        #  sheet: '2 inner layer'
        #- layers: [ In3.Cu, Dwgs.User ]
        #  sheet: '3 inner layer'
        #- layers: [ In4.Cu, Dwgs.User ]
        #  sheet: '4 inner layer'
        - layers: [ B.Cu, Dwgs.User ]
          sheet: 'Bottom copper layer'
        - layers: [ B.Mask, Dwgs.User ]
          sheet: 'Top Mask layer'
        - layers: [ B.SilkS, Dwgs.User ]
          sheet: 'Bottom layer'
        - layers: [ F.Paste,Dwgs.User]
          sheet: 'Top paste Fabrication layers'
        - layers: [ B.Paste,Dwgs.User]
          sheet: 'Bottom paste Fabrication layers'
      output: "%v_%p_Rev%r-%i_%D.%x"

**Expected behavior**
Rendering the Mask layer as other layers 

**Screenshots**
Bottom layer:
[image]
Mask:
[image]

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow?  used with GitLab
- Using a docker image? kicad_auto:ki6
- Using a desktop system? Windows

**Additional context**
Add any other context about the problem here.

MarkusSBS, 2022-12-16:
i was expecting that the output looks like this:
[image]

the text in the lower right  is text on the Mask layer.

But the output of kibot looks like inverted and the text is missing

set-soft, 2022-12-16:
Hi @MarkusSBS !
This is on purpose!
I don't see any point in printing the negative of the physical mask. If you need it for fabrication then use any of the plot outputs, like the gerber or PDF output.
The masks are rendered as the other layers: as you see them.
The fact that KiCad prints the mask inverted is most probably because this is how you draw it in the program: inverted. You draw the apertures, not the mask.

About the text: I assume you are drawing "F.Mask" in the solder mask so it gets in the frame ... I understand the hack, but you can't expect to get solder mask *outside* the PCB edge ... sitting on nowhere. 

I'll close this because I don't see any bug here.
If you want to:
1) Have some mechanism to print negative solder masks
2) Have some text variable defined using the layer name (which should solve your problem because you are using a custom wks)
Please open feature requests for them

closed 2022-12-16 by set-soft

MarkusSBS, 2022-12-16:
ok sounds logical what is happenig there. 

The problem is that you sometimes want the mask data, the same as the gerbers with, titel frame for Dokumentation.
 for example if you have a produktion run and you have problem later on or a disput

set-soft, 2022-12-16:
> The problem is that you sometimes want the mask data, the same as the gerbers with, titel frame for Dokumentation.
> for example if you have a produktion run and you have problem later on or a disput

If you really need it just fill a feature request. In fact it can be generated when `inverted` is true, which currently doesn't help at all for masks.

set-soft, 2022-12-16:
Oh! BTW @MarkusSBS, what you are hacking with text outside the board is almost implemented. Try using the **LAYER** text variable.

set-soft, 2022-12-16:
Hi @MarkusSBS !
For the KiCad negative solder mask try disabling `realistic_solder_mask`

MarkusSBS, 2022-12-19:
realistic_solder_mask is what i was searching for thanks