INTI-CMNB/KiBot #352: [BUG] with pcb_print logo in wks template is missing

MarkusSBS, 2022-12-16:
**Describe the bug**
in the export with PCB_print logos added to the doc frame are missing. This is working for the SCH but not for PCB

**To Reproduce**
  - name: 'print_pcb_pdf'
    comment: "PDF for the PCB"
    type: pcb_print
    dir: PCB_Docu
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      enable_ki6_frame_fix: true
      #sheet_reference_layout: pcb_print.kicad_wks
      scaling: 1.0
      pages:
        - layers: [ F.SilkS, Dwgs.User ]
          sheet: 'Top Silk layer'
        - layers: [ F.Mask, Dwgs.User ]
          sheet: 'Top Mask layer'
        - layers: [ F.Cu, Dwgs.User ]
          sheet: 'Top copper layer'
        #- layers: [ In1.Cu, Dwgs.User ]
        #  sheet: '1 inner layer'
        #- layers: [ In2.Cu, Dwgs.User ]
        #  sheet: '2 inner layer'
        #- layers: [ In3.Cu, Dwgs.User ]
        #  sheet: '3 inner layer'
        #- layers: [ In4.Cu, Dwgs.User ]
        #  sheet: '4 inner layer'
        - layers: [ B.Cu, Dwgs.User ]
          sheet: 'Bottom copper layer'
        - layers: [ B.Mask, Dwgs.User ]
          sheet: 'Top Mask layer'
        - layers: [ B.SilkS, Dwgs.User ]
          sheet: 'Bottom layer'
        - layers: [ F.Paste,Dwgs.User]
          sheet: 'Top paste Fabrication layers'
        - layers: [ B.Paste,Dwgs.User]
          sheet: 'Bottom paste Fabrication layers'

**Expected behavior**
rendering the logo like for the SCH

**Screenshots**

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? GitLab
- Using a docker image? kicad_auto:ki6
- Using a desktop system? Windows

**Additional context**
Add any other context about the problem here.

set-soft, 2022-12-16:
Please provide a small example to reproduce it. Logos are supported by the frame renderer.

MarkusSBS, 2022-12-16:
[kicadexampe.zip](https://github.com/INTI-CMNB/KiBot/files/10246983/kicadexampe.zip)

here a small example  with a frame template and a example projektfile with all configs I have 

i hope this is what you want

MarkusSBS, 2022-12-16:
[image]

MarkusSBS, 2022-12-16:
[kicadexampe.zip](https://github.com/INTI-CMNB/KiBot/files/10247082/kicadexampe.zip)

 small example 
 it uses a custom frame that is located in /kicad/6.0/template 

to have it in the CI runner the docker image does a 

`before_script:
   - git -c http.sslVerify=false clone https://gitlab-ci-token:${CI_JOB_TOKEN}@gitlab.blabla/hardware/kicad-template.git /root/.local/share/kicad/6.0/template//`
 
after this it should look like this:
[image]
 but the frame is empty

set-soft, 2022-12-16:
Hi @MarkusSBS !
I think I found the problem, as a temporal workaround try adding `svg_precision: 6` to the pcb_print options

set-soft, 2022-12-16:
Is fixed in the current `dev` version (1.5.2 or newer)

closed 2022-12-16 by set-soft

MarkusSBS, 2022-12-19:
with svg_precision: 6 logo prinout works now but

the soldermask has a offset?!

not sure if this is a bug or just to mutch tinkering 

[image]

set-soft, 2022-12-19:
> not sure if this is a bug or just to mutch tinkering

I tried with one of the test cases using default and 6 and got:

[light_control-assembly_3.pdf](https://github.com/INTI-CMNB/KiBot/files/10260670/light_control-assembly_3.pdf)
[light_control-assembly_6.pdf](https://github.com/INTI-CMNB/KiBot/files/10260675/light_control-assembly_6.pdf)

Both look the same.

commit 8c3e5b98 (2022-12-16): [PCB_Print][Fixed] Images not showing in custom frames.

Closes #352