INTI-CMNB/KiBot #354: Creating a diff at the docker image. Kibot 1.3.0.

raulgotor, 2022-12-23:
**What do you want to achieve?**
First of all, I put this as a question since I think I'm missing something and is not a bug, but something related to my configuration.

I'd like to create a diff between current schematic and previous one on the repo. I'm running on a docker image (setsoft/kicad_auto:ki6) with Kibot 1.3.0

The issue i'm facing is that the script gets stuck when checking out the repo back to the original branch after being at the previous commit.

Below I attach the corresponding diff section of the yaml file, and the trace output when I run `kibot -v`. Script gets stuck at the last line, and I have to kill the process, resulting in no diff generation

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
- name: basic_diff_sch
     comment: Schematic diff between the last two changes
     type: diff
     dir: diff
     options:
      old: KIBOT_LAST-1
      old_type: git
      add_link_id: true
      pcb: false
      force_checkout: true
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`)
- In a GitHub action? Which tag are you using? (Ex `v1`, `v1_k6`, `v2`, `v2_d`, `v2_dk6`, `v2_k6`)
- Using a docker image? (Ex: `kicad_auto:ki6`, `kicad_auto:latest`, `kicad_auto:dev`, `kicad_auto:dev_k6`, a custom one? can you provide the `Dockerfile`)
- Using a desktop system? Which OS, distro and version?

**Additional context**

```bash
- 'Schematic diff between the last two changes' (basic_diff_sch) [diff]
DEBUG:Output destination: /kikit/Generated_Data/diff (kibot - kiplot.py:343)
DEBUG:Starting tool check for KiCad PCB/SCH Diff (kibot - dep_downloader.py:770)
DEBUG:- Returning `/usr/bin/kicad-diff.py` (kibot - dep_downloader.py:777)
DEBUG:Starting tool check for Git (kibot - dep_downloader.py:770)
DEBUG:- Returning `/usr/bin/git` (kibot - dep_downloader.py:777)
DEBUG:Starting tool check for KiCad Automation tools (kibot - dep_downloader.py:770)
DEBUG:- Returning `/usr/bin/pcbnew_do` (kibot - dep_downloader.py:777)
DEBUG:Saving current changes (kibot - out_diff.py:266)
DEBUG:Executing: /usr/bin/git stash push -m KiBot_Changes_Entry (kibot - kiplot.py:148)
DEBUG:- Output from command: No local changes to save
 (kibot - kiplot.py:144)
DEBUG:Executing: /usr/bin/git submodule foreach 'git stash push -m KiBot_Changes_Entry' (kibot - kiplot.py:148)
DEBUG:Executing: /usr/bin/git rev-parse --abbrev-ref HEAD (kibot - kiplot.py:148)
DEBUG:- Output from command: feature/automated_diff
 (kibot - kiplot.py:144)
DEBUG:Current branch is feature/automated_diff (kibot - out_diff.py:276)
DEBUG:Finding KIBOT_LAST-1 (kibot - out_diff.py:191)
DEBUG:Executing: /usr/bin/git log --pretty=format:%H --skip=1 -n 1 -- /kikit/Pr17.kicad_sch (kibot - kiplot.py:148)
DEBUG:- Output from command: f2cfe849290d31717351e89aedda978890c43626 (kibot - kiplot.py:144)
DEBUG:- f2cfe849290d31717351e89aedda978890c43626 (kibot - out_diff.py:208)
DEBUG:Changing to f2cfe849290d31717351e89aedda978890c43626 (kibot - out_diff.py:280)
DEBUG:Executing: /usr/bin/git checkout --force --recurse-submodules f2cfe849290d31717351e89aedda978890c43626 (kibot - kiplot.py:148)
DEBUG:- Output from command: Note: switching to 'f2cfe849290d31717351e89aedda978890c43626'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at f2cfe84 small bom fixes, excluding holes from BoM
 (kibot - kiplot.py:144)
DEBUG:Executing: /usr/bin/git describe --exact-match --tags HEAD (kibot - kiplot.py:148)
DEBUG:Not at a tag point (kibot - out_diff.py:219)
DEBUG:Executing: /usr/bin/git rev-parse --abbrev-ref HEAD (kibot - kiplot.py:148)
DEBUG:- Output from command: HEAD
 (kibot - kiplot.py:144)
DEBUG:Executing: /usr/bin/git describe --tags --dirty (kibot - kiplot.py:148)
DEBUG:- Output from command: v1.1.1-5-gf2cfe84
 (kibot - kiplot.py:144)
DEBUG:Executing: /usr/bin/git rev-parse --short HEAD (kibot - kiplot.py:148)
DEBUG:- Output from command: f2cfe84
 (kibot - kiplot.py:144)
DEBUG:Loading sheet from /kikit/Pr17.kicad_sch (kibot - v6_sch.py:1780)
DEBUG:SCH title: `PAPI PCB` (kibot - v6_sch.py:1620)
DEBUG:SCH date: `2022-09-18` (kibot - v6_sch.py:1621)
DEBUG:SCH revision: `v2.0.0` (kibot - v6_sch.py:1622)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1623)
DEBUG:Loading sheet from /kikit/Pr17_ext_sensor.kicad_sch (kibot - v6_sch.py:1780)
DEBUG:SCH title: `PAPI PCB` (kibot - v6_sch.py:1620)
DEBUG:SCH date: `2022-09-18` (kibot - v6_sch.py:1621)
DEBUG:SCH revision: `v2.0.0` (kibot - v6_sch.py:1622)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1623)
DEBUG:Hashing /kikit/Pr17.kicad_sch (kibot - out_diff.py:98)
DEBUG:Hashing /kikit/Pr17_ext_sensor.kicad_sch (kibot - out_diff.py:98)
DEBUG:Executing: /usr/bin/kicad-diff.py -v --only_cache --old_file_hash schc8fef050b579ff7afa164d34aca5132411ac8b20 --cache_dir /tmp/tmp_7jo063t /kikit/Pr17.kicad_sch /kikit/Pr17.kicad_sch (kibot - kiplot.py:148)
DEBUG:- Output from command: INFO:kicad-diff.py:Plotting the schematic
INFO:kicad-diff.py:/kikit/Pr17.kicad_sch SHA1 is schc8fef050b579ff7afa164d34aca5132411ac8b20
 (kibot - kiplot.py:144)
DEBUG:Restoring point feature/automated_diff (kibot - out_diff.py:178)
DEBUG:Executing: /usr/bin/git checkout --force --recurse-submodules feature/automated_diff (kibot - kiplot.py:148)
```

raulgotor, 2022-12-23:
Nevermind, deleting the repo and cloning again fixed the issue, so most likely it was related to repo configuration.

closed 2022-12-23 by raulgotor

set-soft, 2022-12-26:
Hi @raulgotor !
Also note that current KiBot is 1.5.1, which includes changes and fixes to the Diff output. Consult the [changelog](https://github.com/INTI-CMNB/KiBot/blob/master/CHANGELOG.md).
According to the log the problem you experimented was when KiBot tried to update submodules