INTI-CMNB/KiBot #357: [BUG] copy_files output expects KiCad6 party_3rd_dir to be set

andersruneson, 2022-12-29:
**Describe the bug**
I'm trying to run the copy_files output but get a traceback when running KiBot kicad5_auto:latest.

```
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.5.1', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 403, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 497, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 488, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 415, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_copy_files.py", line 269, in run
    self.options.run(output_dir)
  File "/usr/lib/python3/dist-packages/kibot/out_copy_files.py", line 220, in run
    files = self.get_files()
  File "/usr/lib/python3/dist-packages/kibot/out_copy_files.py", line 161, in get_files
    os.path.normpath(os.path.join(GS.pcb_dir, KiConf.party_3rd_dir)),
  File "/usr/lib/python3.9/posixpath.py", line 90, in join
    genericpath._check_arg_types('join', a, *p)
  File "/usr/lib/python3.9/genericpath.py", line 152, in _check_arg_types
    raise TypeError(f'{funcname}() argument must be str, bytes, or '
TypeError: join() argument must be str, bytes, or os.PathLike object, not 'NoneType'
```

I believe the issue might be out_copy_files.py, row 160:
```
self.rel_dirs = [os.path.normpath(os.path.join(GS.pcb_dir, KiConf.models_3d_dir)),
                 os.path.normpath(os.path.join(GS.pcb_dir, KiConf.party_3rd_dir)),
                 GS.pcb_dir]
```
Here we try to join KiConf.party_3rd_dir (which is None since I do not run KiCad6 yet) with pcb_dir.
I tried to add 3RD_PARTY to my kicad_common but that made no change, I guess kicad/config.py discards that variable if not running KiCad6?
`KiConf._solve_var('3RD_PARTY', 'party_3rd_dir', '3rd party', KiConf.guess_3rd_dir, only_k6=True, no_dir=True)`

**To Reproduce**

Below is the affected outputs I'm trying to run.
```
outputs:
- name: eurocircuits_bom
  comment: BOM for EuroCircuits
  type: copy_files
  run_by_default: true
  dir: 'Manufacturers/EuroCircuits'
  options:
    files:
    - source: 'group_row_bom_xlsx'
      source_type: 'output'
      dest: '/'
```

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image: kicad5_auto:latest

andersruneson, 2022-12-29:
Changing 
```
self.rel_dirs = [os.path.normpath(os.path.join(GS.pcb_dir, KiConf.models_3d_dir)),
                 os.path.normpath(os.path.join(GS.pcb_dir, KiConf.party_3rd_dir)),
                 GS.pcb_dir]
```
to
```
self.rel_dirs = [os.path.normpath(os.path.join(GS.pcb_dir, KiConf.models_3d_dir)),
                 os.path.normpath(os.path.join(GS.pcb_dir, '/')),
                 GS.pcb_dir]
```
works as a work around.

set-soft, 2023-01-02:
Hi @andersruneson !
Thanks for reporting! Now the code checks KiCad paths before trying to add them.
The fix is in the `dev` branch (16cb739ae49a046de175b7eb65ce9ebbecf4f3e7)

commit 16cb739a (2023-01-02): [Copy_Files][Fixed] Problems on KiCad 5

- No 3rd party dir

Fixes #357
closed 2023-01-02 by set-soft