INTI-CMNB/KiBot #359: [HOW] WARNING:(W009) KiCad config without environment.vars section

mundodisco8, 2023-01-09:
**What do you want to achieve?**
I would like to know where to get more information about the meaning of some warnings and how to fix them.

**Do you have some PCB/Schematic to use as example?**
I attach a project. Running

```bash
kibot -e Snowflake.kicad_sch -b Snowflake.kicad_pcb
```

Triggers warning W009. The configuration file is `example_template.kibot.yaml`.

[Snowflake.zip](https://github.com/INTI-CMNB/KiBot/files/10372737/Snowflake.zip)

**Do you have some configuration file (.kibot.yaml) that you are using?**
It's the example template, with parts commented. I'm learning my way through and I haven't explored much yet. It's included in the zip file
```yaml
kibot:
  version: 1

global:
  # Acknowledge some of the errors thrown in the process
  filters:
    # S1 is the silkscreen with the text in the back (turn this...). It doesn't have a symbol associated in the schematic
    - filter: 'S1 in board, but not in schematic'
      number: 13
      regex: 'S1'
    # DRC error on courtyard overlap is now a warning
    - filter: 'Courtyard overlap Battery/counter warning'
      number: 58
      regex: 'Footprint U2\n.+BT1'
    - filter: "Battery Internal connection"
      number: 58
      regex: 'BT1 on B.Cu\n.+BT1 on B.Cu'
    # Complaint about the resistor of the potentiometer being `0-50k`
    - filter: 'Pot value'
      number: 20
      regex: '0-50k'

preflight:
  # [dict] Annotates the PCB according to physical coordinates.
  # This preflight modifies the PCB and schematic, use it only in revision control environments.
  # Used to assign references according to footprint coordinates.
  # The project must be fully annotated first.
  # annotate_pcb:
  #   - top_main_axis: y
  #   - top_main_ascending: true
  #   - top_secondary_ascending: true
  #   - top_start: 1
  #   - bottom_main_axis: y
  #   - bottom_main_ascending: true
  #   - bottom_secondary_ascending: true
  #   - bottom_start: 101
  #   - use_position_of: 'footprint'
  #   - grid: 1.0
  # [boolean=false] Annotates all power components.
  # This preflight modifies the schematic, use it only in revision control environments.
  # Used to solve ERC problems when using filters that remove power reference numbers.
  annotate_power: true
  # [boolean=false] Zones are filled before doing any operation involving PCB layers.
  # The original PCB remains unchanged.
  check_zone_fills: true
  # [boolean=false] Option for `run_erc`. ERC warnings are considered errors.
  erc_warnings: false
  # [boolean=false] Fill all zones again and save the PCB.
  fill_zones: true
  # [list(dict)] A list of entries to filter out ERC/DRC messages.
  # Note that ignored errors will become KiBot warnings (i.e. `(W058) ...`).
  # To farther ignore these warnings use the `filters` option in the `global` section.
  # Acknowledge DRC errors
[... 363 lines cut ...]
                text: ''
                # [string=''] Text to add after the field content. Will be added only if the field isn't empty.
                # Any space to separate it should be added in the text.
                # Use \n for newline and \t for tab
                text_after: ''
                # [string=''] Text to add before the field content. Will be added only if the field isn't empty.
                # Any space to separate it should be added in the text.
                # Use \n for newline and \t for tab
                text_before: ''
            # [number=0] Used to group columns. The XLSX output uses it to collapse columns
            level: 0
            # [string=''] Name to display in the header. The field is used when empty
            name: 'Line'
        # [string='modern-blue'] Head style: modern-blue, modern-green, modern-red and classic
        style: 'modern-blue'
        # [string='KiBot Bill of Materials'] BoM title
        title: 'KiBot Bill of Materials'
```

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? Ubuntu:latest with Kicad 6.0.10 installed on it by hand.

**Additional context**
In this particular case, the warning seems to be about the BoM, but it would be great to have a list of warnings to figure out how to deal with them

set-soft, 2023-01-10:
Hi @mundodisco8 !

For KiCad 6.0.10 on Ubuntu your config file should be `~/.config/kicad/6.0/kicad_common.json`
If KiBot can find your KiCad libs this warning isn't important.
I think KiCad starts fresh configs without environment variables, and they are added when you edit them.
Here is how the `environment` section in my config (imported from KiCad 5) looks:

```json
  "environment": {
    "show_warning_dialog": true,
    "vars": {
      "KICAD_TEMPLATE_DIR": "/usr/share/kicad/template",
      "KICAD_USER_TEMPLATE_DIR": "/home/salvador/kicad/template",
      "KIGITHUB": "https://github.com/KiCad",
      "KISYS3DMOD": "/usr/share/kicad/3dmodels/",
      "KISYSMOD": "/usr/share/kicad/footprints"
    }
  },
```

The warning means that KiBot is guessing the KiCad environment, which could be wrong. Is usually right if KiCad was installed from the .deb, but if you compile KiCad to be installed in `/usr/local/` (or other place) KiBot could fail to find the values.

closed 2023-01-10 by set-soft

mundodisco8, 2023-01-11:
brilliant, thanks for the explanation!