INTI-CMNB/KiBot #360: [BUG] Error when using the annotate_pcb preflight

mundodisco8, 2023-01-09:
**Describe the bug**
When I use the annotate_pcb preflight, with any option enabled, I get an error

**To Reproduce**
This is the command I run, I attach a zip file with the project and the kibot config on it.
[Snowflake.zip](https://github.com/INTI-CMNB/KiBot/files/10373703/Snowflake.zip)

```bash
$ kibot -e hardware/Snowflake.kicad_sch -b hardware/Snowflake.kicad_pcb
Using config file: example_template.kibot.yaml
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.5.1', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 389, in main
    outputs = cr.read(cf_file)
  File "/usr/lib/python3/dist-packages/kibot/config_reader.py", line 560, in read
    BasePreFlight.add_preflights(self._parse_preflights(v1))
  File "/usr/lib/python3/dist-packages/kibot/config_reader.py", line 222, in _parse_preflights
    o_pre = BasePreFlight.get_class_for(k)(k, v)
  File "/usr/lib/python3/dist-packages/kibot/pre_annotate_pcb.py", line 129, in __init__
    o.config(self)
  File "/usr/lib/python3/dist-packages/kibot/optionable.py", line 225, in config
    self._perform_config_mapping()
  File "/usr/lib/python3/dist-packages/kibot/optionable.py", line 131, in _perform_config_mapping
    for k, v in self._tree.items():
AttributeError: 'list' object has no attribute 'items'
```
If I comment out the annotate_pcb options:
```yaml
preflight:
  # [dict] Annotates the PCB according to physical coordinates.
  # This preflight modifies the PCB and schematic, use it only in revision control environments.
  # Used to assign references according to footprint coordinates.
  # The project must be fully annotated first.
  annotate_pcb:
    # - top_main_axis: y
    # - top_main_ascending: true
    # - top_secondary_ascending: true
    # - top_start: 1
    # - bottom_main_axis: y
    # - bottom_main_ascending: true
    # - bottom_secondary_ascending: true
    # - bottom_start: 101
    # - use_position_of: 'footprint'
    # - grid: 1.0
```

The same command seems to work

```bash
$ kibot -e hardware/Snowflake.kicad_sch -b hardware/Snowflake.kicad_pcb
Using config file: example_template.kibot.yaml
WARNING:(W040) No annotation problems, skipping power annotation (kibot.pre_annotate_power - pre_annotate_power.py:69)
[...]
```

**Expected behavior**
The schematic gets annotated, or at least I get a more descriptive error.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? Using Ubuntu:latest, with KiCad and KiBot installed by hand, no dependencies missing 

```bash
root@6087c6774112:~/Snowflake# kibot-check
KiBot installation checker

Core:
Linux: 5.10.16.3 (Linux 6087c6774112 5.10.16.3-microsoft-standard-WSL2 #1 SMP Fri Apr 2 22:23:49 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux)
Python: 3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0]
KiCad: 6.0.10-86aedd382b~118~ubuntu22.04.1
Kibot: 1.5.1

Modules:
Colorama: 0.4.4
LXML: 4.8.0
PyYAML: 5.4.1
QRCodeGen: Ok
Requests: 2.25.1
XLSXWriter: 3.0.2
Xvfbwrapper: Ok
markdown2: 2.4.1
mistune: 0.8.4
numpy: 1.21.5

Tools:
Ghostscript: 9.55.0
Git: 2.34.1 (git version 2.34.1)
ImageMagick: 6.9.11.60 (Version: ImageMagick 6.9.11-60 Q16 x86_64 2021-01-25 https://imagemagick.org)
Interactive HTML BoM: 2.5.0.2 (v2.5.0-2)
KiBoM: 1.8.0 (KiBOM Version: 1.8.0)
KiCad Automation tools (kiauto): 2.1.0 (pcbnew_do 2.1.0 - Copyright 2018-2022, INTI/Productize SPRL - License: Apache)
KiCad PCB/SCH Diff (kidiff): 2.4.3 (kicad-diff.py 2.4.3 - Copyright 2020-2022, INTI/Salvador E. Tropea - License:)
KiCost: 1.1.15 (KiCost v1.1.15)
KiKit: 1.2.0.1 (kikit, version 1.2.0-1)
OpenSCAD: 2021.1.0 (OpenSCAD version 2021.01)
Pandoc: 2.9.2.1 (pandoc 2.9.2.1)
RAR: 5.50.0 (RAR 5.50   Copyright (c) 1993-2017 Alexander Roshal   11 Aug 2017)
RSVG tools: 2.52.5 (rsvg-convert version 2.52.5)
Xvfb: Ok (xvfb-run)

Color reference: ok, optional for an output, optional for general use, mandatory for an output, mandatory for general use

Did this help? Please consider commenting it on https://github.com/INTI-CMNB/KiBot/discussions/categories/kibot-check
```

set-soft, 2023-01-10:
Hi @mundodisco8 !

Now KiBot prints an error when you use wrong data type for the pre-flight options.
You must use a dict there.
The docs didn't show which are the options and the example was wrong.
Now should be fixed

mundodisco8, 2023-01-11:
Ah! Right, even the comment on top of that mentioned it, sorry for not noticing!

set-soft, 2023-01-11:
Hi @mundodisco8 !
Yes, but is also true that the example was missleading. This is why I marked it as *also a bug*.

commit db6fa676 (2023-01-10): [DOCs][Fixed] annotate_pcb docs and example

- Missing options in docs
- Wrong example

Fixes #360