INTI-CMNB/KiBot #362: [HOW]

nerdyscout, 2023-01-10:
[image]

i might be confused a bit currently... using Type: `svg_sch_print`, which i guess belongs to the dialog of the screenshot. correct?

anyway... I am looking for an option to generate a schematic as svg with colored background. what am I missing?

set-soft, 2023-01-11:
Hi @nerdyscout !
Yes, `svg_sch_print` (which was adapted by you from `pdf_sch_print`) is this dialog.
It wasn't possible to control it.
Now it should be, I tried:

```yaml
  - name: 'print_sch_svg'
    comment: "Print schematic (SVG)"
    type: svg_sch_print
    dir: Schematic
    options:
      color_theme: User
      background_color: true
```
And got:
[image]

Of course I have a color scheme called *User* and it defines the background as an ugly yellow ;-)
If you'll use CI/CD you'll need to copy the color file to `~/.config/kicad/6.0/colors/` (i.e. `~/.config/kicad/6.0/colors/user.json`)

Let me know if it works for you.

set-soft, 2023-02-06:
Hi @nerdyscout !
Did you have time to test it?

nerdyscout, 2023-02-06:
I just did past weekend... but I haven't figured out how to do this:
`If you'll use CI/CD you'll need to copy the color file to ~/.config/kicad/6.0/colors/ (i.e. ~/.config/kicad/6.0/colors/user.json)`
while simply using 
```
  docs:
    runs-on: ubuntu-latest
    needs: [ERC]
    container:
      image: setsoft/kicad_auto:ki6
    steps:
      - uses: actions/checkout@v3
      - name: run kibot
        run: | 
          kibot -c kibot.yml -e ${{env.schema}} -b ${{env.board}} -d ${{env.dir}} -s run_erc,run_drc -v \
            ibom.html octopart.csv schematic.pdf schematic.svg board_top.pdf board_bottom.pdf
```
the user.json obviously is inside the user dir, not the respository. without building my own docker image I currently don't see how to put this file into.

set-soft, 2023-02-06:
Hi @nerdyscout !
How about including *user.json* in your repo and then doing `cp user.json  ~/.config/kicad/6.0/colors/`?

nerdyscout, 2023-02-18:
srry for delaying this.

closed 2023-02-18 by nerdyscout

set-soft, 2023-04-05:
> the user.json obviously is inside the user dir, not the respository. without building my own docker image I currently don't see how to put this file into.

The 6feddd06a54ccf028cbd90c7b34cf993525f22b3 patch solves this problem. You just need to include your colors in the *kibot_resources/colors* folder of your repo. KiBot will copy these files to the KiCad config dir. The global option `resources_dir` can be used to specify other directory.