INTI-CMNB/KiBot #364: [HOW]

N0cynym, 2023-01-12:
**What do you want to achieve?**
I want to know how I can change the size of my exported media content to full report.

If I export to full txt report the content looks like:

```
# PCB Layers

![Print F.Paste](Layers/1-MyProject_Top_Paste_Layer.pdf){ width=16.5cm height=11.7cm }

![Print F.Silk](Layers/2-MyProject_Top_Silkscreen_Layer.pdf){ width=16.5cm height=11.7cm }

![Print F.Mask](Layers/3-MyProject_Top_Mask_Layer.pdf){ width=16.5cm height=11.7cm }

![Print F.Cu](Layers/4-MyProject_Top_Layer.pdf){ width=16.5cm height=11.7cm }

![Print B.Cu](Layers/5-MyProject_Bottom_Layer.pdf){ width=16.5cm height=11.7cm }

![Print B.Mask](Layers/6-MyProject_Bottom_Mask_Layer.pdf){ width=16.5cm height=11.7cm }

![Print B.Silk](Layers/7-MyProject_Bottom_Silkscreen_Layer.pdf){ width=16.5cm height=11.7cm }

![Print B.Paste](Layers/8-MyProject_Bottom_Paste_Layer.pdf){ width=16.5cm height=11.7cm }

```

**Do you have some PCB/Schematic to use as example?**
I can prepare.

**Do you have some configuration file (.kibot.yaml) that you are using?**

```yaml
- name: 'full_report'
    comment: "Full design report"
    type: report

  - name: 'print_sch'
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: .
    options:
      output: MyProject.pdf

  - name: 'print_drawings'
    comment: "Print Dimension"
    type: pdf_pcb_print
    dir: Layers
    options:
      drill_marks: none
      output_name: 0-MyProject_Drawings.pdf
    layers:
      - layer: User.Drawings
      - layer: Edge.Cuts
  - name: 'print_f_paste'
    comment: "Print F.Paste"
    type: pdf_pcb_print
    dir: Layers
    options:
      drill_marks: none
      output_name: 1-MyProject_Top_Paste_Layer.pdf
    layers:
      - layer: F.Paste
      - layer: Edge.Cuts
      
  - name: 'print_f_silk'
    comment: "Print F.Silk"
    type: pdf_pcb_print
    dir: Layers
    options:
      drill_marks: none
      output_name: 2-MyProject_Top_Silkscreen_Layer.pdf
    layers:
      - layer: F.SilkS
      - layer: Edge.Cuts  

  - name: 'print_f_mask'
    comment: "Print F.Mask"
    type: pdf_pcb_print
    dir: Layers
    options:
      drill_marks: none
      output_name: 3-MyProject_Top_Mask_Layer.pdf
    layers:
      - layer: F.Mask
      - layer: Edge.Cuts
      
  - name: 'print_f_cu'
    comment: "Print F.Cu"
    type: pdf_pcb_print
    dir: Layers
    options:
      output_name: 4-MyProject_Top_Layer.pdf
    layers:
      - layer: F.Cu
      - layer: Edge.Cuts

  - name: 'print_b_cu'
    comment: "Print B.Cu"
    type: pdf_pcb_print
    dir: Layers
    options:
      output_name: 5-MyProject_Bottom_Layer.pdf
      mirror: true
    layers:
      - layer: B.Cu
      - layer: Edge.Cuts

  - name: 'print_b_mask'
    comment: "Print B.Mask"
    type: pdf_pcb_print
    dir: Layers
    options:
      drill_marks: none
      output_name: 6-MyProject_Bottom_Mask_Layer.pdf
      mirror: true
    layers:
      - layer: B.Mask
      - layer: Edge.Cuts

  - name: 'print_b_silk'
    comment: "Print B.Silk"
    type: pdf_pcb_print
    dir: Layers
    options:
      drill_marks: none
      output_name: 7-MyProject_Bottom_Silkscreen_Layer.pdf
      mirror: true
    layers:
      - layer: B.SilkS
      - layer: Edge.Cuts      

  - name: 'print_b_paste'
    comment: "Print B.Paste"
    type: pdf_pcb_print
    dir: Layers
    options:
      drill_marks: none
      output_name: 8-MyProject_Bottom_Paste_Layer.pdf
      mirror: true
    layers:
      - layer: B.Paste
      - layer: Edge.Cuts
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a GitLab CI/CD
- Using docker image `setsoft/kicad_auto_test:ki6`

**Additional context**
I want to create a pdf with the layer pdf contet in full scale and not scaled to `{ width=16.5cm height=11.7cm }`.
How can I edit this value?

set-soft, 2023-01-12:
Hi @N0cynym !
You need to create your own report template.
The *full* report is generated by [report_full.txt](https://github.com/INTI-CMNB/KiBot/blob/master/kibot/resources/report_templates/report_full.txt)

You'll see the `{ width=16.5cm height=11.7cm }` is explicitly specified there.

Once you have a modified template just use the `template` option of the `report` output to specify the file you created.
Consult the PanDoc documentation to learn more about the Markdown extensions supported by  [PanDoc](https://pandoc.org/MANUAL.html). If I recall ok the image size is something specific to PanDoc, not really standard Markdown.

closed 2023-01-12 by set-soft

N0cynym, 2023-01-12:
@set-soft thank you very much.

One more question: is it possible to add `xelatex `?

```
$ pandoc  MyProject.txt --from markdown --pdf-engine=xelatex -o MyProject.pdf
xelatex not found. Please select a different --pdf-engine or install xelatex -- see also /usr/share/doc/pandoc/README.Debian
```