INTI-CMNB/KiBot #366: [BUG]

mundodisco8, 2023-01-13:
Fill as much as you can, remove the rest

**Describe the bug**
When trying to substitute the commit hash in my files with the `set_text_variables:` preflight command, git returns errorcode 128. 

```bash
mundodisco8@LAPTOP:/mnt/c/Users/joel.santos/Desktop/test/BGMP220P-Breakout$ kibot -c ArtifactCreation.kibot.yaml -b Hardware/BGM220P_Breakout.kicad_pcb -e Hardware/BGM220P_Breakout.kicad_sch
ERROR:Failed to execute:
git log -1 --format="%h" $KIBOT_PCB_NAME
return code 128 (kibot - pre_set_text_variables.py:127)
```

This only happens under WSL AND if the files I'm working on are mounted from my windows partition (path `/mnt/c/Users...`). If I clone the repo "inside" WSL (in my home folder, for example), then it works without issue.

```bash
mundodisco8@LAPTOP:/mnt/c/Users/joel.santos/Desktop/test/BGMP220P-Breakout$ kibot -c ArtifactCreation.kibot.yaml -b Hardware/BGM220P_Breakout.kicad_pcb -e Hardware/BGM220P_Breakout.kicad_sch
[NO ERRORS]
```

Running the git command manually in the Windows path (`/mnt/c/...`) works

```bash
mundodisco8@LAPTOP:/mnt/c/Users/joel.santos/Desktop/test/BGMP220P-Breakout$ git log -1 --format="%h" $KIBOT_PCB_NAME
128aec2
```

**To Reproduce**
Running Ubuntu on WSL2, with KiCad 6.0.10 and KiBot installed, no dependencies missing from `kibot-check`. I tried with multiple projects and all have the issue, so I don't expect it to be a specific project issue.

**Workaround** 
Run KiBot on a project that is located "inside" the WSL filesystem.

**Expected behavior**
The command returns the hash and doesn't error out.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
Where are you running KiBot:

WSL2 on Windows 11, running Ubuntu 22.04, all up to date as of today. Installed KiCad and KiBot via apt-get.

**Additional context**
I understand that you might not "support" WSL, but I found it curious that it works when the files are inside the WSL volume, and not mounted from the NTFS volume. As the git command works when called from the command line, I figured out that it might have to be with the way you are calling it. I checked `pre_set_text_variables.py`  but I couldn't or know how to debug it. I have a workaround, so it's not urgent.

set-soft, 2023-01-13:
Hi @mundodisco8 !
Please run KiBot with debug info (i.e. -vvvv) and take a look at what exactly happened during the git call.

Also: 
- Is git installed in the Linux side or in the Windows side? 
- Microsoft recommends to run Linux side programs using data in the Linux side. They recommend copying the data to the Linux side. I think this is mainly for performance.

mundodisco8, 2023-01-16:
I attach the log, hope it seds a light, didn't know there were 4 `v`!

About your comments, it's using WSL's git (it's version 2.34.1, while on Windows I have git for Windows at 2.39. And you are right about using files "contained" in the WSL volume, as it really impacts performance, but in this case, I would be interested in checking the ones outside, so I can do a check before committing to my repo, for example. What I don't get is why thescript works when the same project is contained in the volume, and it doesn't when it's in my "Windows Filesystem". Also, running the command in the command line in the "Windows FS" side seems to work.

```
mundodisco8@LAPTOP-SANTJ:/mnt/c/Users/joel.santos/OneDrive/Desktop/test/BGMP220P-Breakout$ kibot -c ArtifactCreation.kibot.yaml -b Hardware/BGM220P_Breakout.kicad_pcb -e Hardware/BGM220P_Breakout.kicad_sch -vvvv
DEBUG:KiBot 1.5.1 verbose level: 4 (kibot - __main__.py:306)
DEBUG:Detected KiCad v6.0.10 (6.0.10-86aedd382b~118~ubuntu22.04.1 6000010) (kibot - __main__.py:196)
DEBUG:KiCad config path /home/mundodisco8/.config/kicad/6.0 (kibot - __main__.py:246)
DEBUG:Command line arguments:
['/usr/bin/kibot', '-c', 'ArtifactCreation.kibot.yaml', '-b', 'Hardware/BGM220P_Breakout.kicad_pcb', '-e', 'Hardware/BGM220P_Breakout.kicad_sch', '-vvvv'] (kibot - __main__.py:285)
DEBUG:Command line parsed:
{'--board-file': 'Hardware/BGM220P_Breakout.kicad_pcb',
 '--cli-order': False,
 '--copy-and-expand': False,
 '--copy-options': False,
 '--define': [],
 '--dont-stop': False,
 '--dry': False,
 '--example': False,
 '--global-redef': [],
 '--help': False,
 '--help-dependencies': False,
 '--help-filters': False,
 '--help-global-options': False,
 '--help-list-outputs': False,
 '--help-output': None,
 '--help-outputs': False,
 '--help-preflights': False,
 '--help-variants': False,
 '--invert-sel': False,
 '--json': False,
 '--list': False,
 '--makefile': None,
 '--markdown': False,
 '--no-auto-download': False,
 '--no-priority': False,
 '--no-warn': None,
 '--out-dir': '.',
 '--plot-config': 'ArtifactCreation.kibot.yaml',
 '--quick-start': False,
 '--quiet': False,
 '--schematic': 'Hardware/BGM220P_Breakout.kicad_sch',
 '--skip-pre': None,
 '--start': '.',
 '--type': [],
 '--verbose': 4,
 '--version': False,
 'TARGET': []} (kibot - __main__.py:286)
DEBUG:- Processing dependencies for `global` (kibot - dep_downloader.py:985)
DEBUG: - Data: {'Dependencies': [{'name': 'Colorama', 'python_module': True, 'role': 'Get color messages in a portable way', 'debian': 'python3-colorama', 'arch': 'python-colorama'}, {'name': 'Requests', 'python_module': True, 'role': 'mandatory', 'debian': 'python3-requests', 'arch': 'python-requests'}, {'name': 'PyYAML', 'python_module': True, 'debian': 'python3-yaml', 'arch': 'python-yaml', 'module_name': 'yaml', 'role': 'mandatory'}, {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'pcbnew_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto'}, {'name': 'Git', 'url': 'https://git-scm.com/', 'downloader': 'git', 'debian': 'git', 'arch': 'git'}, {'name': 'RSVG tools', 'url': 'https://gitlab.gnome.org/GNOME/librsvg', 'debian': 'librsvg2-bin', 'arch': 'librsvg', 'command': 'rsvg-convert', 'downloader': 'rsvg', 'id': 'RSVG', 'tests': [{'command': ['convert', '-list', 'font'], 'search': 'Helvetica', 'error': 'Missing Helvetica font, try installing Ghostscript fonts'}]}, {'name': 'Ghostscript', 'url': 'https://www.ghostscript.com/', 'url_down': 'https://github.com/ArtifexSoftware/ghostpdl-downloads/releases', 'debian': 'ghostscript', 'arch': 'ghostscript', 'command': 'gs', 'downloader': 'gs'}, {'name': 'ImageMagick', 'url': 'https://imagemagick.org/', 'url_down': 'https://imagemagick.org/script/download.php', 'command': 'convert', 'downloader': 'convert', 'debian': 'imagemagick', 'arch': 'imagemagick', 'extra_arch': ['gsfonts']}, {'name': 'KiCost', 'github': 'hildogjr/KiCost', 'pypi': 'KiCost', 'downloader': 'pytool'}, {'name': 'LXML', 'python_module': True, 'debian': 'python3-lxml', 'arch': 'python-lxml', 'downloader': 'python'}, {'name': 'KiKit', 'github': 'yaqwsx/KiKit', 'pypi': 'KiKit', 'downloader': 'pytool'}, {'name': 'Xvfbwrapper', 'python_module': True, 'debian': 'python3-xvfbwrapper', 'arch': 'python-xvfbwrapper', 'downloader': 'python'}, {'name': 'Xvfb', 'url': 'https://www.x.org', 'command': 'xvfb-run', 'debian': 'xvfb', 'arch': 'xorg-server-xvfb', 'no_cmd_line_version': True}]} (kibot - dep_downloader.py:986)
DEBUG: - Registering dep {'name': 'Colorama', 'python_module': True, 'role': 'Get color messages in a portable way', 'debian': 'python3-colorama', 'arch': 'python-colorama'} (kibot - dep_downloader.py:997)
DEBUG: - Registering dep {'name': 'Requests', 'python_module': True, 'role': 'mandatory', 'debian': 'python3-requests', 'arch': 'python-requests'} (kibot - dep_downloader.py:997)
DEBUG: - Registering dep {'name': 'PyYAML', 'python_module': True, 'debian': 'python3-yaml', 'arch': 'python-yaml', 'module_name': 'yaml', 'role': 'mandatory'} (kibot - dep_downloader.py:997)
DEBUG: - Registering base dep {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'pcbnew_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto'} (kibot - dep_downloader.py:1000)
DEBUG: - Registering base dep {'name': 'Git', 'url': 'https://git-scm.com/', 'downloader': 'git', 'debian': 'git', 'arch': 'git'} (kibot - dep_downloader.py:1000)
[... cut ...]
DEBUG:- Checking version for `/usr/bin/git` (kibot - dep_downloader.py:594)
DEBUG:- No particular version needed (kibot - dep_downloader.py:608)
DEBUG:- Running ['/usr/bin/git', '--version'] (kibot - dep_downloader.py:560)
DEBUG:- Looking for version in `git version 2.34.1` (kibot - dep_downloader.py:586)
DEBUG:- Found version (2, 34, 1, 0) (kibot - dep_downloader.py:621)
DEBUG:- Returning `/usr/bin/git` (kibot - dep_downloader.py:806)
ERROR:Failed to execute:
git log -1 --format="%h" $KIBOT_PCB_NAME
return code 128 (kibot - pre_set_text_variables.py:127)
```

set-soft, 2023-01-16:
Hi @mundodisco8 !
This is most probably related to the shell. Similar to what happends [here](https://github.com/INTI-CMNB/KiBot/discussions/325#discussioncomment-4475159).
The current `dev` code adds a check for Bash.
I also added more debug information.
Can you try using the `dev` code?
You can just do a git clone of the repo and run the script from the repo, no need to install.

mundodisco8, 2023-01-17:
Hey, thanks for your support, I appreciate it.

I did what you asked, and this is my log (I'm adding only the set_text_variables preflight log, and attach the rest as a file) :

```
DEBUG:Preflight apply run_drc (kibot.pre_base - pre_base.py:80)
DEBUG:Preflight apply run_erc (kibot.pre_base - pre_base.py:80)
DEBUG:Preflight apply set_text_variables (kibot.pre_base - pre_base.py:80)
DEBUG:- Current variables: {'GIT_HASH': 'COMMIT_HASH', 'TODAY': 'DATE'} (kibot - pre_set_text_variables.py:114)
DEBUG:Starting tool check for Git (kibot - dep_downloader.py:809)
DEBUG:- Checking binary tool Git (kibot - dep_downloader.py:703)
DEBUG:- Looking for tool `git` at system level (kibot - dep_downloader.py:637)
DEBUG:- Checking version for `/usr/bin/git` (kibot - dep_downloader.py:604)
DEBUG:- No particular version needed (kibot - dep_downloader.py:618)
DEBUG:- Running ['/usr/bin/git', '--version'] (kibot - dep_downloader.py:570)
DEBUG:- Looking for version in `git version 2.34.1` (kibot - dep_downloader.py:596)
DEBUG:- Found version (2, 34, 1, 0) (kibot - dep_downloader.py:631)
DEBUG:- Returning `/usr/bin/git` (kibot - dep_downloader.py:816)
DEBUG:Starting tool check for Bash (kibot - dep_downloader.py:809)
DEBUG:- Checking binary tool Bash (kibot - dep_downloader.py:703)
DEBUG:- Looking for tool `bash` at system level (kibot - dep_downloader.py:637)
DEBUG:- Checking version for `/usr/bin/bash` (kibot - dep_downloader.py:604)
DEBUG:- No particular version needed (kibot - dep_downloader.py:618)
DEBUG:- Running ['/usr/bin/bash', '--version'] (kibot - dep_downloader.py:570)
DEBUG:- Looking for version in `GNU bash, version 5.1.16(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.` (kibot - dep_downloader.py:596)
DEBUG:- Found version (5, 1, 16, 0) (kibot - dep_downloader.py:631)
DEBUG:- Returning `/usr/bin/bash` (kibot - dep_downloader.py:816)
DEBUG:Executing: / u s r / b i n / g i t ' ' l o g ' ' - 1 ' ' - - f o r m a t = '"' % h '"' ' ' '$' K I B O T _ P C B _ N A M E (kibot - pre_set_text_variables.py:131)
ERROR:Failed to execute:
git log -1 --format="%h" $KIBOT_PCB_NAME
return code 128 (kibot - pre_set_text_variables.py:134)
ERROR:stdout:

stderr:
fatal: option '-' must come before non-option arguments
 (kibot - pre_set_text_variables.py:135)
```

Now, I tried to play a bit with the script. I started to hack to try to get the minimal code that would call the command:

```python
from subprocess import run, PIPE
import re
import shlex

git_command = None
bash_command = None

def check_tool_dep(context, dep, fatal=False):
    cmd, ver = check_tool_dep_get_ver(context, dep, fatal)
    return cmd

re_git = re.compile(r'([^a-zA-Z_]|^)(git) ')

text = ""
command = 'git log -1 --format="%h" $KIBOT_PCB_NAME'
if not text and command:
    if re_git.search(command):
        # git_command = self.ensure_tool('git')
        git_command = "/usr/bin/git"
        command = re_git.sub(r'\1'+git_command.replace('\\', r'\\')+' ', command)
    if not bash_command:
        # bash_command = self.ensure_tool('Bash')
        bash_command = "/usr/bin/bash"
    cmd = [bash_command, '-c', command]
    print(f"JOEL: old command: {command}")
    print('Executing: '+shlex.join(command))
    result = run(cmd, stdout=PIPE, stderr=PIPE, universal_newlines=True)
    if result.returncode:
        print('Failed to execute:\n{}\nreturn code {}'.format(command, result.returncode))
        print('stdout:\n{}\nstderr:\n{}'.format(result.stdout, result.stderr))
        sys.exit(FAILED_EXECUTE)
    if not result.stdout:
        print(W_EMPTREP+"Empty value from `{}`".format(command))
    text = result.stdout.strip()
    print(text)
```

If I execute this, it works, I get the hash. Weird... Anyway, hope the stderr helps, but I can't figure out why it's being returned, when calling the command in the command line works, and calling it from a script works...

mundodisco8, 2023-01-17:
Played a bit more, and I think I have the solution. If I hijack the command and cut it to `git log -1 --format=%h`, it executes successfully, so it's the "$KIBOT_PCB_NAME" that causes the error in the script.

Now, I checked my path and it contains a folder with spaces on its name!!

I tested this again by moving it to a path without spaces and it works, and then renaming the folder to add a space and it fails again. 

The reason why running with files contained on the WSL instance worked is because it didn't have spaces. Likewise, running the command in the command line worked because `$KIBOT_PCB_NAME` was empty.

set-soft, 2023-01-17:
Hi @mundodisco8 !
Thanks for testing it.
I don't fully understand why the debug shows

```
DEBUG:Executing: / u s r / b i n / g i t ' ' l o g ' ' - 1 ' ' - - f o r m a t = '"' % h '"' ' ' '$' K I B O T _ P C B _ N A M E (kibot - pre_set_text_variables.py:131)
```

But this seems to be harmless anyways.
Ok, the solution should be as simple as double quoting the variable: "$KIBOT_PCB_NAME"
I added it to the examples and also a note about it.

closed 2023-01-17 by set-soft

mundodisco8, 2023-01-18:
If I can be picky, I would like to suggest a change in the logging message. I'm unit testing your module with the weirdest file names 😁 ...

So, I copied the settings to another project, run it and got an `WARNING:(W073) Empty value from `git log -1 --format="%h" "$KIBOT_PCB_NAME"` (kibot - pre_set_text_variables.py:130)`. Quite puzzled, I run the command and it worked, it was a Git repo, all that stuff. After a bit of fiddling, I realised that i was calling kicad with a `-b Hardware/boardname`, but the real path was `hardware/Boardname` (lowercase H). As I'm running the script in WSL, casing matters, and when checking the git command against an inexistent file, it returned "empty" (not error or anything). Surprisingly, the rest of the script worked, and I got gerbers and everything, maybe because paths are handled by Python and it's not that fuzzy.

My point is, testing against all the possible inputs is going to be a nightmare, so maybe the log message, instead of 

```bash
Empty value from `git log -1 --format="%h" "$KIBOT_PCB_NAME"`
```

Should have been 

```bash
Empty value from `git log -1 --format="%h" "Hardware/Boardname"`
```

With the environment varaiable substituted. Maybe even add a "revise your board path". What do you think?

set-soft, 2023-01-18:
Hi @mundodisco8 !
Your case is very special, replacing the environment variable in this case is trivial, but trying to emulate the shell interpretation of the command isn't.
I wonder how this problem happens. Why Python can find the file but git can't.
I added a debug message that tries to show such a difference, cayou try it?
Use debug information (`-v`) and look for a message like this:

```
DEBUG:Using PCB: `kibot/tests/board_samples/kicad_6/3Rs.kicad_pcb` (kibot - kiplot.py:764)
DEBUG:Real PCB name: `kibot/tests/board_samples/kicad_6/3Rs.kicad_pcb` (kibot - kiplot.py:765)
```

If the names are different we can add a warning about it.

mundodisco8, 2023-01-31:
Sorry, I missed this. I removed the quotes from the variable name in the config file
```yaml
set_text_variables:
    - name: 'GIT_HASH'
      command: 'git log -1 --format="%h" $KIBOT_PCB_NAME'
      before: '<'
      after: '>'
    - name: 'TODAY'
      command: 'date +"%Y-%m-%d"'
``` 

This is what I ran

```bash
src/kibot -c /mnt/c/Users/joel.santos/Path\ With\ Spaces/Desktop/Calavera/ArtifactCreation.kibot.yaml -b /mnt/c/Users/joel.santos/Path\ With\ Spaces/Desktop/Calavera/hardware/Calavera.kicad_pcb -e /mnt/c/Users/joel.santos/Path\ With\ Spaces/Desktop/Calavera/hardware/Calavera.kicad_sch -vvvv

This is the output

```text
DEBUG:KiBot 1.5.2 verbose level: 4 (kibot - __main__.py:306)
DEBUG:Detected KiCad v6.0.10 (6.0.10-86aedd382b~118~ubuntu22.04.1 6000010) (kibot - __main__.py:196)
DEBUG:KiCad config path /home/mundodisco8/.config/kicad/6.0 (kibot - __main__.py:246)
DEBUG:Command line arguments:
['src/kibot', '-c', '/mnt/c/Users/joel.santos/Path With Spaces/Desktop/Calavera/ArtifactCreation.kibot.yaml', '-b', '/mnt/c/Users/joel.santos/Path With Spaces/Desktop/Calavera/hardware/Calavera.kicad_pcb', '-e', '/mnt/c/Users/joel.santos/Path With Spaces/Desktop/Calavera/hardware/Calavera.kicad_sch', '-vvvv'] (kibot - __main__.py:285)
DEBUG:Command line parsed:
{'--board-file': '/mnt/c/Users/joel.santos/Path With Spaces/Desktop/Calavera/hardware/Calavera.kicad_pcb',
 '--cli-order': False,
 '--copy-and-expand': False,
 '--copy-options': False,
 '--define': [],
 '--dont-stop': False,
 '--dry': False,
 '--example': False,
 '--global-redef': [],
 '--help': False,
 '--help-dependencies': False,
 '--help-filters': False,
 '--help-global-options': False,
 '--help-list-outputs': False,
 '--help-output': None,
 '--help-outputs': False,
 '--help-preflights': False,
 '--help-variants': False,
 '--invert-sel': False,
 '--json': False,
 '--list': False,
 '--makefile': None,
 '--markdown': False,
 '--no-auto-download': False,
 '--no-priority': False,
 '--no-warn': None,
 '--out-dir': '.',
 '--plot-config': '/mnt/c/Users/joel.santos/Path With Spaces/Desktop/Calavera/ArtifactCreation.kibot.yaml',
 '--quick-start': False,
 '--quiet': False,
 '--schematic': '/mnt/c/Users/joel.santos/Path With Spaces/Desktop/Calavera/hardware/Calavera.kicad_sch',
 '--skip-pre': None,
 '--start': '.',
 '--type': [],
 '--verbose': 4,
 '--version': False,
 'TARGET': []} (kibot - __main__.py:286)
DEBUG:- Processing dependencies for `global` (kibot - dep_downloader.py:995)
DEBUG: - Data: {'Dependencies': [{'name': 'Colorama', 'python_module': True, 'role': 'Get color messages in a portable way', 'debian': 'python3-colorama', 'arch': 'python-colorama'}, {'name': 'Requests', 'python_module': True, 'role': 'mandatory', 'debian': 'python3-requests', 'arch': 'python-requests'}, {'name': 'PyYAML', 'python_module': True, 'debian': 'python3-yaml', 'arch': 'python-yaml', 'module_name': 'yaml', 'role': 'mandatory'}, {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'pcbnew_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto'}, {'name': 'Git', 'url': 'https://git-scm.com/', 'downloader': 'git', 'debian': 'git', 'arch': 'git'}, {'name': 'RSVG tools', 'url': 'https://gitlab.gnome.org/GNOME/librsvg', 'debian': 'librsvg2-bin', 'arch': 'librsvg', 'command': 'rsvg-convert', 'downloader': 'rsvg', 'id': 'RSVG', 'tests': [{'command': ['convert', '-list', 'font'], 'search': 'Helvetica', 'error': 'Missing Helvetica font, try installing Ghostscript fonts'}]}, {'name': 'Ghostscript', 'url': 'https://www.ghostscript.com/', 'url_down': 'https://github.com/ArtifexSoftware/ghostpdl-downloads/releases', 'debian': 'ghostscript', 'arch': 'ghostscript', 'command': 'gs', 'downloader': 'gs'}, {'name': 'ImageMagick', 'url': 'https://imagemagick.org/', 'url_down': 'https://imagemagick.org/script/download.php', 'command': 'convert', 'downloader': 'convert', 'debian': 'imagemagick', 'arch': 'imagemagick', 'extra_arch': ['gsfonts']}, {'name': 'KiCost', 'github': 'hildogjr/KiCost', 'pypi': 'KiCost', 'downloader': 'pytool'}, {'name': 'LXML', 'python_module': True, 'debian': 'python3-lxml', 'arch': 'python-lxml', 'downloader': 'python'}, {'name': 'KiKit', 'github': 'yaqwsx/KiKit', 'pypi': 'KiKit', 'downloader': 'pytool'}, {'from': 'KiKit', 'role': 'Separate multiboard projects'}, {'name': 'Xvfbwrapper', 'python_module': True, 'debian': 'python3-xvfbwrapper', 'arch': 'python-xvfbwrapper', 'downloader': 'python'}, {'name': 'Xvfb', 'url': 'https://www.x.org', 'command': 'xvfb-run', 'debian': 'xvfb', 'arch': 'xorg-server-xvfb', 'no_cmd_line_version': True}, {'name': 'Bash', 'url': 'https://www.gnu.org/software/bash/', 'debian': 'bash', 'arch': 'bash'}, {'name': 'Blender', 'url': 'https://www.blender.org/', 'debian': 'blender', 'arch': 'blender'}]} (kibot - dep_downloader.py:996)
DEBUG: - Registering dep {'name': 'Colorama', 'python_module': True, 'role': 'Get color messages in a portable way', 'debian': 'python3-colorama', 'arch': 'python-colorama'} (kibot - dep_downloader.py:1007)
DEBUG: - Registering dep {'name': 'Requests', 'python_module': True, 'role': 'mandatory', 'debian': 'python3-requests', 'arch': 'python-requests'} (kibot - dep_downloader.py:1007)
DEBUG: - Registering dep {'name': 'PyYAML', 'python_module': True, 'debian': 'python3-yaml', 'arch': 'python-yaml', 'module_name': 'yaml', 'role': 'mandatory'} (kibot - dep_downloader.py:1007)
[... cut ...]
DEBUG:- Found version (5, 1, 16, 0) (kibot - dep_downloader.py:631)
DEBUG:- Returning `/usr/bin/bash` (kibot - dep_downloader.py:816)
DEBUG:Executing: / u s r / b i n / g i t ' ' l o g ' ' - 1 ' ' - - f o r m a t = '"' % h '"' ' ' '$' K I B O T _ P C B _ N A M E (kibot - pre_set_text_variables.py:135)
ERROR:Failed to execute:
git log -1 --format="%h" $KIBOT_PCB_NAME
return code 128 (kibot - pre_set_text_variables.py:138)
ERROR:stderr:
fatal: option '-' must come before non-option arguments
 (kibot - pre_set_text_variables.py:142)
```

set-soft, 2023-01-31:
Hi @mundodisco8 !
What I need is to try to reproduce your upper/lower-case problem.
You wrote:
```
i was calling kicad with a -b Hardware/boardname, but the real path was hardware/Boardname (lowercase H). 
```
So I added a code that prints things like:
```
DEBUG:Using schematic: `/mnt/c/Users/joel.santos/Path With Spaces/Desktop/Calavera/hardware/Calavera.kicad_sch` (kibot - kiplot.py:741)
DEBUG:Real schematic name: `/mnt/c/Users/joel.santos/Path With Spaces/Desktop/Calavera/hardware/Calavera.kicad_sch` (kibot - kiplot.py:742)
```
The first line is the name you indicated at the command line, the second is a trick that should find the actual name of the file in the filesystem.
I need just this part of the log to see if we can detect the difference between the name you passed in the command line and the name KiBot detects.
If we can detect this difference KiBot could print a warning about it. Explaining you passed a name that doesn't match the exact name of the file and that this could create problems when KiBot passes this (wrong) name to other tools, even when KiBot can use the (wrong) name.

mundodisco8, 2023-01-31:
Let's see if I got it

With the same paths (real linux path `/mnt/c/Users/joel.santos/Path\ With\ Spaces/Desktop/Calavera/hardware/Calavera.kicad_pcb`)

I call

```bash
src/kibot -c /mnt/c/Users/joel.santos/Path\ With\ Spaces/Desktop/CALAVERA/ArtifactCreation.kibot.yaml -b /mnt/c/Users/joel.santos/Path\ With\ Spaces/Desktop/Calavera/HARDWARE/Calavera.kicad_pcb -e /mnt/c/Users/joel.santos/Path\ With\ Spaces/Desktop/Calavera/HARDWARE/Calavera.kicad_sch -vvvv
```
Where I replaced `hardware` from the real path with `HARDWARE` in the command, and also, because it wasn't in the path for the config file, replaced `calavera` with `CALAVERA`.

Relevant log lines

```text
DEBUG:Using configuration file: `/mnt/c/Users/joel.santos/OneDrive - Brompton Bicycle Ltd/Desktop/CALAVERA/ArtifactCreation.kibot.yaml` (kibot - __main__.py:145)
DEBUG:Using schematic: `/mnt/c/Users/joel.santos/OneDrive - Brompton Bicycle Ltd/Desktop/Calavera/HARDWARE/Calavera.kicad_sch` (kibot - kiplot.py:741)
DEBUG:Real schematic name: `/mnt/c/Users/joel.santos/OneDrive - Brompton Bicycle Ltd/Desktop/Calavera/HARDWARE/Calavera.kicad_sch` (kibot - kiplot.py:742)
DEBUG:Using PCB: `/mnt/c/Users/joel.santos/OneDrive - Brompton Bicycle Ltd/Desktop/Calavera/HARDWARE/Calavera.kicad_pcb` (kibot - kiplot.py:773)
DEBUG:Real PCB name: `/mnt/c/Users/joel.santos/OneDrive - Brompton Bicycle Ltd/Desktop/Calavera/HARDWARE/Calavera.kicad_pcb` (kibot - kiplot.py:774)
```

Am I doing wrong? I don't see any changes in capitalization. I'm on latest commit on dev

```bash
~/KiBot > src/kibot --version
KiBot 1.5.2 - Copyright 2018-2023, Salvador E. Tropea/INTI/John Beard - License: GPL v3+
~/KiBot > git status
On branch dev
Your branch is up to date with 'origin/dev'.
```

set-soft, 2023-02-01:
Hi @mundodisco8 !
Thanks for testing, nothing wrong on your side, is just the detection fails. I found it on internet and hoped it was usable.