INTI-CMNB/KiBot #367: [BUG] KiBot hangs when Docker VirtioFS is enabled

cdwilson, 2023-01-14:
_Just a heads up, this might be a Docker bug, but I wanted to post it here in case it's not (or if anybody else runs into this in the mean time)._

When I enable the VirtioFS option in Docker, KiBot hangs. When I change the option back to gRPC FUSE, things start working again.

[image]

**To Reproduce**
```
git clone https://github.com/cdwilson/test-kibot.git
cd test-kibot/
make release
```

In the example above, it hangs in the DRC check step (although if I disable DRC, it will hang other places also). Here's the screencast showing where it hangs: [pcbnew_run_drc_screencast.ogv.zip](https://github.com/INTI-CMNB/KiBot/files/10416235/pcbnew_run_drc_screencast.ogv.zip)

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad_auto:dev_k6`
- Using a desktop system? Which OS, distro and version? `macOS Ventura 13.1` / `Docker Desktop 4.16.1 (95567)`

set-soft, 2023-01-14:
Hi @cdwilson !
Thanks for reporting it, I don't have access to macOS.
From the video you posted it looks like KiBot is waiting for the DRC report creation.
This sound strange, but giving the fact that this problem happens when using a different filesystem sync system ...

In order to determine it: Does it always happens when using KiAuto? (KiBot calling KiCad GUI to do something)

Also: I think it isn't hanging is just waiting a 10 minutes time-out. How do you stop KiBot? Stopping the container?
We need to get the debug logs, when KiAuto fails it stores some interposer logs. You can also get them using debug options (-vvvv for maximum debug). But I'm not sure you'll get them by stopping the container. Please try to obtain the debug information. So we can determine if KiAuto is waiting for I/O activity.

I'm also assuming KiAuto is using the interporser mechanism, but I may be wrong. Which CPU has your machine? Is an x86 CPU or an ARM (i.e. M1) CPU?

If using the interposer this is very strange. I took a look at VirtioFS site and it seems to be something that works at filesystem level. The interposer works at user level (intercepting the standard C lib), so I don't think it could be fooled.

set-soft, 2023-03-18:
Hi @cdwilson !
If you don't have more information about it I'll close this issue.

set-soft, 2023-04-05:
As we don't have more information I'm closing this issue, the limitation is documented here, but we can't go farther.

closed 2023-04-05 by set-soft

Uvis-Acurable, 2023-10-04:
I just encounter a problem where KiBot would just halt with VirtioFS enabled in docker.

I saw it happen in creating GenCad and exporting 3d images. I get: RuntimeError('Timed out waiting for `{}`'.format(strs))

It actually produces the output correctly but just halts.

set-soft, 2023-10-04:
Hi @Uvis-Acurable !

When I consulted the VirtioFS couldn't find any clue about why (and how) it manages to just remove the C library call to file close. From your log:

```
INFO:Wait for GenCAD file creation
DEBUG:Found IO IO:open:/workdir/outputs/Fabrication/GenCAD/AcuPebble_hardware_E01-gencad.cad (pcbnew_do - interposer.py:391)
DEBUG:Waiting for ['IO:close:/workdir/outputs/Fabrication/GenCAD/AcuPebble_hardware_E01-gencad.cad'] starts=True times=1 (pcbnew_do - interposer.py:147)
```

And then a time-out. No call to close (close64, fclose, etc.). 

What is happening? No idea

set-soft, 2023-10-12:
Hi @Uvis-Acurable and @cdwilson !

I'm adding an experimental mechanism that can be used to force the use of file open/close detection using the process ID. This is what we use when no interposer is available. So this should work for VirtioFS.

The mechanism can be enabled defining the *KIAUTO_USE_PID_FOR_CREATE* environment variable, like this:

```
export KIAUTO_USE_PID_FOR_CREATE=1
```

And then running KiBot/KiAuto in the same shell.  But this will need to install KiAuto from the git repo, or wait until the next KiAuto release.