INTI-CMNB/KiBot #368: [HOW] Produktion Information and layer stackup

MarkusSBS, 2023-01-16:
Hallo 

1. Not sure if How or featur request.

But is it possible to Generate  a Layer Stackup, Produktion Details or Reports  with information Kicad already has with a wks frame? 

For example:

[image]

or the DRC ERC output 

2. Or is it possible to include files in the same repository into the Kibot Archive.

(Realy nice soulution would be including Kicad data into Latex to have nice formated reports)

set-soft, 2023-01-16:
Hi @MarkusSBS !
1. Short answer: currently we can't do it.  Generating a WKS looks too complex. A library component seems more universal. If we generate a WKS we must edit the currently used WKS and the position is hard to customize.
2. Not sure if I understand: you can include files from the repo using the `compress` output (using `from_cwd` option). You can also include the stack-up as a table in a PDF report (it gets generated using LaTeX as intermediate format) using the `report` output. Is not as nice as what you show but ...

[image]

Here is the full example:

[light_control-report.pdf](https://github.com/INTI-CMNB/KiBot/files/10427473/light_control-report.pdf)

Reports are customizable, so you can add more information, rearrange it, etc.
The above example was generated by the `report_full` output of the following config:

```yaml
# Example KiBot config file
kibot:
  version: 1

global:
  solder_mask_color_top: blue
  solder_mask_color_bottom: red
  pcb_finish: ENIG
  copper_thickness: "35"

outputs:
  - name: 'report_simple'
    comment: "Simple design report"
    type: report
    output_id: _simple
    options:
      template: simple

  - name: 'print_front'
    comment: "Top copper and silkscreen"
    type: pcb_print
    dir: Layers
    output_id: -front
    options:
      title: 'Fake title for front copper and silk (%f)'
      pages:
        - scaling: 2.0
          layers:
            - layer: F.Cu
            - layer: F.SilkS
            - layer: Edge.Cuts

  - name: 'print_bottom'
    comment: "Bottom copper and silkscreen"
    type: pcb_print
    dir: Layers
    output_id: -bottom
    options:
      title: 'Fake title for bottom copper and silk'
      pages:
        - scaling: 2.0
          layers:
            - layer: B.Cu
            - layer: B.SilkS
            - layer: Edge.Cuts

  - name: 'print_sch_svg'
    comment: "Schematic"
    type: svg_sch_print
    dir: .
    options:
      output: Schematic.svg

  - name: 'report_full'
    comment: "Full design report"
    type: report
    dir: report
    options:
      do_convert: true
```

Note that it used the PDFs from the `pcb_print` outputs. The `report_full_svg` template uses the SVGs (you must create them and it collects them). Of course you can create a template with the files you want instead of letting KiBot to guess.

MarkusSBS, 2023-01-16:
` 1. Short answer: currently we can't do it. Generating a WKS looks too complex. A library component seems more universal. If we generate a WKS we must edit the currently used WKS and the position is hard to customize.`

Ah what i mean was to print this infomation Inside of a WKS and include it in the printout not generating a dynamic WKS. 
But i thing this is obsolete by 2 

I will try your example if it fits my needs i think its basicly what i want

MarkusSBS, 2023-01-17:
report does what I want but the usage is not obvious.

Is there a best practise to injekt a template for it into the docker container and is there a information how to fix the layout like font and text possiton/white spaces?  
The example report_full is shifted to the right and has massiv whitespaces on top an bottom

set-soft, 2023-01-17:
Hi @MarkusSBS !

> Is there a best practise to injekt a template for it into the docker container

Put the file in your repo, you always need access to your repo. This is the same for any other file used by KiBot.

> is there a information how to fix the layout like font and text possiton/white spaces?

Consult the PanDoc documentation. What KiBot actually generates is a markdown file. PanDoc is used to convert the markdown into some nice PDF file. The big margins looks normal to me.

MarkusSBS, 2023-01-18:
/usr/local/lib/python3.9/dist-packages/kibot/resources/report_templates/ is this the only folder that is checkt for a template?

are there other search path or is it possible to specify a search path?

set-soft, 2023-01-18:
Hi @MarkusSBS !
You can copy your template to any place in your repo.
Just specify the path relative to the point where you run KiBot (usually the root of the repo).
The /usr/local/lib/python3.9/dist-packages/kibot/resources/report_templates/  is for built-in templates.

MarkusSBS, 2023-01-19:
is there a special syntax to use Text Vars inside the pandoc  template? 

Or do i need to define them first? 

My current test: 
```
---
title:  | 
		| ACME GmbH
		| PCB Specifications
subtitle:	|
			| Projekt ${SBS_Produkt} Rev ${SBS_REVISION}
			| ${SBS_SAP}
author: ${SBS_Prj_Reviewer}
date: ${MyDate}
footer: ${git_hash}
geometry: "left=2cm, right=2cm, top=3cm, bottom=3cm"
---

Git hash: ${git_hash}
\newpage
```

[image]

set-soft, 2023-01-20:
Hi @MarkusSBS !
The ${VAR} was to expand the report variables. But expanding text variables is a good idea.
The 47bdc2615417dc46d643c71161fce18c51ed504e patch allows expanding text variables, and also the most common WKS variables for the PCB and the environment variables.

I tested it using the `light-control.kicad_pcb` example and:

```yaml
kibot:
  version: 1

preflight:
  set_text_variables:
    - name: PROJECT
      text: Test project
    - name: SAP
      text: XYZ
    - name: AUTHOR
      text: Juan de los palotes
    - name: date
      command: 'git log -1 --format="%as" -- "$KIBOT_PCB_NAME"'
    - name: git_hash
      command: 'git log -1 --format="%h" "$KIBOT_PCB_NAME"'
      before: 'Git hash: <'
      after: '>'
    - name: gh
      command: 'git log -1 --format="%h" "$KIBOT_PCB_NAME"'

outputs:
  - name: 'report_vars'
    comment: "Various variables"
    type: report
    options:
      template: tests/data/report_vars.txt
```

```
---
title:  | 
		| ACME GmbH
		| PCB Specifications
subtitle:	|
			| Project ${PROJECT} Rev ${REVISION}
			| ${SAP}
author: ${AUTHOR}
date: ${ISSUE_DATE}
footer: ${git_hash}
geometry: "left=2cm, right=2cm, top=3cm, bottom=3cm"
---

Git hash: ${git_hash}
\newpage
```

I got them expanded. Which pandoc format supports the title and other YAML stuff?

MarkusSBS, 2023-01-20:
`I got them expanded. Which pandoc format supports the title and other YAML stuff? `
dont know i just used the docker container you provided 
```
netsoft/kicad_auto:dev_k6 
with apt-get update && apt-get -y install pandoc texlive-latex-recommended
```
and it just worked

MarkusSBS, 2023-01-20:
i have testet it with setsoft/kicad_auto_test:dev_k6 an it works

MarkusSBS, 2023-01-25:
can be closed for now

closed 2023-01-25 by MarkusSBS

set-soft, 2023-04-28:
Related to this: KiCad 7 now supports:
- Place | Add Board Characteristics
- Place | Add Stackup Table

set-soft, 2024-04-15:
Ok, now is possible to insert the above tables using KiCad and keep them synced using two new preflights

set-soft, 2024-04-17:
And this new preflight can create this

[image]