INTI-CMNB/KiBot #370: [BUG]

MarkusSBS, 2023-01-17:
**Describe the bug**
report_full not working do to missing dependencies 

**To Reproduce**
run output job with setsoft/kicad_auto:dev_k6 docker container 

**Expected behavior**
generating report PDF 

**Screenshots**
```
- 'Full design report' (report_full) [report]
DEBUG:Output destination: /builds/hardware/kibot_test/Fabrication/report (kibot - kiplot.py:359)
DEBUG:Eurocircuits Pattern class for `minimum` is 7 because the clearance is 0.2, track is 0.13 and OAR is 0.13 (kibot - out_report.py:93)
DEBUG:Eurocircuits Pattern class for `measured` is 4 because the clearance is 0.2, track is 0.8 and OAR is 0.15 (kibot - out_report.py:93)
DEBUG:Eurocircuits Pattern class for `defined` is 7 because the clearance is 0.2, track is 0.13 and OAR is 0.13 (kibot - out_report.py:93)
DEBUG:Eurocircuits Drill class for `minimum` is D because the drill is 0.3 (kibot - out_report.py:106)
DEBUG:Eurocircuits Drill class for `measured` is B because the drill is 0.5 (kibot - out_report.py:106)
DEBUG:Eurocircuits Drill class for `defined` is C because the drill is 0.4 (kibot - out_report.py:106)
DEBUG:Report template: `/usr/local/lib/python3.9/dist-packages/kibot/resources/report_templates/report_full.txt` (kibot - out_report.py:752)
DEBUG:- Evaluating `edge_connector or castellated_pads or edge_plating` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `edge_connector or castellated_pads or edge_plating` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `edge_connector` (kibot - out_report.py:742)
DEBUG:- Result `` (kibot - out_report.py:747)
DEBUG:- Evaluating `castellated_pads` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `edge_plating` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `stackup` (kibot - out_report.py:742)
DEBUG:- Result `yes` (kibot - out_report.py:747)
DEBUG:- Evaluating `stackup and impedance_controlled` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `stackup and impedance_controlled` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `stackup` (kibot - out_report.py:742)
DEBUG:- Result `yes` (kibot - out_report.py:747)
DEBUG:- Evaluating `stackup` (kibot - out_report.py:742)
DEBUG:- Result `yes` (kibot - out_report.py:747)
DEBUG:- Evaluating `stackup` (kibot - out_report.py:742)
DEBUG:- Result `yes` (kibot - out_report.py:747)
DEBUG:- Evaluating `stackup` (kibot - out_report.py:742)
DEBUG:- Result `yes` (kibot - out_report.py:747)
DEBUG:- Report context: `stackup` (kibot - out_report.py:767)
DEBUG:- Evaluating `stackup` (kibot - out_report.py:742)
DEBUG:- Result `yes` (kibot - out_report.py:747)
DEBUG:- Report context: `defined_tracks` (kibot - out_report.py:767)
DEBUG:- Report context: `used_tracks` (kibot - out_report.py:767)
DEBUG:- Report context: `defined_vias` (kibot - out_report.py:767)
DEBUG:- Report context: `used_vias` (kibot - out_report.py:767)
DEBUG:- Report context: `hole_sizes_no_vias` (kibot - out_report.py:767)
DEBUG:- Report context: `oval_hole_sizes` (kibot - out_report.py:767)
DEBUG:- Report context: `drill_tools` (kibot - out_report.py:767)
DEBUG:- Evaluating `schematic_svgs` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `schematic_svgs` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `schematic_svgs` (kibot - out_report.py:742)
DEBUG:- Result `False` (kibot - out_report.py:747)
DEBUG:- Evaluating `layer_pdfs` (kibot - out_report.py:742)
DEBUG:- Result `True` (kibot - out_report.py:747)
DEBUG:- Evaluating `layer_pdfs` (kibot - out_report.py:742)
DEBUG:- Result `True` (kibot - out_report.py:747)
DEBUG:- Evaluating `layer_pdfs` (kibot - out_report.py:742)
DEBUG:- Result `True` (kibot - out_report.py:747)
DEBUG:- Report context: `layer_pdfs` (kibot - out_report.py:767)
DEBUG:Report output: `/builds/hardware/kibot_test/Fabrication/report/kicad_ci_test-report.txt` (kibot - out_report.py:779)
DEBUG:Starting tool check for Pandoc (kibot - dep_downloader.py:809)
DEBUG:- Returning `None` (kibot - dep_downloader.py:816)
ERROR:Missing `pandoc` command (Pandoc), install it (kibot - dep_downloader.py:773)
ERROR:Home page: https://pandoc.org/ (kibot - dep_downloader.py:773)
ERROR:Download page: https://github.com/jgm/pandoc/releases (kibot - dep_downloader.py:773)
ERROR:Debian package: pandoc (kibot - dep_downloader.py:773)
ERROR:- Recommended extra Debian packages: texlive texlive-latex-base texlive-latex-recommended (kibot - dep_downloader.py:773)
ERROR:Arch package: pandoc (kibot - dep_downloader.py:773)
ERROR:- Recommended extra Arch packages: texlive-core (kibot - dep_downloader.py:773)
ERROR:In CI/CD environments: the `kicad_auto_test` docker image contains it. (kibot - dep_downloader.py:773)
ERROR:Output that needs it: report (kibot - dep_downloader.py:773)
ERROR:Used to create PDF/ODF/DOCX files (kibot - dep_downloader.py:773)
ERROR:Try running the installation checker: kibot-check (kibot - dep_downloader.py:773)
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab
- Using a docker image? setsoft/kicad_auto:dev_k6

**Additional context**
Config: 

  - name: 'report_full'
    comment: "Full design report"
    type: report
    dir: report
    options:
      do_convert: true

MarkusSBS, 2023-01-17:
i belive it is do to the size of latex and pandoc.

What is the preferred way to include this into the workflow?

set-soft, 2023-01-17:
Hi @MarkusSBS !
This was on-purpose.
The current policy (that I'm changing) is to provide light-weight docker images by default.
The `ghcr.io/inti-cmnb/kicad6_auto_full` contains Pandoc.
My idea was to make the workflows faster, but it generates extra questions and situations like this one: the `dev` gets created using the light-weight images. Or this: the GitHub action uses the light-weight image.
And now I'm introducing Blender render, which is far from trivial to setup. 

> What is the preferred way to include this into the workflow?

This should be enough, but I also add `texlive-latex-base`. Look here: [kicad_auto_test](https://github.com/INTI-CMNB/kicad_auto_test)

I'll be trying to incorporate Blender in the next hours. The size of the images will grow notably, but I think the added functionality is worth the extra weight.

closed 2023-01-17 by set-soft