INTI-CMNB/KiBot #371: [BUG] file_id is often ignored by printing or export

N0cynym, 2023-01-17:
**Describe the bug**
If I print or export and set under `options` the `output_name` I miss the `file_if` if I use variants.

**To Reproduce**
```yaml
variants:
  - name: 'default'
    comment: 'default'
    type: kibom
  - name: 'variant1'
    comment: 'my variant'
    type: kibom
    file_id: _variant1
    variant: variant1
    pre_transform: '_var_rename'

outputs:    
  - name: 'print_sch'
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: .
    options:
      output: Schematics.pdf
```
**Expected behavior**
My goal is to export my content the following file naming structure to make it sortable:

- Schematics-myProject-pre_transform
- Gerber-myProject-pre_transform
- Drill-myProject-pre_transform
- PCBprint-myProject-pre_transform

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow GitLab
- kicad_auto:dev_k6`

set-soft, 2023-01-17:
Hi @N0cynym !
I don't fully understand it.
What do you get with your example? What do you expect? What's wrong?
About what you want: 
- What about using `dir: Schematics` for the stuff you want to separate as schematics?
- You can also use `output_id` and some output pattern like `%I-%f-%v.%x`

N0cynym, 2023-01-20:
I'll close this issue and use the generated name from kibot so kibot is than correctly adding variant name behind.

The files will be sorted by folder name and not file name.

Thanks for your reply!

closed 2023-01-20 by N0cynym