INTI-CMNB/KiBot #372: [BUG] Using the copy_files output in the compress output does not work

alexanderwiller, 2023-01-18:
**Describe the bug**
When trying to add files copied with `copy_files` to some archive with `compress`, an internal error appears:

```
- 'PDF with board specification' (board_specification) [copy_files]
- '' (generic_manufacturing_output) [compress]
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 403, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 501, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 492, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 413, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 192, in run
    self.options.run(self.expand_filename(output_dir, output))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_compress.py", line 236, in run
    files, dirs_outs = self.get_files(output)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_compress.py", line 156, in get_files
    files_list, out_dir, out = get_output_targets(f.from_output, self._parent)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 398, in get_output_targets
    files_list = out.get_targets(out_dir)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_base.py", line 114, in get_targets
    return self.options.get_targets(out_dir)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_copy_files.py", line 210, in get_targets
    return sorted([os.path.join(out_dir, v) for v in files.values()])
AttributeError: 'list' object has no attribute 'values'
```
Apparently, a list is provided, but a dict expected.

**To Reproduce**
Run kibot with any board/schematic and specify the following two outputs:
```
outputs:
- name: board_specification
  comment: PDF with board specification
  type: copy_files
  options:
    files:
    - source: "PCB specifications.pdf"

- name: generic_manufacturing_output
  type: compress
  options:
    format: ZIP
    files:
    - from_output: board_specification
```

**Expected behavior**
The copied files are added to the ZIP archive.

**Environment (please complete the following information):**
Where are you running KiBot: `ghcr.io/inti-cmnb/kicad6_auto:dev`

set-soft, 2023-01-18:
Hi @alexanderwiller !
b6240d5fea09750d01a3ab4f390de89ac74c265f should fix it.
ALso note that the above example is copying the PDF over itself, now reported as error.

closed 2023-01-18 by set-soft

alexanderwiller, 2023-01-25:
Thanks, it is fixed.

Regarding
>  ALso note that the above example is copying the PDF over itself, now reported as error.

I do not understand - the above example does work for me without creating any error.

set-soft, 2023-01-25:
> I do not understand - the above example does work for me without creating any error.

In your example:

```yaml
- name: board_specification
  comment: PDF with board specification
  type: copy_files
  options:
    files:
    - source: "PCB specifications.pdf"
```

The `PCB specifications.pdf` will be copied to the folder specified by `-d`, but if you omit it the file will be copied over itself. I sugest using a source sub-dir, so you don't erase the file by accident. 
The current `dev` code avoids such a situation, but the stable release can do it.