INTI-CMNB/KiBot #374: [BUG] DRC warnings are treated as errors

geissdoerfer, 2023-01-22:
Running DRC checks on a project that has DRC warnings but no DRC errors fails.

I'm using the KiBot docker image with GitHub Action `v2_k6` with the following kibot_config.yml:

```
kibot:
  version: 1

preflight:
  check_zone_fills: true
  run_drc: true
```

Output from the action:

```
| *****************************************************************************************
| *****************************************************************************************
| 
| 
| KiBot GitHub Action v2
| 
| 
| KiBot 1.5.1 - Copyright 2018-2022, Salvador E. Tropea/INTI/John Beard - License: GPL v3+
| ii  kicad          6.0.10+dfsg-1~bpo11+1 amd64        Electronic schematic and PCB design software
| Debian: 11.6
| pcbnew_do
| 2.1.1 -
| Copyright
| 2018-2023, 
| INTI/Produc
| tize SPRL -
| License:
| Apache 2.0
| KiCost
| v1.1.15
| PcbDraw
| 0.9.0-5
| iBoM: v2.5.0-2
| 
| 
| *****************************************************************************************
| *****************************************************************************************
| Options: -c 'kibot_test.yml' -d '.' -b 'kicad/riotee-board.kicad_pcb' -e 'kicad/riotee-board.kicad_sch'
| - Running the ERC
| WARNING:(W058) Missing KiCad main config file /root/.config/kicad/6.0/kicad_common.json (eeschema_do - misc.py:181) (kibot - kiplot.py:128)
| - Running the DRC
| ERROR:Found 0 DRC errors and 9 unconnected pad/s or warnings (pcbnew_do - pcbnew_do:1257) (kibot - kiplot.py:125)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(199.2225 mm, 71.3100 mm): Line on F.Silkscreen
|     @(199.2200 mm, 71.0100 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(199.2225 mm, 71.3100 mm): Line on F.Silkscreen
|     @(199.2200 mm, 71.1600 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(199.9300 mm, 64.1100 mm): Line on F.Silkscreen
|     @(198.7222 mm, 64.1800 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(197.9225 mm, 71.1600 mm): Line on F.Silkscreen
|     @(199.2200 mm, 71.1600 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(197.9225 mm, 71.1600 mm): Line on F.Silkscreen
|     @(197.9200 mm, 71.1600 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(199.2225 mm, 71.1600 mm): Line on F.Silkscreen
|     @(199.2200 mm, 71.0100 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(199.2225 mm, 71.1600 mm): Line on F.Silkscreen
|     @(199.2200 mm, 71.1600 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(199.2225 mm, 71.1600 mm): Line on F.Silkscreen
|     @(197.9200 mm, 71.1600 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| WARNING:(W058) (silk_overlap) Silkscreen overlap
|     @(198.6300 mm, 64.1100 mm): Line on F.Silkscreen
|     @(198.7222 mm, 64.1800 mm): Line on F.Silkscreen (pcbnew_do.kiauto.file_util - file_util.py:149) (kibot - kiplot.py:128)
| ERROR:DRC errors: 9 (kibot.pre_run_drc - pre_run_drc.py:75)
[Kibot run ERC and DRC/test]   ❌  Failure - Main INTI-CMNB/KiBot@v2_k6
[Kibot run ERC and DRC/test] exit with `FAILURE`: 5
[Kibot run ERC and DRC/test] 🏁  Job failed
```

Expected behavior: DRC should pass. This seems to work for the ERC, where there is an explicit option (`erc_warnings`) to treat ERC warnings as errors.

set-soft, 2023-01-23:
Hi @geissdoerfer !

Use the `ignore_unconnected` option, they go to the same counter: `ERROR:Found 0 DRC errors and 9 unconnected pad/s or warnings`

geissdoerfer, 2023-01-23:
Thanks @set-soft that solves my immediate problem. But it will lead to unconnected pads being ignored which is a serious issue that I would absolutely want to make the DRC fail.

Wouldn't something like `drc_warnings` make sense, similar to what `erc_warnings` does already?

set-soft, 2023-01-23:
Hi @geissdoerfer !
Warnings were introduced by KiCad 6, KiCad 5 has only violations.
You should get 0 warnings, perhaps using exclusions.
I'll eventually add correct support for warnings, but this will be a change in behavior, which isn't good.
So it must be done carefully.

closed 2023-01-23 by set-soft

matthijskooijman, 2023-05-15:
I was also a bit suprised  by this behavior. Would it make sense to keep this issue open as a reminder to improve this?

> Wouldn't it make more sense to pass the DRC with warnings and having something like drc_warnings to make warnings also fail the DRC, similar to what erc_warnings does already?

This seems like a good approach (though maybe using something more clear like `drc_fail_on_warnings` or something - maybe add `erc_fail_on_warnings` as an alias of `erc_warnings` as well too?). If compatibility is an issue here, the default value could of course be true (though maybe consider switching the default to false at some point, to be consistent with `erc_warnings`?).

set-soft, 2024-04-17:
Just as a note: this problem is solved by the new KiCad 8 preflights named *erc* and *drc*. 
They use a modern errors classification, and can also generate HTML reports

commit 91077773 (2023-01-23): [DOCs] Clarified how KiCad 6 warnings are counted

Closes #374