INTI-CMNB/KiBot #377: [FEATURE] Text Variables in output file name

MarkusSBS, 2023-01-25:
I think there is no easy way to use text variable in output files.

For example a self defined variable for a material Number.

Workaround is to set a textvar to a sheet comment and use %C but i can imagine that this can be to limeting in the future

set-soft, 2023-01-25:
Hi @MarkusSBS !
This is already implemented, here is an example (using the outputs from #378):

[#377.zip](https://github.com/INTI-CMNB/KiBot/files/10504567/377.zip)

Here we use:

```yaml
preflight:
  set_text_variables:
    - variable: "V1"
      text: "PREFIX_FOR_ALL"
```

And then we expand V1 as with any other text variable: `output: "${V1}_%v.%p_Rev%r-%i_%D-%T.TXT"`

The .zip includes the generated files:

```
$ ls  *
PREFIX_FOR_ALL_.light_control_Rev-bom_2023-01-25-20-19-16.csv  PREFIX_FOR_ALL_.light_control_Rev-both_pos_2023-01-25-20-19-16.csv

dril:
PREFIX_FOR_ALL_.light_control_Rev--NPTH_2023-01-25-20-19-16.TXT  PREFIX_FOR_ALL_.light_control_Rev--PTH_2023-01-25-20-19-16.TXT

Gerbers:
PREFIX_FOR_ALL_.light_control_Rev-B_Cu_2023-01-25-20-19-16.GBL     PREFIX_FOR_ALL_.light_control_Rev-B_Silkscreen_2023-01-25-20-19-16.GBO  PREFIX_FOR_ALL_.light_control_Rev-F_Mask_2023-01-25-20-19-16.GTS        PREFIX_FOR_ALL_.light_control_Rev-In1_Cu_2023-01-25-20-19-16.G1
PREFIX_FOR_ALL_.light_control_Rev-B_Mask_2023-01-25-20-19-16.GBS   PREFIX_FOR_ALL_.light_control_Rev-Edge_Cuts_2023-01-25-20-19-16.GML     PREFIX_FOR_ALL_.light_control_Rev-F_Paste_2023-01-25-20-19-16.GTP       PREFIX_FOR_ALL_.light_control_Rev-In2_Cu_2023-01-25-20-19-16.G2
PREFIX_FOR_ALL_.light_control_Rev-B_Paste_2023-01-25-20-19-16.GBP  PREFIX_FOR_ALL_.light_control_Rev-F_Cu_2023-01-25-20-19-16.GTL          PREFIX_FOR_ALL_.light_control_Rev-F_Silkscreen_2023-01-25-20-19-16.GTO

PCB_Docu:
PREFIX_FOR_ALL_.light_control_Rev-assembly_2023-01-25-20-19-16.pdf
```

Is that what you want?

MarkusSBS, 2023-01-26:
ah thats sound like what i want.
I will test it 

maybe this part should be added to the docu

commit fce9a2d5 (2023-01-26): [DOCs] Documented the text variable expansion in `output`

Closes #377
closed 2023-01-26 by set-soft