INTI-CMNB/KiBot #378: [BUG] %Cn expanded for PDF output but not for gerber, drill, BOM or PnP files

MarkusSBS, 2023-01-25:
For the PCB 

output: option %Cn expandes only for the PDF not for gerber, Dill, BOM or PnP files 

**Config**
```
- name: 'gerbers'
    comment: "Gerbers for the board house"
    type: gerber
    dir: Gerbers
    options:
      # generic layer options
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: true
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: false
      force_plot_invisible_refs_vals: false
      tent_vias: true
      use_protel_extensions: true
      create_gerber_job_file: false
      output: "%C1_%v.%p_Rev%r-%i_%D-%T.%x"
      gerber_precision: 4.6
      use_gerber_x2_attributes: false
      use_gerber_net_attributes: false
      disable_aperture_macros: true
      line_width: 0.1
      uppercase_extensions: true
      subtract_mask_from_silk: true
      inner_extension_pattern: '.g%n'
      edge_cut_extension: '.gml'
    layers:
      - 'copper'
      - 'F.SilkS'
      - 'B.SilkS'
      - 'F.Mask'
      - 'B.Mask'
      - 'Edge.Cuts'
      - 'F.Paste'
      - 'B.Paste'

  - name: 'drill'
    comment: "Drill files"
    type: excellon
    dir: dril
    options:
      pth_and_npth_single_file: false
      pth_id: '-PTH'
      npth_id: '-NPTH'
      output: "%C1_%v.%p_Rev%r-%i_%D-%T.TXT"
      
      
  - name: 'print_pcb_pdf'
    comment: "PDF for the PCB"
    type: pcb_print
    dir: PCB_Docu
    options:
      #svg_precision: 6
      realistic_solder_mask: false
      force_edge_cuts: true
      keep_temporal_files: false
      #enable_ki6_frame_fix: true
      #sheet_reference_layout: pcb_print.kicad_wks
      scaling: 1.0
      pages:
        - layers: [ F.SilkS, Dwgs.User ]
          sheet: 'Top Silk layer'
        - layers: [ F.Mask, Dwgs.User ]
          sheet: 'Top Mask layer'
        - layers: [ F.Cu, Dwgs.User ]
          sheet: '%ld (%ln)'
          repeat_for_layer: F.Cu
          repeat_layers: copper
        - layers: [ B.Mask, Dwgs.User ]
          sheet: 'Top Mask layer'
        - layers: [ B.SilkS, Dwgs.User ]
          sheet: 'Bottom layer'
        - layers: [ F.Paste,Dwgs.User]
          sheet: 'Top paste Fabrication layers'
        - layers: [ B.Paste,Dwgs.User]
          sheet: 'Bottom paste Fabrication layers'
      output: "%C1_%v.%p_Rev%r-%i_%D-%T.%x"

 - name: 'position'
    comment: "Pick and place file"
    type: position
    options:
      output: "%C1_%v.%p_Rev%r-%i_%D-%T.%x"
      format: CSV
      units: millimeters
      separate_files_for_front_and_back: false
      only_smd: true
      columns:
        - id: Ref
          name: Designator
        - Val
        - Package
        - id: PosX
          name: "Mid X"
        - id: PosY
          name: "Mid Y"
        - id: Rot
          name: Rotation
        - id: Side
          name: Layer

  - name: 'bom'
    comment: "BoM"
    type: bom
    options:
      output: "%C1_%v.%p_Rev%r-%i_%D-%T.%x"
      ref_separator: ','
      columns:
        - field: Value
          name: Comment
        - field: References
          name: Designator
        - Footprint
        - field: 'LCSC#'
          name: 'LCSC Part #'
      csv:
        hide_pcb_info: true
        hide_stats_info: true
        quote_all: true
```

**Expected behavior**
same behavior for all output files 

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab`
- Using a docker image?  `kicad_auto:dev_k6`

set-soft, 2023-01-25:
Hi @MarkusSBS !
How do you set C1? Is it in the PCB *and* the schematic?
I tested it defining the value in the PCB and not in the schematic and it was replaced by all but the BoM, which is logic since the schematic doesn't define it.

Here are the files I used:

[#378.zip](https://github.com/INTI-CMNB/KiBot/files/10504488/378.zip)

The results are included for: `kibot -d pp`

MarkusSBS, 2023-01-26:
i have set it in the schematic and the bom.

that is what i get for the gerber files:
`_.Zellsim_RevA2-Front_25.01.2023-142344.GTL`

it is deffined with:
`"%C1_%v.%p_Rev%r-%i_%D-%T.%x"`

set-soft, 2023-01-26:
This is different from what I get with the files I sent you.
Please create a small example where the problem is reproduced, I can't reproduce it.
How do you invoke KiBot?
Take a look at the debug information looking for something like this:

```
DEBUG:Expanding `%C1_%v.%p_Rev%r-%i_%D-%T.%x` in PCB context for <kibot.out_gerber.GerberOptions object at 0x7fad663497f0> parent: 'Gerbers for the board house' (gerbers) [gerber] (kibot - optionable.py:347)
DEBUG:PCB title: `light_control` (kibot.gs - gs.py:361)
DEBUG:PCB date: `2023-01-25` (kibot.gs - gs.py:362)
DEBUG:PCB revision: `` (kibot.gs - gs.py:363)
DEBUG:PCB company: `` (kibot.gs - gs.py:364)
DEBUG:PCB comment 1: `COMMENT1_STR` (kibot.gs - gs.py:366)
DEBUG:PCB comment 2: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 3: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 4: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 5: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 6: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 7: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 8: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 9: `` (kibot.gs - gs.py:366)
DEBUG:Expanded `COMMENT1_STR_.light_control_Rev-F_Cu_2023-01-26-09-19-12.gtl` (kibot - optionable.py:401)
```

Here you can see the gerber expansion:

```
DEBUG:Expanding `%C1_%v.%p_Rev%r-%i_%D-%T.%x` in PCB context for <kibot.out_gerber.GerberOptions object at 0x7fad663497f0> parent: 'Gerbers for the board house' (gerbers) [gerber] (kibot - optionable.py:347)
```

Then you see how the title block is processed:

```
DEBUG:PCB title: `light_control` (kibot.gs - gs.py:361)
DEBUG:PCB date: `2023-01-25` (kibot.gs - gs.py:362)
DEBUG:PCB revision: `` (kibot.gs - gs.py:363)
DEBUG:PCB company: `` (kibot.gs - gs.py:364)
DEBUG:PCB comment 1: `COMMENT1_STR` (kibot.gs - gs.py:366)
DEBUG:PCB comment 2: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 3: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 4: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 5: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 6: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 7: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 8: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 9: `` (kibot.gs - gs.py:366)
```

And then the result:

```
DEBUG:Expanded `COMMENT1_STR_.light_control_Rev-F_Cu_2023-01-26-09-19-12.gtl` (kibot - optionable.py:401)
```

I used `-vvvv`

MarkusSBS, 2023-01-26:
For my defined Fabrication job the deebug shows:
[image]

```
DEBUG:Expanding `%C1_%v.%p_Rev%r-%i_%D-%T.%x` in PCB context for <kibot.out_gerber.GerberOptions object at 0x7fd72700e190> parent: 'Gerbers for the board house' (gerbers) [gerber] (kibot - optionable.py:347)
DEBUG:Replacing KiCad text variables: ${SBS_Date} ->  (kibot.gs - gs.py:340)
DEBUG:Replacing KiCad text variables: ${SBS_Produkt} -> Zellsim (kibot.gs - gs.py:340)
DEBUG:Replacing KiCad text variables: ${SBS_REVISION} -> A2 (kibot.gs - gs.py:340)
DEBUG:Replacing KiCad text variables: ${SBS_SAP} ->  (kibot.gs - gs.py:340)
DEBUG:PCB title: `Zellsim` (kibot.gs - gs.py:361)
DEBUG:PCB date: `25.01.2023-141205` (kibot.gs - gs.py:362)
DEBUG:PCB revision: `A2` (kibot.gs - gs.py:363)
DEBUG:PCB company: `ACME GmbH` (kibot.gs - gs.py:364)
DEBUG:PCB comment 1: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 2: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 3: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 4: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 5: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 6: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 7: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 8: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 9: `` (kibot.gs - gs.py:366)
```

`DEBUG:Replacing KiCad text variables: ${SBS_SAP} ->  (kibot.gs - gs.py:340) `
 ${SBS_SAP} should be replaced bei the Content of the text var  but it is empty?

in the  KiBot YAML it is set with:
```
  set_text_variables:
    - name: 'SBS_SAP'
      text: '66.6666.66'
```
this is also used in the Documentation job 
 there it is expanded corret: 

```
DEBUG:Replacing KiCad text variables: ${SBS_Date} -> 26.01.2023 (kibot.gs - gs.py:340)
WARNING:(W075) Trying to reformat PCB time, but not in ISO format (26.01.2023) (kibot.gs - gs.py:222)
DEBUG:Replacing KiCad text variables: ${SBS_Produkt} -> Zellsim (kibot.gs - gs.py:340)
DEBUG:Replacing KiCad text variables: ${SBS_REVISION} -> Z9 (kibot.gs - gs.py:340)
DEBUG:Replacing KiCad text variables: ${SBS_SAP} -> 66.6666.66 (kibot.gs - gs.py:340)
DEBUG:PCB title: `Zellsim` (kibot.gs - gs.py:361)
DEBUG:PCB date: `26.01.2023` (kibot.gs - gs.py:362)
DEBUG:PCB revision: `Z9` (kibot.gs - gs.py:363)
DEBUG:PCB company: `ACME GmbH` (kibot.gs - gs.py:364)
DEBUG:PCB comment 1: `66.6666.66` (kibot.gs - gs.py:366)
DEBUG:PCB comment 2: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 3: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 4: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 5: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 6: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 7: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 8: `` (kibot.gs - gs.py:366)
DEBUG:PCB comment 9: `` (kibot.gs - gs.py:366)
```

MarkusSBS, 2023-01-26:
the quick solution for this is maybe just using what you sad in #376 but the behavior here is still weird

MarkusSBS, 2023-01-26:
maybe related to the problem #36 i cant reproduce anymore

set-soft, 2023-01-26:
Hi @MarkusSBS !

I tried using a variable and the problem didn't show up.
Look in the log where the set_text_variables is executed, like this:

```
DEBUG:Preflight apply set_text_variables (kibot.pre_base - pre_base.py:80)
DEBUG:- Current variables: {} (kibot - pre_set_text_variables.py:118)
DEBUG:  - SBS_SAP -> 66.6666.66 (kibot - pre_set_text_variables.py:148)
DEBUG:- Expanding %X patterns in variables (kibot - pre_set_text_variables.py:150)
DEBUG:Expanding `66.6666.66` in SCH context for set_text_variables: True parent: None (kibot - optionable.py:347)
DEBUG:Expanded `66.6666.66` (kibot - optionable.py:401)
DEBUG:- New list of variables: {'SBS_SAP': '66.6666.66'} (kibot - pre_set_text_variables.py:159)
DEBUG:Preflight run set_text_variables (kibot.pre_base - pre_base.py:84)
```

I used:

```yaml
  set_text_variables:
    - name: 'SBS_SAP'
      text: '66.6666.66'
```

Again: **How are you invoking KiBot**
Are you sure you are running the pre-flights?

> maybe related to the problem https://github.com/INTI-CMNB/KiBot/issues/36 i cant reproduce anymore

#361 ?

May be, but this looks like some particular sequence in the use/invocation.

MarkusSBS, 2023-01-26:
For Docu: 

` - "[ -f *.kicad_sch ] && kibot -vvvv -d Fabrication -s run_drc,run_erc print_sch print_pcb_pdf report_full report_pcb_svg "`

For fab/gerber:

`- "[ -f *.kicad_pcb ] && kibot -vvvv  -d Fabrication -s all position bom gerbers drill archiv"`

MarkusSBS, 2023-01-26:
```
DEBUG:Parsing preflight options: {'run_erc': True, 'update_xml': True, 'run_drc': True, 'check_zone_fills': True, 'ignore_unconnected': False, 'set_text_variables': [{'name': 'SBS_Prj_Reviewer', 'command': 'git log -1 --pretty=format:"%cn"'}, {'name': 'SBS_Prj_Creator', 'command': 'git log -1 --pretty=format:"%an"'}, {'name': 'git_hash', 'command': 'git log -1 --format="%h"', 'before': 'Git:'}, {'name': 'SBS_Date', 'command': 'date -d @`git log -1 --format="%at"` +%d.%m.%Y'}, {'name': 'SBS_SAP', 'text': '66.6666.66'}, {'name': 'SBS_REVISION', 'text': 'Z9'}, {'name': 'SBS_PLATFORM', 'text': '120'}, {'name': 'SBS_Doc_status', 'text': 'test'}]} (kibot - config_reader.py:245)
DEBUG:Parsing preflight run_erc (kibot - config_reader.py:254)
DEBUG:Parsing preflight update_xml (kibot - config_reader.py:254)
DEBUG:Parsing preflight run_drc (kibot - config_reader.py:254)
DEBUG:Parsing preflight check_zone_fills (kibot - config_reader.py:254)
DEBUG:Parsing preflight ignore_unconnected (kibot - config_reader.py:254)
DEBUG:Parsing preflight set_text_variables (kibot - config_reader.py:254)
```

hm there is no `DEBUG:Preflight apply set_text_variables ` but it is parsing the preflight corret

MarkusSBS, 2023-01-26:
AH OH user error  the -s option was out off an example an it skipps all preflight. now with all my addons this will break the build.

For the example its bassicly dont care

MarkusSBS, 2023-01-26:
https://gitlab.com/set-soft/kicad-ci-test/-/blob/master/.gitlab-ci.yml

set-soft, 2023-01-26:
Done, I changed the example.

closed 2023-01-26 by set-soft