INTI-CMNB/KiBot #379: [BUG] error 1 (kibot - out_kibom.py:417)

N0cynym, 2023-01-27:
**Describe the bug**
The following constellation in reproduce is causing the error `error 1 (kibot - out_kibom.py:417)`

**To Reproduce**
```yaml
  - name: 'bom_html'
    comment: "Bill of Materials in HTML format"
    type: bom
    dir: .
    options:
      html:
         logo: false
         generate_dnf: false
         #dnf_list: false
      format: HTML  # HTML or CSV
      columns:
        - Row
        - References
        - Quantity Per PCB
        - field: Value
          name: Intranet ID
      normalize_values: false
  - name: 'bom_csv'
    comment: "Bill of Materials in CSV format"
    type: kibom
    dir: .
    options:
      format: CSV  # HTML or CSV
      conf:
          columns:
            - Value
            - Quantity Per PCB
          hide_pcb_info: true
```

**Expected behavior**
I want to export my csv with just:

```yaml
          columns:
            - Value
            - Quantity Per PCB
```

This is working fine with:

```yaml
  - name: 'bom_csv'
    comment: "Bill of Materials in CSV format"
    type: kibom
    dir: .
    options:
      format: CSV  # HTML or CSV
      conf:
          columns:
            - Value
            - Quantity Per PCB
          hide_pcb_info: true
```

But if I do one export with type `bom` and one with type `kibom` it's failing.

**Environment (please complete the following information):**
Where are you running KiBot:
- In a GitLab CI/CD workflow
- Using a docker image `kicad_auto:dev_k6`

N0cynym, 2023-01-27:
This is how it works with `bom` how I wanted it:

```yaml
  - name: 'bom_csv'
    comment: "Bill of Materials in CSV format"
    type: bom
    dir: .
    options:
      format: CSV  # HTML or CSV
      csv:
         hide_pcb_info: true
         hide_stats_info: true
      columns:
            - Value
            - Quantity Per PCB
```

set-soft, 2023-01-27:
Hi @N0cynym !
Why are you using KiBoM *and* the internal BoM?
I can't reproduce your problem, I tried with: 
[#379.zip](https://github.com/INTI-CMNB/KiBot/files/10519363/379.zip)

Can you post the whole error using debug? i.e. `kibot -vvvv`

N0cynym, 2023-01-27:
I'll do this. Thanks for having a look to my issue!

closed 2023-01-27 by set-soft