INTI-CMNB/KiBot #380: [FEATURE] LCSC/JLCPCB (3D) parts import

ModischFabrications, 2023-01-27:
**Is your feature request related to a problem? Please describe.**
3D models and some symbols/footprints are missing when building via CI, but I don't want to copy around my parts lib. I am using simple, mostly unedited files imported directly from LCSC. 

**Describe the solution you'd like**
I would love to have an integration or workflow description to import parts via a LCSC->KiCAD importer. This one is what I am currently using, I think integration is feasible through a lookup of the BOM tags: https://github.com/uPesy/easyeda2kicad.py

With this cmmand: 
`easyeda2kicad --full --output "path\to\local\dir\KiCad\lcsc" --v5 --overwrite --lcsc_id=C488349`

It might be possible to jerry-rig it from outside, but I'm not experienced enough to do that. 

**Describe alternatives you've considered**
I could copy my file libraries into every single project, but I really don't want to do that.

**Additional context**
This is where/how I'm planning to use it, note the empty footprints on 3D renderings: https://github.com/ModischFabrications/EasyLight_PCB

Feel free to postpone forever if it's too specific or complex.

ModischFabrications, 2023-01-27:
Minor addition: There is a typo I found because I got that message a few hundred times, see #381

set-soft, 2023-01-28:
Hi @ModischFabrications !
Didn't know you can get such an information from the EasyEDA site.
The bad thing is that `easyeda2kicad` script uses some Python extensions that are rare and doesn't even install on Debian 11.6.
The good thing is that I was able to isolate the code that downloads the 3D models. Is around 9 kiB and doesn't depend on external packages (pure Python standard libs)
The generated WRL needs some adjustments ... colors are preserved, but the materials are lost. KiCad has some pre-defined materials, using them you can generate high quality renders using Blender.

ModischFabrications, 2023-01-28:
Great, that's awesome to hear! It might be interesting to contact the owner of the repo for questions or upstream fixes, it seems that it's actively maintained.

set-soft, 2023-01-31:
Hi @ModischFabrications !
I was playing with this concept, but found various problems:

1. EasyEDA is providing an OBJ (Wavefront) 3D model, which is easy to convert to a WRL (just surfaces).  But KiCad also needs a STEP model (a solid representation) to allow exporting the PCB as a 3D model for 3D validations.
2. The models needs offsets and/or rotations. This is bad because it generates problems. I.e. if we export the PCB as a VRML model its bounding box looks bigger.
3. The colors are quite poor for high quality render, this is because KiCad uses special known colors.

What the user usually needs is an STEP model, which solves **1**. Then you use StepUp (FreeCad plug-in) to solve **2** and generate the equivalent surface (WRL). StepUp allows some color mapping, not always.

I implemented some mechanism to try to solve **3**. 
I will also look closer to what EasyEDA returns to find if the offsets can be fixed.
Both things needs to download the models again, using KiBot.

I tried your PCB, disabling the SOT-23-3 (which has a huge offset) and got a good render image with downloaded models and some heuristic to map the colors:

[image]

set-soft, 2023-01-31:
Ok, the above patch looks like a good start point. We can refine it.

closed 2023-01-31 by set-soft

ModischFabrications, 2023-01-31:
Wow, that's great! I'm impressed by how good the render looks. I don't fully grasp the problem, but I'm happy to use whatever is there. Thanks!

set-soft, 2023-02-01:
Hi @ModischFabrications !

Taking a look at your PCB I see Q1 and Q2 uses *SOT-23-3P_L2.9-W1.3-H1.0-LS2.4-P0.95.wrl*, but with a huge offset: (-861.570000 mm, 843.280000 mm)
This offset seems to be related to the EasyEDA footprint, but wrong for KiCad.
Looking at the EasyEDA information I see the PAD 1 is at  (4003.937, 3003.74) and the 3D model uses (608, -320) adjust.
So the PAD 1 with 3D adjust is (4003.937, 3003.74) - (608, -320) = (3395.937, 3323.74)
Translating it to mm is (3395.937, 3323.74)*0.254 = (862.57, 844.23)
The adjust you use is rotated 180º in the Z axis, this is (-x y z), so your adjust is (861.57, 843.28)
If we apply it to the (862.57, 844.23) we get (862.57, 844.23)-(861.57, 843.28) = (1, 0.95) 
Looking at the footprint in the PCB this is exactly the coordinate of PAD 1.

So far it looks logic, but for one detail: the footprint in the PCB doesn't match the EasyEDA (bizarre) footprint.
This adjust is needed for a footprint with PAD 1 at (4003.937, 3003.74)*0.254 = (1017, 762.95) mm and not (1, 0.95)

If I edit this component in pcbnew, go to the 3D model and replace the name by a 3D model downloaded from EasyEDA the adjust isn't needed.

Is this the same for the file you have? (D:/Kreativ/PCB/KiCad/lcsc.3dshapes/SOT-23-3P_L2.9-W1.3-H1.0-LS2.4-P0.95.wrl)

I can't run the easyeda.py code (missing dependencies and pip fails to install them) but looking at your PCB the footprint name is *lcsc:SOT-23-3_L2.9-W1.3-P1.90-LS2.4-BR*, and looking at the KiCad PCB file PAD 1 is at (1, 0.95), no need to adjust it.

set-soft, 2023-02-01:
Ok, so I removed the big offset and now Q1 and Q2 are there:

[image]

BTW: I got this render using the current `dev` branch of KiBot and Blender (included in the *ghcr.io/inti-cmnb/kicad6_auto_full:latest* docker image, Blender 3.4.1 isn't available for Debian 11.6 and the one packaged for Debian testing doesn't have support for the IA powered denoise filter, so you need much more than 10 samples to get this image. So I use Blender from the docker image even on my system).

The KiBot config I used is:

```yaml
# KiBot Blender export test 1
kibot:
  version: 1

import:
  - file: PCB2Blender_2_1

outputs:
  - name: '3d_export'
    comment: "Exports the PCB in blender format"
    type: blender_export
    disable_run_by_default: _PCB2Blender_2_1
    options:
      pcb3d: _PCB2Blender_2_1
      render_options:
        transparent_background: true
        samples: 10
      outputs:
        - type: blender
        - type: render
```

You should be able to reproduce it on CI/CD, or using docker images on Windows.

Note that currently you always get a green PCB, regardless of the color indicated in KiCad.

ModischFabrications, 2023-02-08:
Sorry, took me a while to work on my PCB again. I have noticed a few issues with the 3d previews as well, notably that Q1/2 and the big switch don't have any. 
[image]

I didn't care enough about it enough to investigate further, but your explanation sounds plausible. Do I summarize correctly that both definitions have huge offsets, but work together because they cancel each other out? There are also some other errors in the imported files, so this isn't surprising. It would be great to have a full PCB preview, but I can live with partial previews as long as everything else works. 

Do I understand correctly that the imported JLCPCB parts are wrong only in the 3D rendering and there is not much I can do about it during an import? Or is it something that I should've set up somewhere? It sound like something that the script I used could fix, but that might a misconception.

The render look great, I'm very impressed by the quality possible. I wasn't aware of the possibility to set color in KiCAD, so that's nothing I'm missing. My problem would be more that the specular highlight is very dominant, I prefer the "ugly, but visible" render preview.

set-soft, 2023-02-09:
Hi @ModischFabrications !

I don't know how exactly did you get the footprints you are using. I guess they were imported using easyeda2kicad scripts (which doesn't run in my system). Assuming this: you are right about the "cancellation"  effect, but the problem is that the 3D model gets the "uncancelled" coordinates. So you have to reset them manually.

The SW0 issue is different. The easyeda2kicad script was interpreting the *dissolve* parameter as *transparency*, they are related, but works differently. This made SW0 fully transparent, and hence invisible. I reported this and I think is fixed. KiBot doesn't have this problem.

About the light: KiBot creates a light automatically, using a simple heuristic. You can open the blender output (KiBot can generate it) using blender and play with the light, then tell KiBot where do you want to place the light. Similar for the camera, KiBot creates a default camera, asks Blender to make it focus the board and then moves it a little farther (to avoid problems with small boards and tall components), but you can tell KiBot where to place the camera. Of course this process takes time. Note that you don't need to install Blender, the kicad_auto_test (aka kicad_auto_full) docker images have a fully functional Blender, I use it because the Debian package for Blender is useless (IA filtering disabled because the packager failed to solve some issues with the Intel guy that maintains the IA filter)

ModischFabrications, 2023-02-09:
Yes, I imported them directly via script, I had no problems using it. After stumbling across this problem a few times, I think I'll pay more attention to using standard footprints, especially for these SOT-23 parts with standardized measurements. 

I personally won't make manual adjustments because I use KiBot as a single-shot "show me what you think this PCB is" type of integration, a quick social media post is the biggest usage these renders will ever see. My local installation should never see or care about it's visual representations, copying over the workflow files *should* be all that is needed for pretty outputs. I understand the use case and don't want to discourage anyone from doing that, but I treat the pipeline as a black box that spits out usable files and focus on the PCB itself. 

Thanks for fixing follow-up problems as well!

commit df085fa8 (2023-01-31): Added mechanism to download 3D models from EasyEDA

- Using LCSC codes

Closes #380