INTI-CMNB/KiBot #383: [FEATURE]SVG support for Logo

N0cynym, 2023-02-01:
**Is your feature request related to a problem? Please describe.**

`ERROR:In section 'bom_html' (bom): Only PNG images are supported for the logo (kibot - error.py:40)`

**Describe the solution you'd like**
Supporting vector graphics as logo in bill of materials.

**Additional context**
I do currently create one PCB documentation with `pdfunite` and export my generated xlsx spreadsheet with `soffice`.
Unfortunately just png is supported which is let the logo look not so nice compared to the rest of the `bom`.
It's just a request if this is possible.

N0cynym, 2023-02-01:
Ok I found this commit https://github.com/INTI-CMNB/KiBot/commit/1dea3970bc063340d17251ecc8a11e860fb51f31and will test to scale my logo.

N0cynym, 2023-02-01:
So I could not work out how the scaling is working correctly.
`logo_scale: -10` removes my png

Which logo is used in default?

set-soft, 2023-02-01:
Hi @N0cynym !

> So I could not work out how the scaling is working correctly. `logo_scale: -10` removes my png

What do you expect using a negative scale?
How does your logo look using the default scale (2)?
Too big? Use 1 or less than 1. 
Too small? Use more than 2.

But note that (as documented) not all spreadsheet software applies the scale. Also note that the size you get is dependent on the spreadsheet. So you must experiment it for the software you use.

> Which logo is used in default?

The KiBot logo, a 370x200 PNG embedded in the code.

About adding support for SVG: This can be done, but for this KiBot must convert the image to PNG, I don't see any advantage of doing this inside KiBot.

N0cynym, 2023-02-01:
If I use the scale >1 the logo is to big.

A negative scale will remove the logo.

Trying a positive scale <1 and >0 give me:

`ERROR:In section 'bom_xlsx' (bom): Option `logo_scale` must be a number not `string` (kibot - error.py:40)`

set-soft, 2023-02-01:
Are you using `.` as decimal separator?
I modified the XLSX test 7 like this:

```yaml
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: 'bom_internal'
    comment: "Bill of Materials in HTML format"
    type: bom
    dir: BoM
    options:
      format: XLSX
      xlsx:
        logo: docs/images/bot.png
        logo_scale: 0.1
        col_colors: false
```

And worked perfectly, I got it scaled to a good size, the original PNG looks huge.

set-soft, 2023-02-02:
Hi @N0cynym !

YAML numbers use a point as decimal separator, like in most programming languages this isn't affected by the user locale. Also note that using quotes you are converting it to a string.

About the default: is already documented:
- `logo_scale`: [number=2] Scaling factor for the logo. Note that this value isn't honored by all spreadsheet software.

Here the docs says the value must be a number, and its default is 2.

About SVG: For the HTML could be possible, but not for the XLSX. I don't see much advantage in doing it for the HTML, in particular knowing that SVGs are problematic.

What KiBot could do is convert the SVG to PNG, but this is really low priority. I always wanted to create an output to convert image formats, so it could help in the future

N0cynym, 2023-02-02:
Ok thank you very much! 👍 

I'll have a look into it if I see that a new feature is availible.

set-soft, 2024-03-30:
Ok, the above patch adds support for SVGs. Is a good starting point, please report any issue.

commit f1eaee6c (2024-03-30): [BoM][Added] Support for SVG format in the logos

Converting them to PNG

Closes #383
closed 2024-03-30 by set-soft