INTI-CMNB/KiBot #385: [BUG] Can't export fiducial marks in variant PaP

N0cynym, 2023-02-02:
**Describe the bug**
Based on your help in #373 I can now export my fiducial marks in default variant.

But this is not working in other variants.

**To Reproduce**
Please test following with my attached files:

```
kibot -s all position
kibot -g variant=variant_1 -s all position
kibot -g variant=variant_2 -s all position
```

**Expected behavior**
I want to export REF points (fiducial mark) as default in each variant.

[Test_PaP.zip](https://github.com/INTI-CMNB/KiBot/files/10566386/Test_PaP.zip)

**Results**

Default

```
### Module positions - created on Thu 02 Feb 2023 07:14:06  ###
### Printed by KiBot
## Unit = mm, Angle = deg.
## Side : top
# Ref    Val       Package                    PosX       PosY       Rot        Side
C1       100 nF    C_0805_2012Metric          153.5000   -86.5000   180.0000   top 
C2       100 nF    C_0805_2012Metric          153.5000   -91.5000   180.0000   top 
C3       100 uF    CP_EIA-6032-28_Kemet-C     143.5000   -96.0000   0.0000     top 
R1       100 Ohm   R_0805_2012Metric          136.7100   -86.3600   0.0000     top 
R2       100 Ohm   R_0805_2012Metric          136.7100   -91.4400   0.0000     top 
Ref1     Brdref    Fiducial_0.5mm_Mask1.5mm   133.0000   -82.0000   0.0000     top 
Ref2     Brdref    Fiducial_0.5mm_Mask1.5mm   157.0000   -99.0000   0.0000     top 
Ref3     Brdref    Fiducial_0.5mm_Mask1.5mm   133.0000   -99.0000   0.0000     top 
## End
```

Variant 1

```
### Module positions - created on Thu 02 Feb 2023 07:15:29  ###
### Printed by KiBot
## Unit = mm, Angle = deg.
## Side : top
# Ref   Val      Package             PosX       PosY       Rot        Side
C1      100 nF   C_0805_2012Metric   153.5000   -86.5000   180.0000   top 
C2      220 nF   C_0805_2012Metric   153.5000   -91.5000   180.0000   top 
R1      0 Ohm    R_0805_2012Metric   136.7100   -86.3600   0.0000     top 
R2      50 Ohm   R_0805_2012Metric   136.7100   -91.4400   0.0000     top 
## End
```

Variant 2

```
### Module positions - created on Thu 02 Feb 2023 07:16:52  ###
### Printed by KiBot
## Unit = mm, Angle = deg.
## Side : top
# Ref   Val      Package             PosX       PosY       Rot        Side
C1      100 nF   C_0805_2012Metric   153.5000   -86.5000   180.0000   top 
C2      10 nF    C_0805_2012Metric   153.5000   -91.5000   180.0000   top 
R1      50 Ohm   R_0805_2012Metric   136.7100   -86.3600   0.0000     top 
R2      0 Ohm    R_0805_2012Metric   136.7100   -91.4400   0.0000     top 
## End
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a `GitLab` CI/CD workflow
- Using docker image `kicad_auto:dev_k6`

closed 2023-02-02 by set-soft

N0cynym, 2023-02-03:
@set-soft sorry that I need to ask again:

I get the warning
```
WARNING:(W015) Unknown option `exclude_filter` (kibot - optionable.py:149)
```
if I try to export now following:

```
 - name: 'position'
    comment: "Pick and place file"
    type: position
    dir: .
    options:
      exclude_filter: _none
      include_virtual: true
      format: ASCII   # CSV or ASCII format
      units: millimeters  # millimeters or inches
      separate_files_for_front_and_back: true
```

Where should I add the `exclude_filter`?

set-soft, 2023-02-03:
Hi @N0cynym !
Is in the docs! If you search `exclude_filter` in the README.md the first match is in the `ibom` variant, the second in the `kibom` variant. Only the `bom` output has a local `exclude_filter`

N0cynym, 2023-02-03:
Sorry @set-soft  I tried serveral times without succsess.

To get how it works if I want to export my pick and place list with fiducial markers:

1. Means I need 2 outputs in KiBot config: `bom` and `position`
2. I need to add to the options of `bom` `exclude_filter: _none`
3. `kibot -g variant=default -s all -d Generated position`

Is this now correct?
If not, could you provide me a kibot config sample?

set-soft, 2023-02-06:
Hi @N0cynym !
Just a small example:
```yaml
kiplot:
  version: 1

variants:
  - name: 'default'
    comment: 'default'
    type: kibom
    exclude_filter: _none

  - name: 'position'
    comment: "Pick and place file"
    type: position
    dir: .
    options:
      include_virtual: true
      format: ASCII   # CSV or ASCII format
      units: millimeters  # millimeters or inches
      separate_files_for_front_and_back: true

  - name: 'bom_csv'
    comment: "Bill of Materials in CSV format"
    type: bom
    dir: .
    options:
      exclude_filter: _mechanical
      format: CSV  # XLSX, HTML or CSV
      csv:
         hide_pcb_info: true
         hide_stats_info: true
      columns:
            - Value
            - Quantity Per PCB
```

As you can see we are keeping all components in the variant, but we are applying the *_mechanical* exclude filter to the BoM, so we don't get the fiducials.

set-soft, 2023-02-08:
Hi @N0cynym !

> To understand the documentation correctly, exclude_filter: _mechanical means set exclude filter back to default?

Aprox., this is the default *KiBoM* exclude filter. That you needed to remove to generate a list of positions including mechanical stuff