INTI-CMNB/KiBot #386: [FEATURE] precise github actions tags

flaviut, 2023-02-03:
**Is your feature request related to a problem? Please describe.**

I'm using the `INTI-CMNB/KiBot@v2_k6` github action, and all the sudden my actions seem to be failing. This seems to be related to a recent update of the `v2_k6` action, and the new version of KiCad crashes (segfaults) on my project.

**Describe the solution you'd like**

It makes sense to have an evergreen tag like `v2_k6`, but it would also be nice to have a pinned version, like `v2.0.0_k6` that never changes and allows a working github actions workflow to work forever, regardless of any changes that may occur.

**Describe alternatives you've considered**

None

set-soft, 2023-02-03:
Hi @flaviut !
You already have `v2_k6_1_2_0` and `v2_k6_1_3_0` as explained in the [docs](https://github.com/INTI-CMNB/KiBot#github-actions-tags).

I'm changing the title to *more github actions tags* because 1.4.0 and 1.5.1 doesn't have tags

set-soft, 2023-02-03:
Done, added tags for 1.4.0 and 1.5.1

closed 2023-02-03 by set-soft

flaviut, 2023-02-03:
My mistake, I didn't see that part of the docs. Thank you!

commit 63158f63 (2023-02-03): [DOCs] Documented the new GHA tags

Closes #386