INTI-CMNB/KiBot #388: [BUG] When using variant, the erc_check should be performed per-variant

jnavila, 2023-02-07:
Fill as much as you can, remove the rest

**Describe the bug**
When using variants for alternative population of functions, some output pins of alternative functions are tagged in contention error by ERC , but that would not happen for each variant. 

**To Reproduce**
Create a schematic with 2 op-amps whose output are linked. Create variants which eliminates one of them. Enable erc in Kibot → you get an erc error.

**Expected behavior**
The erc should be performed for each variant, once each schematic has been processed. Reports of ERC errors should be per-variant. 

**Environment (please complete the following information):**
Where are you running KiBot: 
- Using a docker image? kicad5_auto:1.2.0
- Using a desktop system? Windows 11 and Linux ubuntu 22.04

jnavila, 2023-02-07:
Hi, 

My config files contain several variants, that's the pain point, and obviously, my request does not fall into the preflight checks, but more on variant management. I don't  think that the behavior needs to be checked, but more that an additional command be created once a variant has been created.

Your last remark is very interesting. I usually generate multiple outputs for a given variant, and the types of outputs are the same across variant. Right now, I managed to prevent copy-paste by using yaml references, but this feels a bit odd because I would expect this feature to be described in the kibot language, not by a the yaml level workaround. Anyway, let me know how the enhancement goes.

set-soft, 2023-02-07:
Hi @jnavila !
Don't use copy & paste!
Try doing what I told you: 
1. Create one config file with all your variants/filter, preflights and outputs. But just one output for all the variants, lets say: just one BoM output.
2. Include an sch_variant and pcb_variant, but with the run_by_default disabled.
3. Run KiBot once for each variant asking to generate the variants.
4. Run KiBot once for each variant asking to generate the outputs, passing the variant files.

You'll avoid repetition in the config file and get the preflights