INTI-CMNB/KiBot #390: [FEATURE] Check if update from schematic is needed

andersruneson, 2023-02-10:
This may already be implemented and I've missed it, for give me for my ignorance in that case.

**Is your feature request related to a problem? Please describe.**
Sometimes we release a board design where there are changes in the schematic that was not implemented in the PCB.

**Describe the solution you'd like**
It would be very nice if it's possible to implement a pre-flight check for if the PCB have pending changes from the Schematic. This may need KiCad API support to work out, I don't know.

**Describe alternatives you've considered**
Of course to check this manually, but we do our release by doing a tag in Gitlab so we may not even have KiCad installed on the same computer.

**Additional context**
Attached a screenshot for the Update PCB from Schematic dialog, when there is any updated needed the pre-flight may end the rest of the "run/build".
[image]

set-soft, 2023-02-13:
Hi @andersruneson !
I think what you want is the `check_pcb_parity` option from the `update_xml` preflight. As documented in the `run_drc`:
- Note that the KiCad 6 *Test for parity between PCB and schematic* option is not supported.
        If you need to check the parity use the `update_xml` preflight.
Let me know if it works for you.

andersruneson, 2023-02-13:
Ah, thanks. I think that is what I'm looking for. (still at KiCad5 but will eventually be able to upgrade)

set-soft, 2023-02-13:
Hi @andersruneson !
Note that KiCad 7.0.0 is being released, support for 5.x will be dropped in various tools, not on KiBot itself, but various tools are dropping it.

andersruneson, 2023-02-13:
We are working on a new component database tool, it's a lot of work to migrate for us but the new database interface in Ki7 looks nice for us.

What tools are you referring to?

set-soft, 2023-02-14:
KiKit is one example that tools that are dropping KiCad 5 support.