INTI-CMNB/KiBot #391: [FEATURE] KiRi integration

FrankvVeelen, 2023-02-12:
As an alternative to the kidiff tool (which works okay, but doesn't work for analyzing differences between more than a few commits/versions, I think it'd be cool if kibot could integrate [Kiri](https://github.com/leoheck/kiri), as it (at least for me) works a bit nicer with tools like jenkins (generating a html page on which i can then analyse differences between commits).

If you need help with implementing this and could give me a few pointers where i should start please let me know.

set-soft, 2023-02-13:
When I took a look at Kiri it used a mechanism for schematics that isn't good for KiBot integration.
I know Kiri is much better for interactive usage, but not for CI/CD.
I'll keep the request, but I'm not sure about it.

Nudelsalad, 2023-11-13:
Would be an awesome feature! Does Kiri even interfere with kibot if it is triggered separately? With the --no-server flag given the output could be maybe nicely relinked with navigate_results maybe?;) There are workarounds to get this working but up to your answer I guess there is nothing in the pot in the near future?

set-soft, 2023-11-14:
Hi @Nudelsalad !

I see @leoheck moved to *kicad-cli* for KiCad 7. Which removes the need for [PlotKiCadSch](https://github.com/jnavila/plotkicadsch), a real overkill since this is an OCaml tool.

Still using [KiCad-Diff](https://github.com/Gasman2014/KiCad-Diff) for the PCB, which is redundant tool since KiBot uses [KiDiff](https://github.com/INTI-CMNB/KiDiff).

Things are getting closer, but I don't have time right now.

leoheck, 2023-11-14:
Hi guys, KiCad-Diff is a small script, and for Kiri, you don't even need the GUI, I don't see any big problem of having it installed. 

Kiri still uses KiCad-Diff because kicad-cli for layouts takes a looong time to export each layer, and considering a big set of layers per commit and also a big set of commits, it takes a lot of time. 

But it is also possible to use kicad-cli for this. If KiCad-Diff is not installed (or maybe you have to pass a flag to Kiri), it will use kicad-cli instead. It may work ok if comparing 2 commits. 

For instance, here, we just compare 2 commits, removing previous runs, and displaying the commands kiri is running, without launching the browser.

```bash
kiri -r -D --kicad-cli -t 2 -l 
```

And this already takes 12.2s on my machine which is a good machine I guess (Dell XPS, i9)
The Board used in this example has 22 layers. Not all of them are important but they are/were present in the Kicad project.

set-soft, 2023-11-24:
Hi @leoheck !

I'll take a look at this. I'll post my questions here. First one: why `dos2unix` is a mandatory dependency?

leoheck, 2023-11-24:
Nope, I just ended up adding it as a workaround since most of the time people do things that they don't know, and they blame my tool for that and I don't always have time to debug everything. You could remove the lines using it. I can end up removing or adding a test to not use it if it is missing.

leoheck, 2023-11-27:
I am not sure how KiBot works or how it is used, but Kiri could just generate the images and the website without having to launch it... then the result (the .kriri folder,) can be exported, there is also an argument to make a pack of it. Inside, this folder it has a python script to launch the webserver using the previously generated files.

set-soft, 2023-11-27:
@leoheck : I was playing with KiRi and found a couple of issues:

1. When the files are in KiCad 5 format it tries to use *plotkicadsch*, even when it used *kicad-cli* for the other commits (forced by --kicad-cli). You can see it using the following [repo](https://github.com/INTI-CMNB/kibot_variants_arduprog.git)
2. In the above case when *plotkicadsch* fails (not installed) it tries to use *xdootool*. But if the process fails the keyboard is left changed to *us*. I suspect one of the problems is that KiRi thinks this is KiCad 5, but started KiCad 7.

It would be nice if *--kicad-cli* forces the use of *kicad-cli* even for KiCad 5 files.

Other questions: 

- I see you use shell scripts and Python, why not just Python? Is much more powerful and is even more portable, you won't need to worry for things like *readlink*, *sed*, *realpath*, etc. And your shell scripts are Bash specific!
- By using something like Python you'll be able to drastically reduce the "bin" pollution, you have 19 things in *bin*, KiRi can't be installed in a regular system, in particular: you even have an *env.sh* script, that could collide with anything.

leoheck, 2023-11-27:
> It would be nice if --kicad-cli forces the use of kicad-cli even for KiCad 5 files.

Yeah, definitely, I will give it a look. 

> I see you use shell scripts and Python, why not just Python? Is much more powerful and is even more portable, you won't need to worry for things like readlink, sed, realpath, etc. And your shell scripts are Bash specific!
By using something like Python you'll be able to drastically reduce the "bin" pollution, you have 19 things in bin, KiRi can't be installed in a regular system, in particular: you even have an env.sh script, that could collide with anything.

Yeah, python is awesome, but shell script is easier for me. Actually, this tool started as an experiment, but then later I adapted it to work on other systems and for others too. It just started as a bunch of commands that I put together in a script.   Having everything made in Python would be better definitely it would also make it possible to add on Kicad package manager (people would definitely love this). I thought many times to start changing it to Python only, but it would take a lot of time and effort for me to do it. Maybe if I have some spare time in the future. It is not a big and complex tool. The tool is simple, actually, it gets the commit list from git and iterates through them exporting the schematics/layers to images.

set-soft, 2023-11-27:
Hi @leoheck !

I tried the following toy test:
[t1.zip](https://github.com/INTI-CMNB/KiBot/files/13476301/t1.zip)

Running the *run_kiri.sh* script I got a nice list of commits and can browse the schematic changes. But the PCB looks the same for all the commits. Looking at the logs I see things like:

```
# 4/6 1db11f8 | 2023-11-27 11:12:45 | KiBot test | Reference

    Retrieving commits:
    - Source path: /home/salvador/0Data/0Test/#391/t1/.
    - Output path: /home/salvador/0Data/0Test/#391/t1/.kiri/1db11f8
...
    # Plotting schematics 1db11f8
...
   # Plotting layout _local_, F.Cu
```

Always plotting *_local_*, not the corresponding commit. Can you reproduce this bug?

leoheck, 2023-11-27:
Schematics are hard to spot the differences actually. 

But I can see them here... 

[image]
[image]
[image]

You can easy see the same with this script
```
schs=($(find .kiri/*/_KIRI_/sch -name "*.svg"))
for sch in ${schs[@]}; do
        # echo "Diffing ${schs[1]} and ${sch}"
        diff -q --report-identical-files ${schs[1]} ${sch}
done
```

Right now, when the local files have not commited changes, they are included automatically on the list of commits.

set-soft, 2023-11-27:
Hi @leoheck !

Schematics are ok, the problem is with the PCB.
The log says that is always plotting the layout for *_local_*, so you can't see any difference ;-)

set-soft, 2023-11-27:
I also found another problem: If I include the same sub-sheet twice KiRi generates:

```
light_control|light_control.kicad_sch||light_control|light_control
sub|sub.kicad_sch||sub2|light_control-sub2
sub|sub.kicad_sch||sub|light_control-sub
```

This is good, but when you look into the generated files only two of them are generated (*light_control.svg* and *sub.svg*). You may claim that *sub* and *sub2* are the same, but this isn't correct, changes in the annotation (reference numbers) can't be displayed if you don't have both rendered.

Here is the test case: [test.zip](https://github.com/INTI-CMNB/KiBot/files/13477847/test.zip)

Nudelsalad, 2023-11-27:
> About the link to navigate_results, you mention "workarounds", which ones do you mean?
> 
> From what I see you need to start the KiRi webserver and read the pages from it. As a workaround I tried `google-chrome --allow-file-access-from-files index.html` and got it working, but is quite specific to a particular case.

Hi @set-soft and @leoheck! Nice to see this discussion going on,
My workaround is not really a solution to these problems, but here is a little insight: 
I installed Kiri directly into the kibot docker container and let them both run their jobs in gitlab-ci. There, the output jobs are pushed to gitlab pages, and with separate links in the readme file or in the index.html I generated, you can access both outputs. The kibot link points to Test-navigate.html in the kibot/web directory and the kiri link points to index.html also in the kiri/web directory.
As you might see this is a dirty solution and the container is quite big with all my libs and the full kibot image ~3.2GB
My scope was to get it up and running in a short time, not to dig very deep.

Unfortunately I am having some performance issues with kiri output on large projects. I am not sure why it is really lagging on a large schematic. Thinking it must be an issue on the gitlab side, I also spun up the local kiri server and removed all the auxiliary project files. This left me with no clue as I have both a powerful server and a powerful desktop machine. But only on the schematic side!  On the other hand, looking at the PCB seems fine and smooth.
I'll mail you both the example :)

leoheck, 2023-11-27:
> This is good, but when you look into the generated files only two of them are generated (light_control.svg and sub.svg). You may claim that sub and sub2 are the same, but this isn't correct, changes in the annotation (reference numbers) can't be displayed if you don't have both rendered.

Yeah, handling multiple schematic pages is hard, especially the way Kicad stores these things. This behavior can be improved too. In the past, I thought that just one is enough since I was interested in circuit changes only. Perhaps there is a better way to display this.. maybe showing all the instances of each page... however only the designators are going to change, I feel no one will check these changes. I may be wrong tho.

leoheck, 2023-11-27:
> Unfortunately I am having some performance issues with kiri output on large projects.

Is this for a single comparison or for comparing all git history? You can run Kiri with `-t 3` to compare only the last 3 changes, including the local if it has chances.

You could also try `kiri-docker` if you are using linux or a linux shell.

Check it here: https://github.com/leoheck/kiri-docker
There is this kiri command that is a wrapper to the normal kiri command.
It mount the local repo inside the docker, launch kiri there, and displays the result in the local browser.
This could work for you, maybe.

set-soft, 2023-11-28:
> Testing here, I saw no changes as you are saying. But after cleaning the cache, (flag -r) it started to show the changes. Could you try that too? The local changes should not need this, I will have to check this behavior too, thanks for finding this.

Tried it and got the same. Did you try using the `run_kiri.sh` script? I suspect the issue could be related to `--kicad-cli`

set-soft, 2023-11-28:
> Unfortunately I am having some performance issues with kiri output on large projects. I am not sure why it is really lagging on a large schematic. Thinking it must be an issue on the gitlab side, I [...]

I got your example, do you mean that doing zoom and pan in the browser is really slow? 

I noticed that in my machine at work (not really fast, 3rd gen i7). But if you look at the SVGs they are upto 3.6 MiB. The script is comparing 2 of them. Perhaps the method used is not the best (is this the feColorMatrix stuff @leoheck ), or just the pan & zoom class (svg-pan-zoom). But yes, is slow.

leoheck, 2023-11-28:
Hi guys (@set-soft @Nudelsalad), I tried to summarize the issues here, https://github.com/leoheck/kiri/issues/104
Do you mind checking if it is missing something or if I misunderstood anything?

======

@set-soft I tried your script once (and again now), the local layers are not being displayed for some reason (the image was just red), as you observed. However, they were generated. I also could not see the differences in the layers of other commits with the browser.

Found the issue, the layouts of the `_local_`  have different names. the names are not being normalized. They should be  normalized to this format `layer-00.svg`

[image]

leoheck, 2023-11-28:
[image]

@set-soft do you have this rename here? I don't have it right now, since I reinstalled the OS, this may the the issue on my side. I will check if I can skip it to use the normal rm tool... since it is more spread and maybe reliable. this causes issues on other distros since not always people have the rename that is the perl script since there are others available.

[image]

leoheck, 2023-11-28:
for some reason, the _local_ changes produce an image with a different kicad version. but there are no local changes made by the user. 
[image]

However, the changes are not being displayed yet.

leoheck, 2023-11-28:
@set-soft I could find an issue when generating pcb layers with kicad-cli it is fixed in the repository already, could you check if it works for you?

leoheck, 2023-11-28:
[image]

kicad-cli can be used with kicad 5 layout but not with schematics since it plost only kicad_sch and not sch. This is a poor decision they made. Poor because they have the kicad parser and the kicad exported. They could have just used it.

But I will try to fix the issue with the keyboard changing when xdotool fails.

leoheck, 2023-11-29:
@Nudelsalad , I found out that your experience can be much better optimizing the svg files... 
You have a lot of images.. and this may take some time. It took some time on my i9. 
Just copy and paste this inside the .kiri folder of your project, this will optimize all the svgs of the schematic files.

```bash
optimize_svg()
{
	svgs=($(find */_KIRI_/sch/ -name "*.svg"))
	total="${#svgs[@]}"
	i=0
	for svg in ${svgs[@]}; do
		i=$((i+1))
		orig_size=$(du -sh ${svg}  | tr "\t" " " | cut -d" " -f1)
		cmd="svgo -q --final-newline -i ${svg}"
		eval ${cmd}
		new_size=$(du -sh ${svg}  | tr "\t" " " | cut -d" " -f1)
		printf "%3d/%d | %4s > %4s | %s\n" ${i} ${total} "${orig_size}" "${new_size}" "${cmd}"
	done
}

optimize_svg
```

The output is something like this... 
[image]

I could add this to kiri and a flag to enable it (or maybe to disable it) when not needed. 

You are going to need to have the svgo installed. I installed it on my Ubuntu using this, https://snapcraft.io/install/svgo/ubuntu

leoheck, 2023-11-29:
@set-soft dos2unix is definitely not needed anymore. thanks for pointing this out.

leoheck, 2023-11-29:
@Nudelsalad I added this optimization to the kiri, directly. You can test it with the latest version of the repo. It is possible to disable it with the flag (uppercase) `-R`. However, the `svgo` tool must be available in the PATH otherwise Kiri will skip this step.

set-soft, 2023-11-29:
> Hi guys (@set-soft @Nudelsalad), I tried to summarize the issues here, [leoheck/kiri#104](https://github.com/leoheck/kiri/issues/104) Do you mind checking if it is missing something or if I misunderstood anything?

Very good @leoheck ! I'm updating KiRi with your last patches.

> @set-soft do you have this rename here? 

I have the *rename* tool installed.

> @set-soft I could find an issue when generating pcb layers with kicad-cli it is fixed in the repository already, could you check if it works for you?

I'm running KiRi right now and I can see it generating images for each hash, not just *_local_*, yes, fixed!

> @Nudelsalad , I found out that your experience can be much better optimizing the svg files...

Do you know which optimizations are applied? Can you look for a light weight equivalent? I want to avoid Node.js tools, they are slow and bloated.

I'm experimenting with the following approach to draft an integration:
- KiDiff generates the SVG files, it can handle SCH and PCB conversion using KiAuto
- KiBot chooses which files to use, it can handle *git* repos very well, including submodules.
- Then I copy the KiRi engine (kiri_server, kiri.js, kiri.css, etc.) to the destination folder.

I already modified KiDiff and KiBot is generating almost all the files needed by KiRi. Will post results here.

leoheck, 2023-11-29:
> Do you know which optimizations are applied?

I do not know this. 

I just used the first one I could find and it worked really well. So I pushed it to check if @Nudelsalad could tell us if this improves the overall experience for him. 

Also, the produced files are normalized, so I can compare them against the others to see if they have meaningful changes and this may allow me to hide layers (maybe schematics too) that do not need to be compared at that time making the review process faster.

We can use a different one, for sure, right now I just don't know any other. 

svgo is doing its normal optimization which I don't know what it is. But I also asked it to reduce the precision of the values and asked for a nested svg.

I put this behavior in a separate file, you can also check and experiment with modifying it in this file here  ttps://github.com/leoheck/kiri/blob/main/bin/optimize_svg_for_web

> I'm experimenting with the following approach to draft an integration:
KiDiff generates the SVG files, it can handle SCH and PCB conversion using KiAuto
KiBot chooses which files to use, it can handle git repos very well, including submodules.
Then I copy the KiRi engine (kiri_server, kiri.js, kiri.css, etc.) to the destination folder.
I already modified KiDiff and KiBot is generating almost all the files needed by KiRi. Will post the results here.

Thats nice. Yes, please share the results here, I am eager to see it. I realized recently that Kiri can show and compare the commits, and the local changes too, but not the staging area.

leoheck, 2023-11-29:
I added an option to compare 2 commits, using this syntax `kiri -c HASH..HASH` kind of what git does. the HEAD syntax works too and also the LOCAL, local, _local_. This may help to have some improvements since it won't need to compare the whole list of commits.

Nudelsalad, 2023-11-29:
Hey guys,
I just tried out the newest kiri updates and somehow the issue of pan & zoom is gone, the kiri website is now working flawlessly but somehow the size had not been significantly reduced. I guess thats really due to the project size. 
I noticed two things:

1.  Installing all npm dependencies with svgo consumes nearly 600MiB in my container
2. Creating the artifacts of the output with kiri takes for a limit of the 5 last commits on my server 13 minutes which is quite longggg. But I guess this is due to kidiff sadly no verbose output is possible there. In the end it works, but its kind of odd user experience and blocking my runner.

Second thing i tested was another svg optimizer tool there are few out there, two seem to be beneficial for here:

1. [scour](https://github.com/scour-project/scour) (pip package) easy to install just the need to add it to your requirements.txt and lets go
2. [svgcleaner](https://github.com/RazrFalcon/svgcleaner) which seems to be more advanced than scour and it was able to reduce the size of the already reduced image with svgo to another 26.37% and it seems to be easy integratable into your already existing script 

```
optimize_svg()
{
	svgs=($(find */_KIRI_/sch/ -name "*.svg"))
	total="${#svgs[@]}"
	i=0
	for svg in ${svgs[@]}; do
		i=$((i+1))
		orig_size=$(du -sh ${svg}  | tr "\t" " " | cut -d" " -f1)
		cmd="svgcleaner ${svg} ${svg}"
		eval ${cmd}
		new_size=$(du -sh ${svg}  | tr "\t" " " | cut -d" " -f1)
		printf "%3d/%d | %4s > %4s | %s\n" ${i} ${total} "${orig_size}" "${new_size}" "${cmd}"
	done
}

optimize_svg
```

Here the output of my job:
[image]

Here the output of svgcleaner after calling

`svgcleaner 0603_Dashpanel.svg 0603_Dashpanel.svg`

[image]

@set-soft you mentioned that you are about to integrate the kiri files and mentioned kiri-server. Is kiri-server really needed in kibot? As I am continuing to dream that navigate_results just links the web/index.html of the kiri_out to maybe the index.html starting page of kibot with maybe a third Option next to PCB, Schematic and maybe Review/Revisions

leoheck, 2023-11-29:
> the issue of pan & zoom is gone

Hm, I increased the speed of the zoom a bit. This could have done the magic.

set-soft, 2023-11-29:
> 1. Installing all npm dependencies with svgo consumes nearly 600MiB in my container

Normal for a Node.js tool, this is why I asked for an alternative.

> 2. Creating the artifacts of the output with kiri takes for a limit of the 5 last commits on my server 13 minutes which is quite longggg. But I guess this is due to kidiff sadly no verbose output is possible there. In the end it works, but its kind of odd user experience and blocking my runner.

I'm testing the KiRi integration using a toy repo (the one I posted here). Using KiBot is much faster than using `kicad-cli`, but is just a toy project.

> Second thing i tested was another svg optimizer tool there are few out there, two seem to be beneficial for here:
> 
> 1. [scour](https://github.com/scour-project/scour) (pip package) easy to install just the need to add it to your requirements.txt and lets go

Looks very light-weight

> @set-soft you mentioned that you are about to integrate the kiri files and mentioned kiri-server. Is kiri-server really needed in kibot? As I am continuing to dream that navigate_results just links [...]

If the *navigate_results* output is deployed to a web server you don't need *kiri_server*.
If you generate the artifacts, download them and try to see the *navigate_results* the *kiri.js* script will fail, unless you use the above mentioned workaround (`google-chrome --allow-file-access-from-files index.html`)

set-soft, 2023-11-29:
Ok, here is a first working result, not committed:  [res1.zip](https://github.com/INTI-CMNB/KiBot/files/13502662/res1.zip)

This is KiBot generating the files needed by KiRi engine. 

I added a very bizarre situation in the schematic to test how KiBot and KiRi reacts. I included 3 sub-sheets, one is shared, another has a repeated name. 

I see big performance issues, it looks like the browser is refreshing stuff all the time, i.e. scroll bars flickering. I'm not optimizing the SVGs yet.

Nudelsalad, 2023-11-29:
> I see big performance issues, it looks like the browser is refreshing stuff all the time, i.e. scroll bars flickering. I'm not optimizing the SVGs yet

@set-soft I also noticed weird flickering of the scroll bar sometimes however it is working fine for me right now with my last project and optimized svgs but i will apply this to more projects and report if it appears again

leoheck, 2023-11-29:
Both `scour` and `svgcleaner` work nice, 

this is scour working,
[image]

using the maximum scrub as it is reported by their github
```bash
cmd="scour -i ${svg} -o output.svg --enable-viewboxing --enable-id-stripping --enable-comment-stripping --shorten-ids --indent=none"
eval "${cmd}"
mv output.svg "${svg}"
```

leoheck, 2023-11-29:
The svg optimizer could optimize in a way that I can compare 2 to ignore pages/layers without changes.. this could make the review process easier. There is always something inside the file, that is not visible, that prevents me for hiding all unchanged layers.

set-soft, 2023-11-30:
I tried *scour* and even when it reduces the file sizes, the real issues still there. I thought it was something related to the complexity of the files, but I can load the SVGs in the browser and they are displayed fast. I suspect the problem is in the pan & zoom class or, most probably, in the way colors are computed (using a complex SVG filter).

One thing that I would like to see is the possibility of diff colors customization. I know it sounds cosmetic, but all the colored diff tools I know for code use added == green and removed == red, KiRi is using cyan for added, which is somehow confusing to me.

What is the *txt_icon*? I'm not sure if I understand it right: Only the commits where the SCH and/or PCB changed are displayed? If this is the case: What is the TXT icon for?

Looking at the JS code I see the layer colors are hardcoded and the colors in the initial *index.html* aren't really important. It would be nice if we could allow color changes, so people can use colors that are easy to recognize. 

The list of commits has a fixed width that in my system isn't good, is this common? Can be automatically adjusted?
[image]

leoheck, 2023-11-30:
> I tried scour and even when it reduces the file sizes, the real issues still there. I thought it was something related to the complexity of the files, but I can load the SVGs in the browser and they [...]

I felt the same when I ran scour. With the svgcleaner and svgo it was a bit better.

> One thing that I would like to see is the possibility of diff colors customization. I know it sounds cosmetic, but all the colored diff tools I know for code use added == green and removed == red, KiRi is using cyan for added, which is somehow confusing to me.

Definitely. It is not just cosmetic, it is important for me too. But I am not sure how to handle those color matrixes. They are quite complex to work with. I would appreciate it if you could try to change them a bit. I don't like the current colors either. 

> What is the txt_icon? I'm not sure if I understand it right: Only the commits where the SCH and/or PCB changed are displayed? If this is the case: What is the TXT icon for?

Text icons are changes that are not in the *sch and *kicad_pcb files, but this was tried a long time ago, today I feel they don't help much when comparing the files. 

> Looking at the JS code I see the layer colors are hardcoded and the colors in the initial index.html aren't really important. It would be nice if we could allow color changes, so people can use colors that are easy to recognize.

I just have to understand how to make it work.

> The list of commits has a fixed width that in my system isn't good, is this common? Can be automatically adjusted?

Sure, but maybe something broke in your browser. I felt this was not behaving like this before. But it needs a bit of love. 

I will keep notes on this, but I can't guarantee I can fix or improve these things otherwise I would have done so in the past since I agree with everything you are commenting here.

leoheck, 2023-12-01:
This seems interesting, I will give it a look at the weekend, cool? Meanwhile, do you have a picture showing how this looks on Kibot? I don't have Kibot installed yet neither do I know how it works and what it produces.

set-soft, 2023-12-01:
Hi @leoheck !

I noticed that the list of layers and sheets embedded in the *index.html* are useless because the JS code replaces them as soon as we load the page (and/or switch to PCB).

This leads to:
1. Bloated *index.html*
2. Code repetition. The JS logic must be also in the generation scripts.

So I tried to avoid filling them. I soon realized the JS assumes the lists are always there so I had to add a few conditionals to avoid accessing data that is not there. Take a look at this patch: 4fb3870d12a0aad00f10442bd59a7013f7871865

leoheck, 2023-12-01:
Ah, there is also a js code that hides equal layers, in this branch... it does not remove pages without changes all the time since values may change a bit between the plots.

leoheck, 2023-12-01:
Is this the new color theme, right?
[image]

set-soft, 2023-12-01:
> Is this the new color theme, right?

Yes, this is the colors used by KiCad 6 and 7 by default

set-soft, 2023-12-01:
> Disabling the initial layers/pages list works, I pushed this too to the new branch. However, I feel the whole navigation is getting slower. Don't you feel the same?

I see some issues, but I think they are related to the same thing that generates the flicker. I'm investigating the color matrix right now, will try to figure out the flicker latter, maybe next week.

set-soft, 2023-12-01:
> Definitely. It is not just cosmetic, it is important for me too. But I am not sure how to handle those color matrixes. They are quite complex to work with. I would appreciate it if you could try to [...]

I was playing with the matrices and the best I got is:

[image]

I can't get a better color for unchanged. The problem is that the pan & zoom is just overlapping both SVGs, so we can just play with the RGBA. In this case I changed one image to be pure green and the other pure red, both with 50% alpha. This leads to green on the newer, red in the older and orange? when both added. It looks like the addition is giving more weight to something, giving more red component, otherwise we should be getting yellow for the red + green.

To get this I used:

```xml
           <svg id="img-1">
              <defs>
                  <filter id="filter-1">
                      <feColorMatrix in=SourceGraphic type="matrix"
                      values="1.0  0.0  0.0  0.0  0.0
                              0.0  1.0  0.0  1.0  0.0
                              0.0  0.0  1.0  0.0  0.0
                              0.0  0.0  0.0  0.5  0.0">
                  </filter>
                  <filter id="filter-12">
                      <feColorMatrix in=SourceGraphic type="matrix"
                      values="-1.0   0.0   0.0  0.0  1.0
                               0.0  -1.0   0.0  1.0  1.0
                               0.0   0.0  -1.0  0.0  1.0
                               0.0   0.0   0.0  0.6  0.0">
                  </filter>
              </defs>
              <image id="diff-xlink-1" height="100%" width="100%" filter="url(#filter-1)"
                  onerror="this.onerror=null; imgError(this);"
                  href="${src1}" xlink:href="${src1}"/>
          </svg>
          <svg id="img-2">
              <defs>
                  <filter id="filter-2">
                      <feColorMatrix in=SourceGraphic type="matrix"
                      values="1.0  0.0  0.0  1.0  0.0
                              0.0  1.0  0.0  0.0  0.0
                              0.0  0.0  1.0  0.0  0.0
                              0.0  0.0  0.0  0.5  0.0">
                  </filter>
                  <filter id="filter-22">
                      <feColorMatrix in=SourceGraphic type="matrix"
                      values="-1.0   0.0   0.0  1.0  1.0
                               0.0  -1.0   0.0  0.0  1.0
                               0.0   0.0  -1.0  0.0  1.0
                               0.0   0.0   0.0  0.6  0.0">
                  </filter>
              </defs>
              <image id="diff-xlink-2" height="100%" width="100%" filter="url(#filter-2)"
                  onerror="this.onerror=null; imgError(this);"
                  href="${src2}" xlink:href="${src2}"/>
          </svg>
```

I also changed the background to pure black to boost the contrast.

Take a look at this patch: e06089cfcd8edabc3741e2c4e6533612527ffc1f

set-soft, 2023-12-11:
Hi @leoheck !
I applied a patch to the kiri.js to avoid reloading the list of pages every time we select a page. Also to avoid an event on-click plus another on-change. Take a look at a7ad89c8d7b2e4a13787d477474fe6956c504c28

About the canvas color: It should be configurable, but I want to solve some issues first

set-soft, 2023-12-11:
Hi @leoheck !
The following patch solves a big issue with sub-sheets: e35fb669a2c3699d3d63af1af54e9b0d4dd5828a
Using it we can recycle the same sheet in the project, or use sub-sheets with the same name (but different path).
The drawback: the names of the generated files are different.
But is an easy fix: just don't rename the files generated by KiCad! They have the sheet hierarchy because this allows pages reuse.

set-soft, 2023-12-11:
This is another important patch: 0cad908deebd987a345a403795adf42e91184da5
It avoids fetching a sheet page that isn't available in a commit, in this case we use the *blank.svg* image.

leoheck, 2023-12-11:
Oh, you are making good progress here, thank you, I will test these last updates. I am going to test this later today since I have too many things todo today. Cool?

Nudelsalad, 2023-12-11:
Very cool! 
Is it then possible to ignore this layer in the sheet viewer? As only those with changes are interesting to review

set-soft, 2023-12-11:
> Very cool! Is it then possible to ignore this layer in the sheet viewer? As only those with changes are interesting to review

Well I think the answer is no. You usually want to know if some layer/sheet was added/removed. If you see a sheet/page all in red, then it was removed, and if is all green it was added.

leoheck, 2023-12-11:
Yeah, but if we can detect this somehow, we could make the keyboard shortcut to navigate through layers/pages to jump over then ones that didn't change.  but they could be still there (with a different font color, maybe) if the user wants to click to check.

set-soft, 2023-12-11:
> Yeah, but if we can detect this somehow, we could make the keyboard shortcut to navigate through layers/pages to jump over then ones that didn't change. but they could be still there (with a different font color, maybe) if the user wants to click to check.

The 901c76b189d918ce6f6a16f9df6b8a8c1acb2fea implements the color part.

[image]
[image]

Nudelsalad, 2023-12-11:
Just great @set-soft!
I would like to add another thought of mine. 
I understand that it is quite hard to handle everything with svgs and that there is not much room for further expansion. Also, I do not want to make more feature RQs while you are in the process of integrating this into Kibot, this is just an idea:

Wouldn't it be cool if the zones were dimmed or completely hidden?
I do not know how you process your files, but I just looked up kicad-cli features for an export option of svgs without filled zones and this feature is not implemented yet - but would be essential I guess.

[image]

In my opinion, the presentation of the new changes is a bit confusing as zones can overlap tracks and vice versa.

As I am not fully involved in the development of your tools, is this possible? Is it worth creating a feature rq for kicad-cli export? Do we share the same opinion? Does it even depend on kicad-cli?

I also have in mind that this is the thread for the integration of kiri into kibot and nothing else, but having two experts working on the same codebase seemed to me a justifiable reason. 

Thanks in advance!

P.S.: As an example, the cadlab.io guys seem to have managed this somehow, although it seems not to be a comparison between svg files, but more a wrapper for whole projects while on a net and change basis the changes seem to be displayed [Link](https://cadlab.io/project/27597/cf6f186abd9961c086e162e4b2411899d16b0cde/circuit/Q29yZS1Cb2FyZC9Db3JlLUJvYXJkLmtpY2FkX3BjYg%3D%3D/diff/7e2f92e613d4e3b820dbb0f8c0400d9ea142389b)

set-soft, 2023-12-12:
Hi @leoheck !
The following patch is interesting, but a little bit disruptive: ad3f8bb7d71f12871a4f23bfae3279563c6baadd
I moved the commits HTML generation to the JavaScript. The goal is to have an index.html that is the same for every project. In order to achieve this I'm creating a file called *commits* containing the commits. The format is documented in the JS, is:

```
HASH|DATE|AUTHOR|DESCRIPTION|SCH_CHANGED|PCB_CHANGED
```

And here is an example:

```
_local_|2023-12-12|Salvador E. Tropea|Local changes not committed|True|True
ab61573|2023-11-27|KiBot test|Cambio 2|True|True
3b3741c|2023-11-27|KiBot test|Cambio 1|True|True
1db11f8|2023-11-27|KiBot test|Reference|True|True
```

The only thing that remains to be moved is the sch/pcb info, this will be the next step. Doing this the *index.html* file will be independent of the project. This will also simplify your shell script a lot.

set-soft, 2023-12-12:
Hi @Nudelsalad !

> Wouldn't it be cool if the zones were dimmed or completely hidden?

In order to keep this idea please open an issue in the [KiDiff](https://github.com/INTI-CMNB/KiDiff)  project. This is the underlying tool.

set-soft, 2023-12-12:
> The only thing that remains to be moved is the sch/pcb info, this will be the next step. Doing this the _index.html_ file will be independent of the project. This will also simplify your shell script a lot.

And now is done: d073b155ef5f4ce43fe21aa259d2c245d6cad409
The *index.html* is always the same, the JS loads the data from the generated files and adjusts the HTML.

set-soft, 2023-12-12:
More for @leoheck : The following patch 6a380e3ddbb071e89fdc8b037725159ad4e25920 moves all the style stuff from the HTML to the CSS.
The main objective is to make the HTML easier to maintain. It could potentially allow some degree of customization using the CSS.

leoheck, 2023-12-12:
@set-soft what if you make a fork of kiri? You could make a for of it, and then a PR to me. It would help me testing and merge all your good stuff at once?

set-soft, 2023-12-12:
> @set-soft what if you make a fork of kiri? You could make a fork of it, and then a PR to me even if it is work in progress. It would help me test and merge all your good stuff at once.

Is a good idea, I'll try it when the functionality is stabilized. I'm trying to better separate:

1. Data (generated files)
2. Layout (the HTML)
3. Style (CSS)

[12 more replies]

closed 2024-01-17 by set-soft