INTI-CMNB/KiBot #394: [BUG] Infinite loop while finding config file

kingelilol, 2023-02-21:
**Describe the bug**
Kibot hangs if there is a dot inside the configfile path.

**To Reproduce**
Move a config file into a dot folder or start the config name with a dot.

**Environment**
Inside the docker kicad_auto:latest
KiBot 1.6.0

**Additional context**
The folder name `.kibot` witch stores the config files is the issue.
It looks like the code retries until there is no dot inside the absolute path left.

```
DEBUG:Using configuration file: `/mnt/tmp/pi_can2spi/.kibot/config.pdfs.yaml` (kibot - __main__.py:145)
^CTraceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.0', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 365, in main
    GS.set_sch(solve_schematic('.', args.schematic, args.board_file, plot_config))
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 708, in solve_schematic
    config = os.path.splitext(config)[0]
  File "/usr/lib/python3.9/posixpath.py", line 125, in splitext
    return genericpath._splitext(p, sep, None, extsep)
  File "/usr/lib/python3.9/genericpath.py", line 121, in _splitext
    def _splitext(p, sep, altsep, extsep):
KeyboardInterrupt
```

set-soft, 2023-02-22:
Reproducing it is far more complex than what you describe.
You  must:

- Have more than one schematic
- Not specify the name of the schematic
- Don't have an schematic corresponding to the PCB
- Have the dot in the path of the config, but not at the begginning ( `-c .xxxx.kibot.yaml` works, `-c .yyy/xxxx.kibot.yaml`  also)
 
Only in this case the problem is exposed.

Now should work

kingelilol, 2023-02-27:
Thanks for fixing it.
And sorry, I will lock closer next time to make it easier to reproduce.

commit fe26f1ab (2023-02-22): Problems to detect the SCH under very particular constraints

Fixes #394