INTI-CMNB/KiBot #396: [BUG]SegFault on pcb_print

arthursaunier, 2023-02-24:
**Describe the bug**
I get a segmentation fault while plotting my pcb with pcb_print.

**To Reproduce**
```
kibot -d ../Fab/test -vvv basic_pcb_print_pdf
```
And you might need a weirdly shaped copper zone on multiple layers to reproduce this problem. 

**Environment:**
Where are you running KiBot:
- Using a desktop system? Which OS, distro and version?
Windows11 with WSL2, with Ubuntu 22.04.1 LTS

**Additional context**
pcb_print fails with a segmentation fault when I try to plot it. It started happening after I added a new copper zone. If I delete that copper zone, I don't get the segmentation fault at all. This quite a big project, 4 layer PCB, with around 400 components and 1200 vias and all. 

You can see the copper zone below:
[image]
[image]
[image]

I'm sorry, I can't add the files of my project here, as this is a work-related design. If you need it to reproduce the bug, I can send it to you privately. 

In this zip file, the debug log from the cmd and my kibot.yaml file:
[bug kibot.zip](https://github.com/INTI-CMNB/KiBot/files/10826068/bug.kibot.zip)

set-soft, 2023-02-24:
Hi @arthursaunier !
Can you create a small PCB example with such a copper area?
I need to reproduce it.
The problem looks like an error in KiCad itself, since you are getting:

```
16:31:07: Warning: Overflow KiROUND converting value 44548886938008567152640000.000000 to i
```

This log isn't from KiBot and an overflow in KiCad is something I never saw before.

Python code can't SIGSEGV by itself, so I think the problem is in the KiCad plotter. But I need a case where it happens.

If creating a PCB with the problem gets difficult you can just do a *binary search*: take a copy of your PCB and remove half of the stuff, if the problem can be reproduced remove half of the remaining, and so on. If the problem can't be reproduced just undo and remove 1/4. Until the remaing PCB is something you can publicly share.

arthursaunier, 2023-02-24:
Hi! 

I already observed this overflow error before on the same project, but it did not lead to a seg fault. I tried to check online but could not find any info about this. 

You should be able to reproduce the problem with the following project (I was able to).

[test.zip](https://github.com/INTI-CMNB/KiBot/files/10826923/test.zip)

Thank you for your work!

set-soft, 2023-02-25:
Thanks @arthursaunier ! 

I was able to reproduce the segment violation using KiCad 6.0.11 and 7.0.0.

I'll try to find the source. It may be some abuse done by KiBot while moving stuff to then plot the pads alone, or may be just KiCad. I'll try to narrow the problem.

set-soft, 2023-02-25:
As a temporal workaround just add:

```
    colored_pads: false
    colored_vias: false
```

to the `pcb_print` options. This disables the problematic code.

GarethID, 2023-02-27:
Hi guys, I've also had this same issue (but haven't had the time to report yet).
I managed to work around it by ensuring all zones are named (uniquely), and that they don't conflict with one another (by setting the priorities all correctly). Since doing that, I haven't had any issues on complex multi-layer zoned boards.

set-soft, 2023-02-27:
Hi @GarethID !
Thanks for sharing your experience.
I managed to reduce @arthursaunier example to a small test case that crashes 7.0.0, at least on my system. 

[test.zip](https://github.com/INTI-CMNB/KiBot/files/10838571/test.zip)

I used v7 because KiCad people isn't paying much attention to v6 bug reports.

This example has just 2 zones:
- "F.GND_plane" priority 0
- "GND_chassis" priority 6

The crash can be consistently reproduced changed their names and priorities.

I suspect the problem is related to the use of already freed memory, so this is very dependent on the memory layout.

I asked in the [KiCad forum](https://forum.kicad.info/t/python-crash-during-zones-fill-after-unfilling-them/40532) and Jon Evans (KiCad developer) encouraged me to report it as a bug (nothing strange in what KiBot does, not at least at first sight). So this is currently under evaluation as [bug 14090](https://gitlab.com/kicad/code/kicad/-/issues/14090).

We'll see the opinion of other KiCad developers

arthursaunier, 2023-02-27:
Hi all, 
@GarethID I tried your fix, changing the names and the priorities of all my zones and checking everything. I still get the error sadly.
@set-soft your workaround works fine for me. I'll follow the bug you reported.
Thanks very much

set-soft, 2023-02-27:
Ok, I found the problem.

Looking at KiCad's code I found I must call `BuildConnectivity()` after a zone fill.

commit 6fd97254 (2023-02-27): [PCB Print][Fixed] KiCad crashing on some complex filled zones

- We must rebuild the conectivity data after a zone fill

Fixes #396
closed 2023-02-27 by set-soft