INTI-CMNB/KiBot #397: KiCAD 5/6 mismatch ?  The design seems to be KiCAD6 and works with KiCAD6.

mdeweerd, 2023-03-01:
While trying to build a test case to emulate the move from KiCAD 6 to KiCAD 7 (moving from KiCAD 5 to 6 first), I stumble on this error (last line):

```WARNING:(W058) Missing KiCad main config file /root/.config/kicad/6.0/kicad_common.json (eeschema_do - misc.py:181) (kibot - kiplot.py:134)
WARNING:(W058) Using old format files is not recommended. Convert them first. (eeschema_do.kiauto.file_util - file_util.py:281) (kibot - kiplot.py:134)
WARNING:(W058) Old file format detected, please convert it to KiCad 6 if experimenting problems (eeschema_do - interposer.py:632) (kibot - kiplot.py:134)
- 'Unique report with characteristics, schematic, etc.' (report) [report]
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:329)
WARNING:(W010) Unable to find KiCad 3D models (kibot - config.py:403)
ERROR:At line 1 of `/usr/share/kicad/symbols/Graphic.kicad_sym`: Mixing KiCad 5 and KiCad 6 files is not allowed (kibot - kiplot.py:257)
ERROR:Line content: `(kicad_symbol_lib (version 20211014) (generator kicad_symbol_editor)` (kibot - kiplot.py:258)
```

This is https://github.com/mdeweerd/kibot-testcases/tree/OLD_FILE_FORMAT_ISSUE/designs/tc6 (tag OLD_FILE_FORMAT_ISSUE) 

I copied and upgraded the KiCAD v5 testcase - which I loaded and saved in KiCAD6.  I also loaded and did ERC using KiCAD6 from the docker image.

To run this you can just do this:
```shell
git clone --branch OLD_FILE_FORMAT_ISSUE git@github.com:mdeweerd/kbot-testcases.git
docker compose run --rm dev_k6
```

mdeweerd, 2023-03-01:
```
Using command line value `header` for global option `variant`
ERROR:No SCH file found (*.sch), use -e to specify one. (kibot.gs - gs.py:377)
```

I explicitly specified the path to the schematic which is in .sch but now there also is .kicad_sch.

When not providing it, kibot tries to automatically select the .sch file. 

We need to select the kicad_sch file!!! - then it works.

This implies a change in kibot's automatic file selection.

mdeweerd, 2023-03-01:
Other thing:: maybe kibot should warn about both files being present and suggest to archive them?

set-soft, 2023-03-02:
> While trying to build a test case to emulate the move from KiCAD 6 to KiCAD 7 (moving from KiCAD 5 to 6 first), I stumble on this error (last line):
> 
> ```
> ERROR:At line 1 of `/usr/share/kicad/symbols/Graphic.kicad_sym`: Mixing KiCad 5 and KiCad 6 files is not allowed (kibot - kiplot.py:257)
> ERROR:Line content: `(kicad_symbol_lib (version 20211014) (generator kicad_symbol_editor)` (kibot - kiplot.py:258)
> ```
> 
> This is https://github.com/mdeweerd/kibot-testcases/tree/OLD_FILE_FORMAT_ISSUE/designs/tc6 (tag OLD_FILE_FORMAT_ISSUE)

This folder has `tc6.sch` a KiCad 5 schematic, and `Graphic.kicad_sym` is a KiCad 6 lib.
The error looks quite logic.

set-soft, 2023-03-02:
> This implies a change in kibot's automatic file selection.

May be, but it clearly asks to remove the KiCad 5 files. So you should do it.

mdeweerd, 2023-03-02:
> May be, but it clearly asks to remove the KiCad 5 files.

Two comments:
1. The message does not suggest that the ".sch" file is not ok, but it suggests that there is an issue with usr/share/kicad/symbols/Graphic.kicad_sym .  So I was checking all the libraries rather than thinking of the .sch as the log clearly showed kicad 6 was used.
2. When not providing the '-e SCHEMA' option to kibot, kibot selects the '.sch' file, not the '.kicad_sch' file even if the latter is the only one available.

set-soft, 2023-03-02:
> Two comments:
>
> 1. The message does not suggest that the ".sch" file is not ok, but it suggests that there is an issue with usr/share/kicad/symbols/Graphic.kicad_sym .  So I was checking all the [...]

```
WARNING:(W058) Using old format files is not recommended. Convert them first. (eeschema_do.kiauto.file_util - file_util.py:281) (kibot - kiplot.py:134)
WARNING:(W058) Old file format detected, please convert it to KiCad 6 if experimenting problems (eeschema_do - interposer.py:632) (kibot - kiplot.py:134)
- 'Unique report with characteristics, schematic, etc.' (report) [report]
```

I think this is clear enough: *convert them*

> 2. When not providing the '-e SCHEMA' option to kibot, kibot selects the '.sch' file, not the '.kicad_sch' file even if the latter is the only one available.

Really? I don't think so. The example you show doesn't have a KiCad 6 file.

closed 2023-03-02 by set-soft

mdeweerd, 2023-03-02:
> ```
> WARNING:(W058) Using old format files is not recommended. Convert them first. (eeschema_do.kiauto.file_util - file_util.py:281) (kibot - kiplot.py:134)
> WARNING:(W058) Old file format detected, [...]

This is what I meant - the next test case just runs "kibot designs/tc7/tc7.kibot.yaml schematic_bw" and it reports:

```
ERROR:No SCH file found (*.sch), use -e to specify one. (kibot.gs - gs.py:414)
```

```shell
git clone --branch MINIMAL_OPTS git@github.com:mdeweerd/kbot-testcases.git
cd kbot-testcases
make minimal_opt_k7
```

set-soft, 2023-03-02:
> This is what I meant - the next test case just runs "kibot designs/tc7/tc7.kibot.yaml schematic_bw" and it reports:

KiBot won't detect the files in this way, just try: `cd designs/tc6/ ; kibot`

The detection is for the current directory

commit d656b338 (2023-03-02): Abort when a .sch and .kicad_sch of the same name are detected

Closes #397