INTI-CMNB/KiBot #398: [BUG] Diff - missing NPTH ?

mdeweerd, 2023-03-01:
While trying to make a test case for the shifted diff issue, I encountered this strange effect.

Test case:
```shell
git clone --branch DIFF_MISSING_NPTH git@github.com:mdeweerd/kbot-testcases.git
cd kbot-testcases
docker compose run --rm diff_k7
```

[image]

While the diff output shows this for F.Cu :
[image]

I was expecting to see the same output (no change) - and at least "consistent" output (the same type of holes should have the same output - all big NTPH holes are invisible, but the smaller ones do not all behave the same).

set-soft, 2023-03-07:
Hi @mdeweerd !

1. You are not using full drill marks, so you are seeing *something* where a THT pad is located, but not how it looks. Switching to full drill marks you'll get:
[image]
And this makes much more sense.
2. You must realize that the plot is showing the copper, not the holes! So for the upper left hole you get nothing, this is because the hole and the copper have the same diameter. But the other two holes are wrong. They have some copper because the drill diameter is smaller than the pad diameter. You should fix it. Doing it you get:
[image]

BTW: Do you really have 0.99 mm drill tools? I think this should be 1 mm

closed 2023-03-07 by set-soft

mdeweerd, 2023-03-07:
Hi @set-soft 

Thank you for your feedback.   I am going to check this footprint again.

As far as the 0.99 mm is concerned - it's in the specification as is.

[image]

https://www.tag-connect.com/wp-content/uploads/bsk-pdf-manager/2019/12/TC2030-IDC-Datasheet-Rev-B.pdf

Apparently drill sizes are not in steps of 0.05:
https://www.pcbway.com/blog/Engineering_Technical/PCB_Drill_Sizes__80_to_44_and_everything_between.html

But not all PCB manufactures show a large range of tools:
https://www.eurocircuits.com/pcb-design-guidelines/drilled-holes/

So we can consider that a lack in DFM by Tag-Connect.