INTI-CMNB/KiBot #400: [Bug/Feature] [BUG/Enhancement] Maintain offset coordinates/scale in diffs

mdeweerd, 2023-03-02:
_Originally posted by @mdeweerd in https://github.com/INTI-CMNB/KiBot/issues/392#issuecomment-1448542716_

I think the test case I join here is a different demonstration of the shift issue mentionned int he post above where I observed a shift on the X-Axis when diffing.
[image].

In this test case, I changed the board in the latest version while maintaining a small board in the previous version.
The difference is shown like this:
[image]

I only changed the board size, not the tracks.

I guess that the original issue is related: the cropping/scaling is different between the two versions (for a reason that I did not identify, but the board is the same) and this results in a shift.

You exectute the testcase like this:

```shell
git clone --branch DIFF_PCB_CHANGE git@github.com:mdeweerd/kbot-testcases.git
cd kbot-testcases
docker compose run --rm diff_k7
# or (also launches docker):
make diff_k7
```shell

The difference output can be found in "tc7_01_20211224_header/Diffs" .

set-soft, 2023-03-02:
I don't think this is:

1. Wrong
2. Easy to solve

If you change the board size is hard to compare both. This is because you must know this **before** generating the images, and this isn't how things work.

The position/scale is automatic, and what is worst, jut think about the cache.

The only thing I think that we could do is just inform the problem. But comparing boards of different size isn't something that I think is practical.

mdeweerd, 2023-03-02:
I was trying to reproduce the issue where I did not change the board size.  I think the original issue and this example are very much related.

I did not investigate how the scaling is determined, but IMHO, the original drawing size and offset can (should) be used - and optionnaly truncated after making the difference.

Regarding cache - if the cache Id is based on the size/offset then the cache id can be unique.

set-soft, 2023-03-02:
> I was trying to reproduce the issue where I did not change the board size. I think the original issue and this example are very much related.
> 
> I did not investigate how the scaling is determined, but IMHO, the original drawing size and offset can (should) be used - and optionaly truncated after making the difference.

Now think about somebody just moving the PCB in the sheet. You can't solve all problems without a really high level of complexity.

> Regarding cache - if the cache Id is based on the size/offset then the cache id can be unique.

Is not about the ID (is a hash), is about knowing how to create things beforehand.

You don't know. Because you don't know if the next change will make the board bigger or smaller.

Each image is generated independently so you can compare any pair using the cache.

mdeweerd, 2023-03-02:
> Now think about somebody just moving the PCB in the sheet. You can't solve all problems without a really high level of complexity.

If you're moving the PCB on the sheet, then you can expect to get differences.  The changes I made to the schematic to get back on the grid generate expected differences.

It is not uncommon for a PCB to change size without redoing the routing (entirely).  The PCB can be extended a bit to add some components or reduced on one side to better fit a case.

> Is not about the ID (is a hash), is about knowing how to create things beforehand.

I suppose it is a git hash, but if the hash would be a hash based on <git hash>_<XxY_XoffsetxYoffset>, then this would be unique.

> Each image is generated independently so you can compare any pair using the cache.

I understand that - and I understand the issue of the size.

Maybe an option to keep the size equal to the sheet's size is not too complex to implement and would protect from this - except when somebody moves his PCB on the sheet itself.

set-soft, 2023-03-02:
> Maybe an option to keep the size equal to the sheet's size is not too complex to implement and would protect from this - except when somebody moves his PCB on the sheet itself.

An option that is disabled by default can be workable.

This will make the diffs harder to see (most area wasted in most cases), but usable.

mdeweerd, 2023-03-02:
I've been further trying to close in on what is happening in my design, and the shift also occurs when comparing one KiCAD6 version with the next KICAD6 version just before moving to KICAD7.

I do not see the difference that could incur this shift.

mdeweerd, 2023-03-03:
Here is a nice test case - no change in the design files, but the PDFs show a diff:

[image]

```shell
git clone --branch DIFF_6_NEW git@github.com:mdeweerd/kbot-testcases.git
cd kbot-testcases
# Check the differences with the baseline
# Only the yaml files are different:
git diff --name-only DIFF_6_BASELINE
# Result:
#  designs/tc6/tc6.diff.kibot.yaml
#  docker-compose.yml

# Using docker, launch (by calling docker comlpose):
# kibot -c designs/tc6/tc6.diff.kibot.yaml -b designs/tc6/tc6.kicad_pcb -e designs/tc6/tc6.kicad_sch -g variant=header -v diff_prev_pcb
docker compose run --rm diff_k6
```